zephyrproject-rtos/zephyr #32610: LwM2M : Implement Notification Storing When Disabled or Offline

JPSELC, 2021-02-24:
This is a key piece of functionality to ensure that payloads get sent if the device is offline temporarily.
This is enabled by Resource ID 6 of the LwM2M Server Object.

JPSELC, 2021-02-24:
I think there are a few aspects.
1. Offline Status : If the device is disabled it is clear, but Offline is not so clear. The application may know it is Offline e.g. if the device is not registered on a network, but there could also be a situation where the device is registered but the server is down. This would be picked up by an LwM2M registration update failure or a notification acknowledge timeout, if that was handled.
2. Storing and retrieving the payloads - this will require user functions
3. Managing the forwarding of Stored Payloads when back online.

carlescufi, 2021-02-24:
@JPSELC is this something you plan to send a Pull Request for?

JPSELC, 2021-02-24:
@carlescufi I haven't looked at it in detail yet. It's something I need but I could also handle at application level. I think it will be quite a large PR involving a lot of changes to lwm2m_engine.c in particular

MaikVermeulen, 2021-04-12:
Any progress? I have upcoming projects that would benefit from handling intermittent connectivity. 
I'm definitely willing to jump in on this! Any particulars I can get started on?

JPSELC, 2021-04-12:
Hi @MaikVermeulen, sorry, not made much progress on this, due to other priorities. The initial challenge is to translate a message which may be in TLV format to JSON for forwarding. I think the data should only be sampled once, as a second read (for JSON format) may not return the same results. That may not be important though.

MaikVermeulen, 2021-04-14:
@JPSELC Forgive me, but I don't yet understand why TLV would have to be translated to JSON. Shouldn't we just store all notifications if the device is offline / keep confirmable notifications until acknowledged? Can't those messages can remain in memory as-is?

JPSELC, 2021-04-14:
The key issue is the time information is lost, because when you are sending stored notifications, the time of receipt by the server is not relevant.
The json format requires a timestamp
See https://github.com/eclipse/leshan/issues/595
See also OMA-TS-LightweightM2M-V1_0-20170208-A p56

JPSELC, 2021-04-15:
I appreciate 1.0.2 spec is the latest : OMA-TS-LightweightM2M-V1_0_2-20180209-A.pdf
but nothing changed with regard to notification storing etc.

MaikVermeulen, 2021-05-10:
Hey guys (@rlubos @JPSELC), I think that my company will soon allocate time for me to dive into this.
The JSON formatting itself is already supported by the LwM2M stack I believe. That leaves the TODO list as follows ([also based on @JPSELC 's comment](https://github.com/zephyrproject-rtos/zephyr/issues/32610#issuecomment-785077991)):
- Check LwM2M client online status (currently it is either online or it eventually stops with a network error)
- **If client disabled/offline** (we would need a new state for offline?) (maybe later we would also like to store notifications until they are acknowledged, related to [#31499](https://github.com/zephyrproject-rtos/zephyr/issues/31499)):
1. Store notification as specified by LwM2M 1.0.2 "Historical notification" (JSON)
2. Send stored notifications after coming online again (this "coming online again" will need some work in the engine, as I already wrote above: currently I believe the client is either online or dying towards network error and then stopped / in need of restart)
3. To detect coming online again -> We will need a retry mechanism. E.g. exponential back-off.

 Please add to this list / provide comments / talk about design where-ever you feel it's needed.

JPSELC, 2021-05-10:
The meaning of Offline is probably more ambiguous than I thought. 
My narrow view was that a registration update had failed, maybe caused by the cell going down or the server going down.
Registration would then be attempted periodically, and when successful, all notifications sent.

JPSELC, 2021-05-10:
@rlubos I don't think offline is related to queue mode, because if a notification occurs, I think it's sent (or sending is attempted) immediately. I think queue is more about commands from the server being queued.

MaikVermeulen, 2021-05-10:
> @rlubos I don't think offline is related to queue mode, because if a notification occurs, I think it's sent (or sending is attempted) immediately. I think queue is more about commands from the server being queued.

I agree, Queue mode is more of a server mechanism, a bit like [DRX](https://en.wikipedia.org/wiki/Discontinuous_reception)).

> Does the LwM2M specification clarify what does it mean that the client is offline in terms of notification storing? I. e. does it mean that the client deregisters from the server (I think that wouldn't make much sense as the observations should be cancelled in that case) or rather is it related to the queued mode, where the client is allowed to disable its receiver after some time of inactivity on its side.
> The meaning of Offline is probably more ambiguous than I thought.
> My narrow view was that a registration update had failed, maybe caused by the cell going down or the server going down.
> Registration would then be attempted periodically, and when successful, all notifications sent.

The LwM2M spec doesn't specify what Offline means. It does indicate that it happens in Queue mode, but as @JPSELC indicates, this might not be relevant for us.
Personally I would define Offline as: Device its LwM2M client is running + the below:
- Not (yet) registered
- Lifetime update failed
- CON message ACK timeout after retries

> Also one more question - the "Notification Storing When Disabled or Offline" resource description does not mention any particular data format, yet I assume it only makes sense to apply the notification storing to the JSON format, as it's the only one that provides timestamping of the data values. Does the specification clarify how to handle the situation when the observer requested a different content format with the "Accept" option? I couldn't find any information.

This is true, but "Historical Notifications" or "Historical versions of Notifications" is only mentioned under the JSON sections, and indeed your points are all true. LwM2M specification 1.2 talks about the Historical Queue Size Attribute, the Historical Queue and the Historical Versions of Notifications. Those notificatiosn use the SenML JSON format, and require the timestamps (for other SenML JSON messages, the timestamps are optional):
> Historical notifications MAY be sent in a format as described in Section SenML JSON. If the queue size reaches hqmax and a new reading is received, the oldest reading MUST be dropped. The LwM2M Client SHOULD empty the queue as soon it becomes aware that connectivity has been restored. The use of "hqmax" is dependent on notification storing being enabled via the "Notification Storing When Disabled or Offline" Resource of the LwM2M Server Object.

Lastly, the Accept option and its effect on historical notifications is not specified.

MaikVermeulen, 2021-05-10:
> Thank you for the feedback. Indeed the Queue mode seems to be decoupled from being offline (still it makes perfect sense to me to keep the device in offline mode if queue mode is used).
>
> I wonder if it should also be possible to allow the application to put the LwM2M engine in "offline" state explicitly. For instance when queue mode is used, and the device wants to put the radio down, it could put the engine in the offline mode so that it won't even attempt to send notifications, but queue them instead.

The LwM2M specification indeed specifies that the client only needs to wake up for the registration update, thus being Offline in between. It would indeed be nice if the engine can control this.

> I think we need to settle on the approach here. Personally, I think wouldn't make sense to store notifications for observers that explicitly requested a content format other than JSON.

When the device is online, I would actually prefer TLV in many cases to reduce data consumption. JSON itself is quite inefficient, and when the device is online, the timestamps are not needed.

rlubos, 2021-05-10:
> When the device is online, I would actually prefer TLV in many cases to reduce data consumption. JSON itself is quite inefficient, and when the device is online, the timestamps are not needed.

Sure, but we still should favour the server preference in the first place. Just a quote from the 1.0.2 specification:

> The Accept Option MAY be included in a LwM2M Server data request, to specify the payload Content-Format this Server prefers to receive. The Client returns the preferred Content-Format if available. If this Accept option is not given or if the LwM2M Client doesn’t support that option, the LwM2M Client will use its own preferred data format reported in the Content-Format of the response message. If the preferred Content-Format cannot be returned, then a 4.06 “Not Acceptable” value MUST be sent as a response.

So my understanding is, if the server requests a specific content format, the client should either reply with the requested one, or reject the request. Thus I don't think there's a point of storing the JSON notification for an observer that requested TLV format.
Other thing is when the server does not request a specfic format. In such case I think we're free to swtich between TLV/JSON formats depending whether the engine is in online or offline state.

MaikVermeulen, 2021-05-10:
> Sure, but we still should favour the server preference in the first place. Just a quote from the 1.0.2 specification:
>
>
> So my understanding is, if the server requests a specific content format, the [...]

Completely agree, no comments!

DominikWindhab, 2021-05-28:
I'd like to join the conversation as we are currently dealing with similar issues.

> The key issue is the time information is lost, because when you are sending stored notifications, the time of receipt by the server is not relevant.

It depends on the level of the observer. /6 Location has a "Timestamp" field. And newer versions of sensor values (e.g. /3347 Pushbutton) do have it too. If the observer is on level 0 or 1 (object or instance e.g. /6/0), Zephyrs LWM2M engine will always pack together the changed values into one message, also including the timestamp. If you only observe on resource level  (e.g. /6/0/0) the timestamp would not be transmitted and hence would be lost. Storing notifications as TLV would be totally fine, as long as observers are correctly used.

> So my understanding is, if the server requests a specific content format, the client should either reply with the requested one, or reject the request. Thus I don't think there's a point of storing the JSON notification for an observer that requested TLV format.

Agreed.

> Personally I would define Offline as: Device its LwM2M client is running + the below:
> 
>     * Not (yet) registered
>     * Lifetime update failed
>     * CON message ACK timeout after retries

There isn't much of a difference between 2) and 3), is there? I think lifetime update is also a CON/ACK message.

To recap the discussion: There are two options how to implement "notification storing".

1. Introduce an "offline" state in the LWM2M engine.
2. Handle "offline" state on application level.

I think introducing an offline state is harder than it looks like. One of the biggest problems will be how to recognize that the device is online again. Network coverage is back again, modem has been turned on, server is up again after a maintenance window, etc. There has to be a trigger from the application to tell the LWM2M engine that the device might be probably online again. Or a registration update retry mechanism. And then there is also the question: When to give up and reset the lwm2m engine (and also delete all stored notifications)?

Handling the offline state on application level is something that needs more attention from a developer, but might be something that is doable by extending the current LWM2M engine. It means that the application would be responsible if setting a resource is save (connection available, registration active). So for example take a temperature sensor where measurment is made every 10 minutes but transmission shall take place once a day in bulk. A scenario like that is really hard to implement. The reason is due to how the LWM2M engine works. A thread is running every 500ms checking if a resource that is currently observed changed. If yes a notify message is created.

Currently there is no way of knowing whether one or multple notify messages have been created by the engine. So you can wait for a specific amount of time before your write the resource again or create a workaround in the engine by implementing a function to check that or use a semaphore. And in addition to that you have to make sure that the LWM2M engine thread does not run during the "set" processs. Otherwise you will end up with 2 messages generated by the engine. But this can be achieved by using k_sched_lock(); for example.

Currently I see 4 reasons why notifications shall be stored:

1) Application wants it so (e.g. to save power and transmit more message in a bulk once instead of everytime new data is available)
2) Connection unavailable (e.g. no network coverage OR modem disabled)
3) Not registered at LWM2M server (e.g. lifetime update failed)
4) CON message ACK timeout after retries (e.g. server did not respond properly)

(4) is when the LWM2M engine created a message for all of the observers but does not receive an ACK in time after a couple of retries. The message will be dropped silently and is lost.

1,2 and 3 are in the hand of the application (3 because reaction to callback of rd_client_event are possible). But implementing a mechanism that allows a developer to "buffer" data is currently very hard. 

I think (4) is a rare case (if 1,2,3 is checked by the application). But 1,2,3 is much more important. Wouldn't it be better to extend the engine in way to make it easy for developers to decide on their own when to set resources and create messages? I think the complexity of implementing an "offline" state that stores notifications is very hard. Instead giving the application developer the tools to decide if a value shall be stored or a notify shall be created immediatly is easier.

By reading the current value of "Notification Storing When Disabled or Offline" the developer can decide whether to buffer the data for later transmission or to discard it immediatly.

DominikWindhab, 2021-06-02:
I was thinking a bit more about this topic.

[This](https://github.com/OpenMobileAlliance/OMA_LwM2M_for_Developers/issues/429#issuecomment-576451186) and [this](https://github.com/OpenMobileAlliance/OMA_LwM2M_for_Developers/issues/429#issuecomment-576536165) post are summarizing the problem problem pretty well.

As an update to LWM2M 1.1 is out of scope, how about the following solution: Create an method that can set the lwm2m stack "online" or "offline" manually. So if for example the application recognizes that there is no GSM network connection we'd put the stack offline. This means no communication will take place initiated by the stack. No notifies, no registration updates, etc.

However, writing of resources by the application would still take place as if nothing happend. When the stack is put in "offline" mode, instead of sending the notifications they would be stored in an internal buffer in RAM. As soon as the stack is put "online" again the notifies would be sent to the backend.

However this not a perfect solution because of the following reasons:

1) If observers are not on instance level timestamps are lost in TLV format.
2) If the registration is not valid anymore (and hence also the observers do not exist or have a different ID) after getting online again all the notifies will be rejected by the server.
3) Resetting the stack (due to a reboot, or to recover from an error) means that all messages stored so far would be lost.

So there is still no guarantee that a message arrives at the backend, but it would be an improvement. I could think of an extentsion for the future to store the notifies in NVM to fix 3). But then there need to be a logic that assumes that an observer with the same path (and maybe also same conditional settings pmin/pmax/lt/gt/etc.) will be created after a new registration of the client. And then the notifies have to be altered to fit that new observer (to fix 2).

TL;DR: How about setting online/offline only via the application and buffer notifies in a local buffer in RAM?

MaikVermeulen, 2021-06-02:
> It depends on the level of the observer. /6 Location has a "Timestamp" field. And newer versions of sensor values (e.g. /3347 Pushbutton) do have it too. If the observer is on level 0 or 1 (object [...]

Notification buffering indeed only makes sense on the higher observer levels where the Timestamp resource is bundled with the measurement. And that's fine, as there plenty of sensors where measurements are actually bundles anyhow. E.g. accelerometer.
This does mean however, that in formats that do not inherently include timestamps (so I believe anything except SenML / JSON), we need to explicitly require the timestamp resources to be added. 
**Remark: Do we expect the user to start adding timestamps everywhere, or do we add them in the storing functionality when storing them?** 
 
> To recap the discussion: There are two options how to implement "notification storing".
>
> 1. Introduce an "offline" state in the LWM2M engine.
> 2. Handle "offline" state on application level.
>
> I think [...]
I think we should add a [Network Management Event Callback](https://docs.zephyrproject.org/latest/reference/networking/net_mgmt.html) to the LwM2M client or engine to handle these cases. The network interfaces (modem or others) should correctly implement the Network Device functionality like those events. The client would go offline when its network interface goes offline. Of course, we also need the handling for when the server is not replying anymore. On top of that, the problems that are not directly client-sided, like an offline server, can be handled by an exponential back-off mechanism that the LwM2M engine/client handles internally.
**Sidenote: I'm currently also looking to the correct handling of IP changes in the Zephyr LwM2M client, those would require this callback as well**
**Sidenote 2: What I currently see is that if the LwM2M stack gets ENETDOWN errors, it will still keep going, while it shouldn't**

> Currently I see 4 reasons why notifications shall be stored:
> 
> 1. Application wants it so (e.g. to save power and transmit more message in a bulk once instead of everytime new data is available)
> 2. Connection unavailable (e.g. no network coverage OR modem disabled)
> 3. Not registered at LWM2M server (e.g. lifetime update failed)
> 4. CON message ACK timeout after retries (e.g. server did not respond properly)
> 
> 1,2 and 3 are in the hand of the application (3 because reaction to callback of rd_client_event are possible). But implementing a mechanism that allows a developer to "buffer" data is currently very hard.
For option 1 indeed the application would need to have a way of triggering the disabled mode. For 2, 3, 4 though, I think those can and should be handled internally.
> I think (4) is a rare case (if 1,2,3 is checked by the application). But 1,2,3 is much more important. Wouldn't it be better to extend the engine in way to make it easy for developers to decide on [...]
I think 2, 3, 4 are equally likely as they are closely related. Personally I feel like actually (1) is the odd case, as it's the only one that the engine can not detect itself but actually requires external control for.

> As an update to LWM2M 1.1 is out of scope, how about the following solution: Create an method that can set the lwm2m stack "online" or "offline" manually. So if for example the application [...]
Please see my proposal of [Network Management Event Callback](https://docs.zephyrproject.org/latest/reference/networking/net_mgmt.html) further up above.
> However, writing of resources by the application would still take place as if nothing happend. When the stack is put in "offline" mode, instead of sending the notifications they would be stored in [...]
Aren't all notifications currently already kept in RAM until acknowledged? We would only need to halt the transmission logic when offline/disabled.
> However this not a perfect solution because of the following reasons:
> 
> 1. If observers are not on instance level timestamps are lost in TLV format.
I already mentioned this above, but I think that because there are objects for which a measurement is actually a collection of resources in an object anyway, I agree that this is an OK limitation.
> 2. If the registration is not valid anymore (and hence also the observers do not exist or have a different ID) after getting online again all the notifies will be rejected by the server.
This one is indeed quite tough. We could keep the notifications for a short timeframe after reconnection, and only send them after the observe has started again. Because if not observed anymore, I guess the server isn't interested in them anymore anyway.
> 3. Resetting the stack (due to a reboot, or to recover from an error) means that all messages stored so far would be lost.
I think this one is perfectly OK, they should both be very rare cases.

> > TL;DR: How about setting online/offline only via the application and buffer notifies in a local buffer in RAM?
> 
> My personal impression was that it should be the application that puts the stack into the offline mode, either based on some other sources than lwm2m engine (for instance in nRF91 you can get notifications about the LTE connection state from some other library) or based on the network error event, generated by the lwm2m lib (on several consecutinve communication failures). But let's hear from others interested in the feature.
Please see my proposal of [Network Management Event Callback](https://docs.zephyrproject.org/latest/reference/networking/net_mgmt.html) further up above.

PS: Later today I will probably create and issue and hopefully start on a PR to address correctly handling IP changes in the LwM2M client, using the [Network Mangement API](https://docs.zephyrproject.org/latest/reference/networking/net_mgmt.html) further up above. After that, I hope to have learned enough to write-up a first draft proposal for implementing the LwM2M offline mode and the notification storing.

JPSELC, 2021-06-04:
Regarding : 4. CON message ACK timeout after retries (e.g. server did not respond properly), I think there should be a Kconfig option which enables the calling of engine_trigger_update as per https://github.com/zephyrproject-rtos/zephyr/issues/31499, which would then result in 3. Not registered at LWM2M server (e.g. lifetime update failed), so there are less conditions to be handled.

DominikWindhab, 2021-06-07:
> Aren't all notifications currently already kept in RAM until acknowledged? We would only need to halt the transmission logic when offline/disabled.

True. But I'm not sure if we would be able to change the observer id after the notify has been created in case of "2) registration not valid anymore" at the moment, are we? Anyways, the lwm2m engine must not throw away any notifications unless at least the server responded (or the buffer is full).

> This one is indeed quite tough. We could keep the notifications for a short timeframe after reconnection, and only send them after the observe has started again. Because if not observed anymore, I guess the server isn't interested in them anymore anyway.

Agreed.

>I think the application only needs to control actively disabling the client to e.g. preserve power.

What are you thinking of? There are two options to do this:
1) unregister from server and turn down the entire stack
2) tell the engine that it should behave as if "offline/disabled" and store notifications until turned up again

> I think 2, 3, 4 are equally likely as they are closely related. Personally I feel like actually (1) is the odd case, as it's the only one that the engine can not detect itself but actually requires external control for.

We are using Nordics nRF91 and I'm not sure what happens if network is not available or modem was turned off. @rlubos is there a difference? Do you know what's going on under the hood and whether this is compatible to what @MaikVermeulen suggests with the Network Management API?

Probably this isn't even necessary. If notifies are not removed until the server responded, and there is a timeframe after a new registration to wait for observers, the online/offline state would just be depending on successful registration/registration updates. Is there a benefit to go down to network-interface level?

rlubos, 2021-06-07:
> I think we should add a Network Management Event Callback to the LwM2M client or engine to handle these cases. The network interfaces (modem or others) should correctly implement the Network Device [...]

That sounds like a good idea, although I think the Network Management events should rather be handled on the application side, and the LwM2M library could expose APIs to put the engine offlline/online. This should fit nicely with the offloaded interafaces as well.

> Aren't all notifications currently already kept in RAM until acknowledged? We would only need to halt the transmission logic when offline/disabled.

Actually, it should be even easier after some recent changes in the LwM2M engine, where the notifications are not sent immediately after creation, but just put on the queue. This way we could just halt the TX queue for the time the engine is offline. 

Still, it seems to me that the engine should not only delay the notification, but also be able to modify them. If you look at the LwM2M specification example of the historical representations, it puts multiple resource values into a single notification message - it'd be more efficient to pack them like this. And if not, we still need to put the "Base Time" and "Time" fileds to the notification message for the historical ones, so the engine needs to know it's online when creating the notification message.

>> If the registration is not valid anymore (and hence also the observers do not exist or have a different ID) after getting online again all the notifies will be rejected by the server.

> This one is indeed quite tough. We could keep the notifications for a short timeframe after reconnection, and only send them after the observe has started again. Because if not observed anymore, I guess the server isn't interested in them anymore anyway.

Yeah I agree this seems to be a tough one. The simplest approach would be to discard all of the stored notifications since the server **have to** renew the observations on new registration according to the spec, but I guess that's not what we aim for. The poposed solution to keep the notifications for a short period after new registration from the client, and discard them only if the server does not renew the observation sounds fine. But as mentioned in some other post before, it'd might be tricky to upate the notification message with a new Token etc.

> We are using Nordics nRF91 and I'm not sure what happens if network is not available or modem was turned off. @rlubos is there a difference? Do you know what's going on under the hood and whether this is compatible to what @MaikVermeulen suggests with the Network Management API?

The LTE Link Controller module for nRF91 allows to register an event callback, which in order allows to detect when the LTE link goes down (see https://github.com/nrfconnect/sdk-nrf/blob/master/applications/asset_tracker_v2/src/modules/modem_module.c#L161 for instance). It should work simillarly to the Network Management, although it's a speratate library, that's why I suggest to handle the network connection status out of the LwM2M engine.

MaikVermeulen, 2021-06-08:
Hey guys, great to see the continued interest and ongoing discussion.
I have tried to make a summary to hopefully help us align our expectations, and possibly to help others interested in the topic get up to speed quickly.

Let me know if I forgot or misinterpreted anything, or if anything is unclear. I will edit the post.
@JPSELC maybe you can add the summary to the original post, or a link to this summary?

### Summary
Goal: Preventing loss of data while experiencing intermittent connectivity, and to better facilitate low power devices.
How: By buffering notifications when the LwM2M client is offline due to connectivity problems, or disabled to save power.

**Implementation**
- Introduce LwM2M Online, Offline and Disabled states
- Expose APIs to control the states to the application, e.g. by Network Management API or LTE Link Controller
- Only buffer notifications for level 2 observes, as the timestamp resource needs to be included
- Buffered notifications are stored in the format requested via the Accept option
	For JSON, timestamps are standard (so must be added)
	For TLV, timestamps are not standard, but must be added manually by the engine to support historical notifications
	For Plain Text and Opaque, buffering is not supported as these formats are only intended for singular resources
- Notification TX queue must be disabled when in the Offline or Disabled states, and enabled again when Online or Enabled again
- Dealing with re-registration:
	- Notifications are kept for some time after re-registration, removed if no new observes are started for these notifications
	- Notification observer ids need to be updated after re-registration
- Notifications are removed when closing the client context, or after observes are cancelled
- Reply timeouts (e.g. due to server offline) can be handled by the RD client Network Error event

**Open issues**
@rlubos @DominikWindhab @JPSELC Please let me know your thoughts about the following open issues (I will update the summary with the agreed upon approaches):
- Do we want to make a distinction between putting the engine offline and disabling it? 
  The only distinction I can think of is that we would maybe want to keep the LwM2M registration alive in disabled mode. That way the server can better tell if something is wrong, and it gives us a periodic window to unload the buffered notifications.

JPSELC, 2021-06-08:
Thanks Maik,
Just one thing. I think the buffering of the messages should be handled by the application, if possible. In a mains-powered application, the messages would need to be stored in Non-volatile memory, whereas a battery-powered application would store in RAM.

DominikWindhab, 2021-06-09:
>Only buffer notifications for level 2 observes, as the timestamp resource needs to be included

Shouldn't this be the opposite? Only level 0 and 1 are allowed? Timestamps would then be part of the instance (if present). Otherwise there is just no timestamp. We could prevent level 2 observes but should we really do that? They would be transmitted without a time information which could be fine. Level 0/1 observes would anyways come with the time.

>For TLV, timestamps are not standard, but must be added manually by the engine to support historical notifications

See before. I think for TLV we cannot add time manually. It comes with the instance or it does not come at all.

>Do we want to make a distinction between putting the engine offline and disabling it?

There is the executeable resource "/1/0/4 Disable" which referes to the "Disabled" in the "/1/0/6 Notification Storing When **Disabled** or Offline". So from my perspective there is an important difference:

- Disabled is triggered by the server
- Offline is triggered by the device-application

Disabled needs a complete unregistration from the LWM2M server. Being offline means that we don't want to unregister, there is just no network connection for example.

>Just one thing. I think the buffering of the messages should be handled by the application, if possible. In a mains-powered application, the messages would need to be stored in Non-volatile memory, whereas a battery-powered application would store in RAM.

I think we agreed on having the buffer in RAM for now. The LWM2M engine takes care of buffering depending on the state. I think having buffering in NVM could be something for future improvements.

The rest looks good to me.

MaikVermeulen, 2021-06-09:
> Shouldn't this be the opposite? Only level 0 and 1 are allowed? Timestamps would then be part of the instance (if present). Otherwise there is just no timestamp. We could prevent level 2 observes [...]

I believe in [the LwM2M engine that the levels start counting from 1](https://github.com/zephyrproject-rtos/zephyr/blob/dce85ebe450053ac1c44e03075cea745cc4f1840/subsys/net/lib/lwm2m/lwm2m_engine.c#L1305). So object level = 1, object instance level = 2, resource level = 3, resource instance level = 4. 

But I agree with your points. If observed specifically at the resource level, apparently no other information is required, but we can still send them. I'm just struggling with thinking of a use-case for this.

> > For TLV, timestamps are not standard, but must be added manually by the engine to support historical notifications
> 
> See before. I think for TLV we cannot add time manually. It comes with the instance or it does not come at all.

OK, so we expect the user to add and fill-in the timestamp resource, that's fine. This should also make our lives of implementing the buffering easier.

> There is the executeable resource "/1/0/4 Disable" which referes to the "Disabled" in the "/1/0/6 Notification Storing When **Disabled** or Offline". So from my perspective there is an important [...]

I can live with this. I guess that any server-sided monitoring should be aware of the registration behavior of the different clients anyway.
Though, I think we need to dig deeper in how to correctly handle the disabled state. How would it become enabled again?

> I think we agreed on having the buffer in RAM for now. The LWM2M engine takes care of buffering depending on the state. I think having buffering in NVM could be something for future improvements.

My preference would also be to first focus on just RAM. How important is it for your @JPSELC ?

DominikWindhab, 2021-06-09:
>I believe in the LwM2M engine that the levels start counting from 1

Yes, sorry, my fault. You are right.

>I'm just struggling with thinking of a use-case for this.

Me too, at the momentI cannot think of a useful use-case. But on the other side, why prevent it? I think it should not behave differently when using JSON, but I'd be happy with both.

>Though, I think we need to dig deeper in how to correctly handle the disabled state. How would it become enabled again?

I think this is what "/1/0/5 Disable Timeout" is for. So it should happen automatically after that timeout.

JPSELC, 2021-06-09:
NVM Buffering is critical for me, unfortunately. Just wondering if we could add hooks for that functionality?
Maybe by default use RAM buffering, but have a mechanism to bypass that via callbacks - a store callback and a restore callback which would allow handling of NVM storage in the application.

MaikVermeulen, 2021-06-09:
> NVM Buffering is critical for me, unfortunately. Just wondering if we could add hooks for that functionality?
> Maybe by default use RAM buffering, but have a mechanism to bypass that via callbacks - a store callback and a restore callback which would allow handling of NVM storage in the application.

To me this sounds perfectly fine. It sounds like it would neatly follow the API the engine already has for the write and read and other callbacks.

DominikWindhab, 2021-06-10:
@MaikVermeulen Are you planning to implement what we are discussed? If you need help or if I can support you please just reach out to me.

MaikVermeulen, 2021-06-10:
> @MaikVermeulen Are you planning to implement what we are discussed? If you need help or if I can support you please just reach out to me.

Yes! Currently this functionality is planned for our next sprint (in about 2 weeks time).
It would be nice if all interested parties can monitor and review the changes as we go, and of course contribute if they like.

@rlubos If you could provide some final feedback I'll make sure the summary is completely up to date.
Furthermore, what would be the best way to cooperate? A shared fork and a draft PR?

DominikWindhab, 2021-06-11:
>In general, personally I'd focus on the JSON notifications first, then it could be decided whether to extend the mechanism for TLV notifications for objects wtih timestamp resource.

I agree that the topic around the timestamp is not easy to solve. But I am not sure if focusing on JSON would be the ideal solution. Usually the server sends a request specifiying the data-format. And from my experience TLV is somwhat standard and is used unless there are specific reasons to use another one. The specification says:

>if the LwM2MClient doesn’t support that option or if the LwM2MServer expresses no data format preference, the LwM2MClient will use its own preferred data format

So as far as I understand it would be possible to respond with JSON although the server requested TLV. I wanted to test this with a backend but I think this would be some work in the engine. A quick-try by removing all writers except JSON in select_writer and do_write_op did not work for me.

Anyhow, to make this work one of the following must be implemented:
1. all responses must be encoded in JSON if notification storing is active always ignoring the requested encoding type OR
2. the selection of response encoding is dependend on the online/offline state of the engine OR
3. the observe must always request JSON if notification storing is active

And in addition to that it requires to have explicit support for JSON using the CONFIG_LWM2M_RW_JSON_SUPPORT=y
This is not always desired by the developer of a constraint system.

>I'm not sure though if sending an outdated data in a notification, w/o a timestamp resource is a good idea though - it doesn't seem to be covered by the spec, and while technically possible, it might be confusing on the server side applications (i.e. how to tell whether the data is outdated or not?).

I think there is not much difference between JSON and TLV, It's just the writer of the content that differs. Anything else should be pretty much the same. I'd really prefer not to change the logic which content encoding to use. I could live with (3) but I do not see much advantage of it.

rlubos, 2021-06-11:
> So as far as I understand it would be possible to respond with JSON although the server requested TLV. I wanted to test this with a backend but I think this would be some work in the engine. A quick-try by removing all writers except JSON in select_writer and do_write_op did not work for me.

Then we have a different understanding of the spec. I've already quoted the specificaction part that makes this requirement (see https://github.com/zephyrproject-rtos/zephyr/issues/32610#issuecomment-836783211). The part you quote, says only that the client may reply with whatever prefered content format only if server **did not specify** its preference or the **Accept option is not handled** by the client. Otherwise, the client must reject the request if it's not able to reply with the selected content format. For me this is a clear requirement that if the server requests TLV, the client shall reply with TLV.

> I think there is not much difference between JSON and TLV, It's just the writer of the content that differs.

The difference is significant when it comes to notification storing. The JSON format has built-in "Time" and "Base Time" fields, while TLV does not, therefore with JSON you can clearly indicate the timestamp for any resource in the notification. Let's not mix in the Timestamp resource, which is only present in a few IPSO objects (and the Location). 

> I could live with (3) but I do not see much advantage of it.

From my point of view, the third option would be the only one that is compliant with the LwM2M specification (at least 1.0.2, I didn't look into the newer ones).

JPSELC, 2021-06-11:
@rlubos You said above in your earlier referenced comment : "In such case I think we're free to swtich between TLV/JSON formats depending whether the engine is in online or offline state."  
Then : "From my point of view, the third option would be the only one that is compliant with the LwM2M specification" 

Can you confirm that you still consider that there are 2 cases that should work for notifications:
1. If content format is not specified, client can use TLV normally and JSON for notifications
2. If content format is specified, it must be JSON, if notifications are required.

Thanks!

rlubos, 2021-06-11:
> @rlubos You said above in your earlier referenced comment : "In such case I think we're free to swtich between TLV/JSON formats depending whether the engine is in online or offline state."
> Then : "From my point of view, the third option would be the only one that is compliant with the LwM2M specification"
> 
> Can you confirm that you still consider that there are 2 cases that should work for notifications:
> 
> 1. If content format is not specified, client can use TLV normally and JSON for forwarding stored notifications
> 2. If content format is specified, it must be JSON, if forwarding of stored notifications is required.
> 
> Thanks!

Ah, in my previous post I replied on a premise that the 3 cases are all about the situation when the server requested some specific content format with an Accept option. But indeed it might not be the case when I read it again. So yeah, I did not change my thinking, if the Accept option was not specified, then the client is free to use whatever content format that suits the client at the moment. The two cases you've listed are indeed **how I see it** in order to be compliant with the spec. Of course, we can discuss this if you disagree or if I miss something in the spec.

DominikWindhab, 2021-06-11:
>For me this is a clear requirement that if the server requests TLV, the client shall reply with TLV.

You are right. I did not read carefully enough. It's about supporting the "option" itself and not about the content of the option.

>Let's not mix in the Timestamp resource, which is only present in a few IPSO objects (and the Location).

This would mean notification are stored if the flag "Notification Storing When Disabled or Offline" is enabled and observer requested JSON. In the end the difference in the code ist just one "if" statement checking the accept-option of the observer to decide whether to put the message in the local buffer or not, So the differences should be in a manageable amount if we would like to try it out also with TLV.

MaikVermeulen, 2021-06-11:
> From my point of view, the third option would be the only one that is compliant with the LwM2M specification (at least 1.0.2, I didn't look into the newer ones).

I agree with @rlubos. Being compliant with the standard should be our main concern, incompatibility issues will likely be worse than slight inconvenience w.r.t. the format.

I also checked [spec 1.2](https://md2html-tool.com/docs/OpenMobileAlliance/LwM2M/master/c47850f/TS_Core/OMA-TS-LightweightM2M_Core-V1_2-20201110-A_full.html) (to see if perhaps OMA changed its thinking. It didn't):
> An LwM2M Server data request MAY contain an option specifying the data formats the Server would prefer to receive for the payload; if this data format is not accepted by the LwM2M Client, the request is rejected; if the LwM2M Client doesn't support that option or if the LwM2M Server expresses no data format preference, the LwM2M Client will use its own preferred data format.

Which IMHO opinion doesn't leave us a lot of playing room. No accept option, client is free to choose. Accept option, client must adhere if it can, may try another if it can't. (So just like in 1.0.2)

Thinking more about it I don't think this should be a problem though. The specification only mentions "Historical notifications" or "Historical versions" in the SenML JSON section, but it I don't think it specifies that no other formats are allowed to do it.
This left me with the following thinking:
- We are free to support both JSON and TLV.
- JSON would always work as it includes inherently timestamps, so it will work for all levels of observes.
- TLV would require the application to add the timestamp resource themselves, so it will only work for level 1 and 2 observes. (This is still compliant with the standard, as long as users that prefer TLV with timestamps adhere to the standard object definitions. If they do not, they can handle that between the client and backend, and the violation is their own choice.)

> Can you confirm that you still consider that there are 2 cases that should work for notifications:
>
> 1. If content format is not specified, client can use TLV normally and JSON for forwarding stored [...]

So to me, the format would be free to choose still. If the user uses JSON, it will always be OK. If the user chooses TLV, they are responsible for the timestamps themselves.

That way this should be solved too:
> In general, personally I'd focus on the JSON notifications first, then it could be decided whether to extend the mechanism for TLV notifications for objects wtih timestamp resource. I'm not sure though if sending an outdated data in a notification, w/o a timestamp resource is a good idea though - it doesn't seem to be covered by the spec, and while technically possible, it might be confusing on the server side applications (i.e. how to tell whether the data is outdated or not?).

JPSELC, 2021-06-11:
I think there could be an option if TLV is specified, where the stored notifications are forwarded in TLV format, without modification or addition of timestamp, and let the user worry about whether a timestamp is included in the object or not. 
For example, a custom object could be used for regular payload which includes a timestamp.
Perhaps we could have a KConfig option to enable this?

rlubos, 2021-06-11:
> Perhaps we could have a KConfig option to enable this?

Well I was actually going to suggest that if we want to do something that is not fully spec-compliant, then fine, but let's make it optional (and disabled by default). So Kconfig option for such features are a way to go.

> I think there could be an option if TLV is specified, where the stored notifications are forwarded in TLV format, without modification or addition of timestamp, and let the user worry about whether a timestamp is included in the object or not.

Well, I don't think we should add a custom resource to a predefined object in any case (that's why I mentioned it would only work with serveral IPSO objects or Location, or a custom object specified by a user).

MaikVermeulen, 2021-06-11:
> Well I was actually going to suggest that if we want to do something that is not fully spec-compliant, then fine, but let's make it optional (and disabled by default). So Kconfig option for such [...]

Indeed, I don't think we need to support non-compliant setups:

If users want timestamped data for standard objects that include the timestamp resource, they are free to choose JSON or TLV. If they want timestamped data for standard objects without timestamp resource, they have to use JSON. If however, they really only want TLV, they will have to create their own object.

If someone really wants TLV with the timestamp resource for a standard object that doesn't support it, apparently they already are modifying (creating custom objects) in their backend anyways, so why not just become compliant.

DominikWindhab, 2021-06-11:
>So to me, the format would be free to choose still. If the user uses JSON, it will always be OK. If the user chooses TLV, they are responsible for the timestamps themselves.

> Well, I don't think we should add a custom resource to a predefined object in any case (that's why I mentioned it would only work with serveral IPSO objects or Location, or a custom object specified by a user).

> If users want timestamped data for standard objects that include the timestamp resource, they are free to choose JSON or TLV. If they want timestamped data for standard objects without timestamp [...]

Totally agree.

JPSELC, 2021-06-11:
So, if I can summarise, again, sorry!

1. If content format is not specified, client can use TLV normally and JSON for forwarding stored notifications
2. If content format is specified as JSON, forwarding of stored notifications is done in JSON format as per specification
3. If content format is specified as TLV _and_ Kconfig Option for TLV Notification Forwarding is set, notifications are forwarded as composed.

I think 1 and 2 are the highest priority.
Regarding whether JSON forwarded notifications are packed or not, I'm not sure we discussed this? I mean where multiple time samples and corresponding time offsets are packed for each resource.

MaikVermeulen, 2021-06-11:
> 1. If content format is not specified, client can use TLV normally and JSON for forwarding stored notifications

Yes. I don't think the standard prohibits switching of formats in this case, so TLV and JSON can be used and switched at will.

> 2. If content format is specified as JSON, forwarding of stored notifications is done in JSON format as per specification

Yes.

> 3. If content format is specified as TLV _and_ Kconfig Option for TLV Notification Forwarding is set, notifications are forwarded as composed.

Almost. I don't think we need a special option for this because (if I'm not mistaken), although the standard specifies historical notifications only in the SenML JSON section, it never states that other formats are prohibited from historical versions. So for TLV, I think we are free to use notification buffering, but the application has to make sure that objects that support the timestamp resource are used. And the observation has to be at object or object instance level.

> Regarding whether JSON forwarded notifications are packed or not, I'm not sure we discussed this? I mean where multiple time samples and corresponding time offsets are packed for each resource.

I don't completely follow. The standard specifies that SenML JSON looks e.g. like this:
```
{“bn”:“/72/“,
 “e”:[
 {"n":"1/2","v":22.4,"t":-5},
 {"n":"1/2","v":22.9,"t":-30},
 {"n":"1/2","v":24.1,"t":-50}],
"bt":25462634
}
```

JPSELC, 2021-06-11:
@MaikVermeulen You think we don't need a KConfig option for 3. ? The reason I suggest that is to a naïve user, they may be expecting all to work well with TLV, and then the notifications may not have any time information, in which case forwarded notifications are effectively garbage. Explicitly enabling it, would be taking the responsibility to include the time information.

Regarding your example of SenML JSON, yes, I agree, that is the standard.
A first attempt might just send each notification separately?
But you are right, ultimately it is better to parse all the stored notifications to pack into this format.

JPSELC, 2021-06-23:
https://github.com/zephyrproject-rtos/zephyr/commit/3138d89fd966abdc243e972c7e2ed8f64f8a6b69 is now added which resolves https://github.com/zephyrproject-rtos/zephyr/issues/31499, and can be used to automatically trigger a registration update when notify timeout occurs.

Simply add this function
static void notify_timeout_cb(void)
{
	lwm2m_rd_client_update();
}

and assign the lwm2m_ctx notify_timeout_cb
e.g. 	client.notify_timeout_cb = notify_timeout_cb;
before the application calls lwm2m_rd_client_start();	
e.g. lwm2m_rd_client_start(&client, dev_str, flags, rd_client_event);

MaikVermeulen, 2021-07-29:
Hi guys, it's been a while. Just wanted to let you that I've not forgotten about this functionality.
I've been a bit swamped with other tasks. Now, holidays are coming up.

I will let you know when I get to this, but of course feel free to start :smiley:!

MaikVermeulen, 2021-09-27:
Good news, from this week onward, I'll get time to work on this.

I think the easiest way would be to resolve #38531.
Then the application can register that callback, and on each ACK just update the resource values to trigger a new notification. The user can choose whether to include the timestamp resource etc. 
Actual storage method of data is completely left to the user application.

Furthermore, if e.g. a timeout is received, the application would just keep the data stored until ACKed.
The application should listen for the event that the registration/observe has become active (so the client has been enabled again), so it knows when to try again.

DominikWindhab, 2021-09-29:
You mean basically, instead of storing the messages temporarily within the LWM2M engine, the user should store them and just set the LWM2M resources not before an ACK of a previous notification arrived.

I'm not sure if this is something that is easy to implement. First thing is that the user can set different resources but the engine will collect all changes, put them in one ore more notifications (depending on the state of observations) and creates a token for each notification that will then in the end be acknowledged by the server. So theres a gap between resources changed by the user and ACKed messages from the server. The user isn't aware of observations, tokens, lwm2m messages etc.

The second problem is the minimum period (pmin) parameter. If this isn't set to 0 then you have to wait for at leas pmin until a new notification can be sent. As we are storing the notifications from earlier times this parameter might be neglectable, but it is enforced by the engine. It just won't send the update until pmin timed out. If pmin would be longer, seconds, or even minutes it would take very long to transmit the buffered values.

MaikVermeulen, 2021-09-29:
For everyone interested:
```
Currently on Github I see that there are 3 or so issues all related to buffering/handling intermittent connectivity.

For focussed discussion, I think it would be nicer to have a central thread on it.
I’ve made a thread ‘Client offline mode, data buffering, etc.’ in the LwM2M channel on the Discord.
```

rlubos, 2023-11-22:
Zephyr LwM2M implementation took a different approach than proposed here. Instead of storing serialized notifications, a [resource data cache](https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/net/lib/lwm2m/Kconfig#L212) was introduced which allows to store timestamped historical values of resources. This allows to send historical data when the device is back online, when notification is generated. Therefore, it seems there's no longer a need to store notifications themselves.

I suggest hereby that we close the issue, since there is an alternative for notification storing. Please reopen, if for some reason notification storing would still be considered a desired approach.

closed 2023-11-22 by rlubos