zephyrproject-rtos/zephyr #32619: samples: usb: audio: Samples for usb audio fail building

PerMac, 2021-02-24:
**Describe the bug**
Samples from /zephyr/samples/subsys/usb/audio fail during building (at least on nrf52840 and nrf5340).

**To Reproduce**
Steps to reproduce the behavior:
1. go to your zephyr dir
2. run: `./scripts/twister -T samples/subsys/usb/audio/ -p nrf52840dk_nrf52840 -v -v --inline-logs`
3. See error

**Expected behavior**
The build works

**Impact**
showstopper

**Logs and console output**
Part of the logs
```
/home/maciej/zephyrproject2/zephyr/subsys/usb/class/audio/audio.c:978:1: note: in expansion of macro 'UTIL_LISTIFY'
  978 | UTIL_LISTIFY(HEADSET_DEVICE_COUNT, HEADSET_DEVICE, HS)
      | ^~~~~~~~~~~~
/home/maciej/zephyrproject2/zephyr/include/net/buf.h:1114:63: error: expected ';', ',' or ')' before '=' token
 1114 |  static const struct net_buf_pool_fixed net_buf_fixed_##_name = {      \
      |                                                               ^
/home/maciej/zephyrproject2/zephyr/subsys/usb/class/audio/audio.c:947:2: note: in expansion of macro 'NET_BUF_POOL_FIXED_DEFINE'
  947 |  NET_BUF_POOL_FIXED_DEFINE(name, 5, size, net_buf_destroy)
      |  ^~~~~~~~~~~~~~~~~~~~~~~~~
/home/maciej/zephyrproject2/zephyr/subsys/usb/class/audio/audio.c:969:2: note: in expansion of macro 'DEFINE_BUF_POOL'
  969 |  DEFINE_BUF_POOL(audio_data_pool_hs_##i, EP_SIZE(dev##_HP, i)); \
      |  ^~~~~~~~~~~~~~~
/home/maciej/zephyrproject2/zephyr/include/sys/util_loops.h:1086:2: note: in expansion of macro 'HEADSET_DEVICE'
 1086 |  F(0, __VA_ARGS__)
      |  ^
/home/maciej/zephyrproject2/zephyr/include/sys/util_internal.h:85:36: note: in expansion of macro 'Z_UTIL_LISTIFY_1'
   85 | #define UTIL_PRIMITIVE_CAT(a, ...) a##__VA_ARGS__
      |                                    ^
/home/maciej/zephyrproject2/zephyr/subsys/usb/class/audio/audio.c:978:1: note: in expansion of macro 'UTIL_LISTIFY'
  978 | UTIL_LISTIFY(HEADSET_DEVICE_COUNT, HEADSET_DEVICE, HS)
      | ^~~~~~~~~~~~
/home/maciej/zephyrproject2/zephyr/subsys/usb/class/audio/audio.c:847:13: error: 'audio_receive_cb' defined but not used [-Werror=unused-function]
  847 | static void audio_receive_cb(uint8_t ep, enum usb_dc_ep_cb_status_code status)
      |             ^~~~~~~~~~~~~~~~
/home/maciej/zephyrproject2/zephyr/subsys/usb/class/audio/audio.c:777:12: error: 'usb_audio_device_init' defined but not used [-Werror=unused-function]
  777 | static int usb_audio_device_init(const struct device *dev)
      |            ^~~~~~~~~~~~~~~~~~~~~
/home/maciej/zephyrproject2/zephyr/subsys/usb/class/audio/audio.c:761:12: error: 'audio_class_handle_req' defined but not used [-Werror=unused-function]
  761 | static int audio_class_handle_req(struct usb_setup_packet *pSetup,
      |            ^~~~~~~~~~~~~~~~~~~~~~
/home/maciej/zephyrproject2/zephyr/subsys/usb/class/audio/audio.c:682:12: error: 'audio_custom_handler' defined but not used [-Werror=unused-function]
  682 | static int audio_custom_handler(struct usb_setup_packet *pSetup, int32_t *len,
      |            ^~~~~~~~~~~~~~~~~~~~
/home/maciej/zephyrproject2/zephyr/subsys/usb/class/audio/audio.c:331:13: error: 'audio_cb_usb_status' defined but not used [-Werror=unused-function]
  331 | static void audio_cb_usb_status(struct usb_cfg_data *cfg,
      |             ^~~~~~~~~~~~~~~~~~~
/home/maciej/zephyrproject2/zephyr/subsys/usb/class/audio/audio.c:287:13: error: 'audio_interface_config' defined but not used [-Werror=unused-function]
  287 | static void audio_interface_config(struct usb_desc_header *head,
      |             ^~~~~~~~~~~~~~~~~~~~~~
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**
Add any other context about the problem here.

jfischer-no, 2021-02-24:
introduced by de84575e2eace2179c8ecfb33ceb975e7581fcfc

mbolivar-nordic, 2021-02-24:
> introduced by [de84575](https://github.com/zephyrproject-rtos/zephyr/commit/de84575e2eace2179c8ecfb33ceb975e7581fcfc)

This is breaking CI for unrelated PRs; I'm sending a revert

PR #32645 (2021-02-25): Revert "sys: util: Update implementation of loop macros"
This reverts commit de84575e2eace2179c8ecfb33ceb975e7581fcfc.

Its merge broke the USB audio sample, which is failing CI in unrelated
PRs.

Fixes: #32619
PR #32692 (2021-02-28): sys: util: Update implementation of loop macros
Bringing back reverted commit, Commit was  initially introduced by #31904 and then reverted by #32645.
The reason for sample failing was that macro `#define UTIL_EXPAND(...) __VA_ARGS__` was accidentally removed from `util_internal.h` and it was used by usb audio. Missing macro was added back to `util_internal.h`. No other changes compared to initial commit.
 
Fixes #32619.
closed 2021-02-25 by nashif