zephyrproject-rtos/zephyr #32621: 2021 GSoC Project Idea: Arduino module based on Zephyr

beriberikix, 2021-02-24:
Description:

Arduino's popularity is renowned as a popular framework for providing a simplified interface to program embedded devices. Recently, Arduino adopted mbed OS as the base RTOS for some of their newer devices. With that work, they separated out [Arduino Core](https://github.com/arduino/ArduinoCore-API) as an independent abstraction layer from [Arduino Core for mbed](https://github.com/arduino/ArduinoCore-mbed). This opens up the possibility for leveraging Arduino Core on other OSes.

The project idea is to create a Zephyr module that leverages the Arduino Core so that a developer can use Zephyr as the underlying OS when they use the Arduino framework on Arduino-compatible devices. The benefits to the user include:

- Access to Arduino APIs as well as advanced Zephyr capabilities
- Broader set of devices than the standard Arduino ecosystem thanks to Zephyrs' device support
- Ability to re-use Arduino tools like the Arduino IDE and wealth of libraries

Note 1: Arduino Core is licensed under the GNU Lesser General Public License and Zephyr is licensed under Apache 2. That means this project will most likely need to be developed out of tree and in a separate repo to keep code and license separation.

See #22247 for a historic discussion.

Difficulty: Medium
Skill Required: C++, Arduino
Hardware requirement: Arduino-compatible board with enough resources for Zephyr (suggestion: [Arduino Zero](https://store.arduino.cc/usa/arduino-zero))
Possible mentor: Jonathan Beri (jberi@golioth.io)

jocelyn-li, 2021-02-25:
@beriberikix thanks for updating wiki page!

DhruvaG2000, 2023-09-07:
@beriberikix  , @MaureenHelm  OK to close this issue now that we have
https://github.com/zephyrproject-rtos/gsoc-2022-arduino-core ? 😃

closed 2023-09-07 by beriberikix