zephyrproject-rtos/zephyr #32649: Support parsing device tree runtime

MrVan, 2021-02-25:
The current Zephyr driver use compiling time device tree info. To make it flexible, I am thinking the support for runtime parsing device tree for aarch64 which is more powerful. And this will make life easier when porting drivers to Zephyr.

@carlocaione @sandeepbrcm @jharris-intel

carlocaione, 2021-02-25:
having that it would be great and I'm totally for it. But being a huge change for Zephyr I think that needs to be discussed. So I'm tagging a few people that can have something to say about this. @pabigot @nashif @ioannisg @mbolivar-nordic @galak

galak, 2021-02-25:
I don't think this makes sense in general as the reason we generate code/defines from devicetree is size / footprint for the majority of systems that utilize Zephyr.

So having a DTB and the associated code to parse it at runtime is overhead that isn't acceptable for those systems.

As we need to maintain support for a non-runtime devicetree usage, not sure what supporting runtime devicetree would get us?

pabigot, 2021-02-25:
This could be useful for more complex applications.  Once we start supporting devices that are left off on startup, and only turned on by the application, a natural extension is to support that with runtime configuration of available devices ("DT overlays" the way Linux supports them through an API).

Using DTB for describing those devices makes more sense than coming up with a home-grown solution.

carlocaione, 2021-02-25:
> So having a DTB and the associated code to parse it at runtime is overhead that isn't acceptable for those systems.

Agree, but nothing prevents us to have both the solutions in place right? For example having a small DT in-memory parsing library to link only when needed.

galak, 2021-02-25:
> Agree, but nothing prevents us to have both the solutions in place right? For example having a small DT in-memory parsing library to link only when needed.

Agreed, would like to see a proposal as to what that might look like.  What the benefits would be, etc.

carlescufi, 2021-02-25:
> Agree, but nothing prevents us to have both the solutions in place right? For example having a small DT in-memory parsing library to link only when needed.

define "small" :) But yes, I do agree with the ability to do that as long as it's always opt-in. It's a bit like userspace support, which is a feature that makes little sense in the smallest devices but can be very useful on the more powerful ones.

jharris-intel, 2021-02-25:
There's a few components here:

1. Runtime devicetree parsing would add code size. (...and size to store the device tree.)
2. Runtime devicetree parsing would add runtime complexity (read: more things to go wrong during boot / when debugging...)
3. Runtime devicetree parsing would increase time-to-boot (due to the time to parse the device tree, among other things.)
3. Runtime devicetree parsing would decrease runtime performance (somewhat, because info in the device tree can't be const-propagated.)

In exchange, you gain "easy" support for a family of related boards.

Note a few limitations (some surmountable):

1. Runtime devicetree still doesn't help you across processor architectures.
2. Runtime devicetree doesn't help you (much) with optional processor features.
3. Runtime devicetree still requires that you have the drivers compiled in ahead of time.

*****

All told? I understand why e.g. Linux does it; for the usecases I'm interested in, I doubt you'd use runtime devicetree support. It's way overkill for e.g. "hey, I want to change this one feature of this one driver across these two boards", or "hey, I want to only boot up the e.g. I2C driver if this board has something attached to I2C", and comes at a cost.

I don't have an issue with Zephyr adding _optional_ runtime devicetree support, so long as the support is actually optional, and doesn't come at a cost of compile-time devicetree support. (Or, to elaborate on that a little. One "classic" failure mode of such things is the "old" feature rapidly becomes a second-class citizen because many things gratuitously rely on the New Shiny feature.)

(I am mildly concerned about tying post-boot driver configuration to runtime DTS, for instance.)

mbolivar-nordic, 2021-02-25:
I think we all agree this could never be made mandatory even if it gets implemented.

I don't have any uses for runtime DT parsing personally and I would be quite concerned about potential extra complexity if we introduce it as an option. However, I'd be open to seeing how it works if someone has an RFC.

> And this will make life easier when porting drivers to Zephyr.

If they are Linux drivers, you can't copy the code anyway for license reasons. I know some BSDs use DT as well but I'm honestly not convinced that this should be a goal. The zephyr device model is so different that I'm not sure I see how the DT conversion is really the hard part here.