zephyrproject-rtos/zephyr #32655: reel_board: tests/kernel/timer/timer_api/ failure

s-kelvin, 2021-02-25:
**Describe the bug**
Assertion failure at test_sleep_abs in tests/kernel/kernel/timer/timer_api/

**To Reproduce**
Steps to reproduce the behavior:
1. west build -p auto -b reel_board tests/kernel/timer/timer_api/
2. west flash
3. See log

**Expected behavior**
Testsuite should pass

**Impact**
Testsuite failure

**Logs and console output**
```
*** Booting Zephyr OS build zephyr-v2.5.0-441-g5de3f850a752  ***
Running test suite timer_api
===================================================================
START - test_time_conversions
 PASS - test_time_conversions
===================================================================
START - test_timer_duration_period
 PASS - test_timer_duration_period
===================================================================
START - test_timer_restart
 PASS - test_timer_restart
===================================================================
START - test_timer_period_0
 PASS - test_timer_period_0
===================================================================
START - test_timer_expirefn_null
 PASS - test_timer_expirefn_null
===================================================================
START - test_timer_periodicity
 PASS - test_timer_periodicity
===================================================================
START - test_timer_status_get
 PASS - test_timer_status_get
===================================================================
START - test_timer_status_get_anytime
 PASS - test_timer_status_get_anytime
===================================================================
START - test_timer_status_sync
 PASS - test_timer_status_sync
===================================================================
START - test_timer_k_define
 PASS - test_timer_k_define
===================================================================
START - test_timer_user_data
 PASS - test_timer_user_data
===================================================================
START - test_timer_remaining
 PASS - test_timer_remaining
===================================================================
START - test_timeout_abs
 PASS - test_timeout_abs
===================================================================
START - test_sleep_abs

    Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/timer/timer_api/src/main.c:746: test_sleep_abs: (end not equal to start + sleep_ticks)
expected wakeup at 118663, got 118664
 FAIL - test_sleep_abs
===================================================================
Test suite timer_api failed.
===================================================================
PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS: Linux Ubuntu 18.04.4 LTS
 - Toolchain zephyr-sdk-0.12.2
 - Zephyr 2.5 commit ID 5de3f850a752

s-kelvin, 2021-02-25:
This fails on Microchip MEC 15xxEVB ASSY6853 as well, with the same assertion failure.

nashif, 2021-02-26:
duplicate of #32697

closed 2021-02-26 by nashif