zephyrproject-rtos/zephyr #32657: canopen sample wont respond on pdo mapping with CO_ODs from objdict.eds

LonesomeUnixCowboy, 2021-02-25:
**Describe the bug**
I try to run the canopen sample with CO_OD.h and CO_OD.c that i have generated from objdict.eds. But when i run the pdo mapping python script i dont get any pdo from my devkit when i push its button.

In contrast when i run the sample with the provided CO_OD.h and CO_OD.c it works fine and the python script gets the pdo.

Im using Object Dictionary Editor v0.8-0-gb60f4eb to generate the files. My devkit is ST Nucleo F091RC.

Why is CO_OD files generated from objdict.eds not sending any pdo but the "stock" CO_ODs does?

**To Reproduce**
Steps to reproduce the behavior:
1. Generate CO_OD.c and CO_OD.h from objdict.eds
2. Change CO_NO_TS to CO_NO_TIME in CO_OD.h
3. Build and run the canopen sample
4. Run pdo mapping python script form canopen sample webpage

**Expected behavior**
Im expecting the same pdo response when running the sample with CO_ODs generated from objdict.eds as with the "stock" CO_ODs.

**Logs and console output**
I see no difference in the candumps when running the sample with the different CO_ODs except the pdo response with the "stock" CO_ODs.

candump with "stock" CO_ODs
```
  can0  60A   [8]  40 00 18 01 00 00 00 00
  can0  58A   [8]  43 00 18 01 8A 01 00 80
  can0  60A   [8]  40 00 18 02 00 00 00 00
  can0  58A   [8]  4F 00 18 02 FE 00 00 00
  can0  60A   [8]  40 00 18 03 00 00 00 00
  can0  58A   [8]  4B 00 18 03 00 00 00 00
  can0  60A   [8]  40 00 18 05 00 00 00 00
  can0  58A   [8]  4B 00 18 05 00 00 00 00
  can0  60A   [8]  40 00 18 06 00 00 00 00
  can0  58A   [8]  4F 00 18 06 00 00 00 00
  can0  60A   [8]  40 00 1A 00 00 00 00 00
  can0  58A   [8]  4F 00 1A 00 00 00 00 00
  can0  60A   [8]  40 01 18 01 00 00 00 00
  can0  58A   [8]  43 01 18 01 8A 02 00 80
  can0  60A   [8]  40 01 18 02 00 00 00 00
  can0  58A   [8]  4F 01 18 02 FE 00 00 00
  can0  60A   [8]  40 01 18 03 00 00 00 00
  can0  58A   [8]  4B 01 18 03 00 00 00 00
  can0  60A   [8]  40 01 18 05 00 00 00 00
  can0  58A   [8]  4B 01 18 05 00 00 00 00
  can0  60A   [8]  40 01 18 06 00 00 00 00
  can0  58A   [8]  4F 01 18 06 00 00 00 00
  can0  60A   [8]  40 01 1A 00 00 00 00 00
  can0  58A   [8]  4F 01 1A 00 00 00 00 00
  can0  60A   [8]  40 02 18 01 00 00 00 00
  can0  58A   [8]  43 02 18 01 8A 03 00 80
  can0  60A   [8]  40 02 18 02 00 00 00 00
  can0  58A   [8]  4F 02 18 02 FE 00 00 00
  can0  60A   [8]  40 02 18 03 00 00 00 00
  can0  58A   [8]  4B 02 18 03 00 00 00 00
  can0  60A   [8]  40 02 18 05 00 00 00 00
  can0  58A   [8]  4B 02 18 05 00 00 00 00
  can0  60A   [8]  40 02 18 06 00 00 00 00
  can0  58A   [8]  4F 02 18 06 00 00 00 00
  can0  60A   [8]  40 02 1A 00 00 00 00 00
  can0  58A   [8]  4F 02 1A 00 00 00 00 00
  can0  60A   [8]  40 03 18 01 00 00 00 00
  can0  58A   [8]  43 03 18 01 8A 04 00 80
  can0  60A   [8]  40 03 18 02 00 00 00 00
  can0  58A   [8]  4F 03 18 02 FE 00 00 00
  can0  60A   [8]  40 03 18 03 00 00 00 00
  can0  58A   [8]  4B 03 18 03 00 00 00 00
  can0  60A   [8]  40 03 18 05 00 00 00 00
  can0  58A   [8]  4B 03 18 05 00 00 00 00
  can0  60A   [8]  40 03 18 06 00 00 00 00
  can0  58A   [8]  4F 03 18 06 00 00 00 00
[... 198 lines cut ...]
  can0  60A   [8]  2F 03 1A 00 00 00 00 00
  can0  58A   [8]  60 03 1A 00 00 00 00 00
  can0  60A   [8]  2F 03 1A 00 00 00 00 00
  can0  58A   [8]  60 03 1A 00 00 00 00 00
  can0  000   [2]  01 0A
  can0  60A   [8]  23 02 21 00 00 00 00 00
  can0  18A   [4]  00 00 00 00
  can0  58A   [8]  60 02 21 00 00 00 00 00
  can0  70A   [1]  05
  can0  70A   [1]  05
  can0  70A   [1]  05
  can0  70A   [1]  05
  can0  70A   [1]  05
  can0  70A   [1]  05
  can0  70A   [1]  05
  can0  70A   [1]  05
  can0  70A   [1]  05
  can0  70A   [1]  05
  can0  70A   [1]  05
  can0  70A   [1]  05
```
**Environment (please complete the following information):**
 - OS: Ubuntu
 - Toolchain: zephyr-sdk-0.12.2
 - Commit SHA or Version used: 89a90963

henrikbrixandersen, 2021-02-26:
@LonesomeUnixCowboy I have take the liberty to change this from bug to question since, as you also noted, the sample works with the code shipped in Zephyr.

Normall you would not use the EDS file for generating the OD as the EDS doesn't contain the full meta data for the OD. Try using the `objdict.xml` file instead. Beware that you will need a revision of https://github.com/robincornelius/libedssharp which is compatible with https://github.com/CANopenNode/CANopenNode/tree/v1.3-master (latest stable CANopenNode, v1.3, the version integrated in the Zephyr Project) for generating the OD implementation files.

Without any indication of what is changed in your custom generated CO_OD.{ch} files, I'm afraid I cannot assist in debugging your issue any further.

LonesomeUnixCowboy, 2021-03-02:
@henrikbrixandersen Your are absolutly right, this is a question and not a bug. I choosed the wrong category and I apologize for this.

Ok, I generated the CO_ODs from the ```objdict.xml``` and now the sample works (it responds to the pdo mapping python script).

I did no change to the ```objdict.eds``` before I generated the CO_ODs. Initially my plan was to copy the TPDO mapping  from the script to the OD (by editing ```objdict.eds```) just to test how/if it worked (also I deleted the mapping in the script). And when it didn't, I restored the  ```objdict.eds``` to its original state and generated the CO_ODs again (from ```objdict.eds```) the to see if that would make the sample work, but still it would not respond.

But now when I generate CO_ODs and ```objdict.eds``` from the ```objdict.xml``` as suggested and the sample responds as expected. I have even made my own custom edits to the ```objdict.xml``` and my devkit responds to my PDOs. I believed that there was no diffrence between the ```objdict.eds``` and ```objdict.xml``` so I edited the .eds. I have to stop believing.

Regarding https://github.com/robincornelius/libedssharp I have used the pre compiled release because I couldn't figure out how build the editor on my machine. It seams to edit/generate ODs fine except that I have to change  ```CO_NO_TS``` to ```CO_NO_TIME``` in CO_OD.h. This is something I can to live with until some kind soul can build the correct version of the editor to use with Zephyr.

Thank you for the help @henrikbrixandersen.

closed 2021-03-02 by LonesomeUnixCowboy

henrikbrixandersen, 2021-03-02:
@LonesomeUnixCowboy Glad you got it working!