zephyrproject-rtos/zephyr #32665: Bluetooth: controller: inclusion of vendor data type and function overrides provided by vendor LLL

cvinayak, 2021-02-25:
**Is your enhancement proposal related to a problem? Please describe.**
Relates to including vendor specific data structure and function implementation overrides in the controller code.
Refer to the discussion here: https://github.com/zephyrproject-rtos/zephyr/pull/32613#discussion_r582450306

**Describe the solution you'd like**
A discussion on alternative schemes to include type overrides, function interface overrides and the folder structure to discuss these is needed.

**Describe alternatives you've considered**
NA

**Additional context**
NA

cfriedt, 2021-07-12:
Marking as low-priority, but feel free to escalate based on your triage @cvinayak

kruithofa, 2022-08-25:
the issue is at best a marginal improvement, and not worth our time doing

closed 2022-08-25 by kruithofa