zephyrproject-rtos/zephyr #32669: [Bluetooth] sample code for Periodic Advertising Sync Transfer

kaiser-ren, 2021-02-25:
**Is your feature request related to a problem? Please describe.**
No, this is NOT related to a problem. 

**Describe the solution you'd like**
Periodic Adv Sync Transfer(PAST) is an important feature especially LE Audio, current Zephyr Bluetooth stack supports PAST(it may need the support from Bluetooth controller), there could be a sample code to illustrate this feature. 
There are several modes of PAST, we can select the typical one, three devices involved: PA, PA sender and PA receiver to do it. 

**Describe alternatives you've considered**
No. 

**Additional context**
None.

Thalley, 2021-02-25:
@kaiser-ren Thanks for raising this. The immediate challenge is that the Zephyr Controller does not support PAST, so I'm unsure how feasible it is to create such a sample at this point, as it will not run with a pure Zephyr build. We can add a HCI host only sample, but I'm not sure that's very typical in Zyphyr. @carlescufi Any comment to that? 

@cvinayak Do we have any timeline for the PAST feature in the Zephyr controller? 

What we could do, for now, is to extend the BT shell to have PAST support, if you need some kind of reference on how to use it.

carlescufi, 2021-02-25:
> We can add a HCI host only sample, but I'm not sure that's very typical in Zyphyr. @carlescufi Any comment to that?

I see no reason not to work on the sample before the controller has support for it. This was already the case for Extended Advertising for example. Additionally, during the development of the controller support for it the sample will be useful.

Thalley, 2021-02-25:
> I see no reason not to work on the sample before the controller has support for it. This was already the case for Extended Advertising for example. Additionally, during the development of the controller support for it the sample will be useful.

Alright, I'll see if I can add a sample soon then :)

kaiser-ren, 2021-02-26:
> @kaiser-ren Thanks for raising this. The immediate challenge is that the Zephyr Controller does not support PAST, so I'm unsure how feasible it is to create such a sample at this point, as it will not run with a pure Zephyr build. We can add a HCI host only sample, but I'm not sure that's very typical in Zephyr. @carlescufi Any comment to that?
> 
> @cvinayak Do we have any timeline for the PAST feature in the Zephyr controller?
> 
> What we could do, for now, is to extend the BT shell to have PAST support, if you need some kind of reference on how to use it.

@Thalley, I'm curious whether nRF52840 support PAST or not.

Thalley, 2021-02-26:
@kaiser-ren Once implemented in the controller, I don't think there's any reason why it shouldn't :)

closed 2021-07-01 by nashif