zephyrproject-rtos/zephyr #32677: z_user_string_nlen() might lead to non-recoverable errors, despite suggesting the opposite

ioannisg, 2021-02-25:
**Describe the bug**

Kernel recoverable errors are not always recoverable

**Context**

The Zephyr Kernel has support for recovering from otherwise fatal errors. One example is z_user_string_nlen()

```
/**
 * @brief Obtain the size of a C string passed from user mode
 *
 * Given a C string pointer and a maximum size, obtain the true
 * size of the string (not including the trailing NULL byte) just as
 * if calling strnlen() on it, with the same semantics of strnlen() with
 * respect to the return value and the maxlen parameter.
 *
 * Any memory protection faults triggered by the examination of the string
 * will be safely handled and an error code returned.
 *
 * NOTE: Doesn't guarantee that user mode has actual access to this
 * string, you will need to still do a Z_SYSCALL_MEMORY_READ()
 * with the obtained size value to guarantee this.
 *
 * @param src String to measure size of
 * @param maxlen Maximum number of characters to examine
 * @param err Pointer to int, filled in with -1 on memory error, 0 on
 *	success
 * @return undefined on error, or strlen(src) if that is less than maxlen, or
 *	maxlen if there were no NULL terminating characters within the
 *	first maxlen bytes.
 */
static inline size_t z_user_string_nlen(const char *src, size_t maxlen,
					int *err)

```

The current implementation for Cortex-M allows the kernel to de-reference whatever pointer is passed by user code in *src.
The function "promises" that any exception will be safely handled by Zephyr, however, this does not seem to be the case when we are building Zephyr running as Non-Secure, in TrustZone enabled Cortex-M builds. In this case, exceptions are triggering SecureFaults, which are not handled by the Zephyr kernel (are handled externally).

I would like to discuss how we can best deal with this case.

Of course, any access to memory that is not available in Non-Secure domain leads to SecureFaults, however, this function's documentation is certainly confusing (i.e. promising that any memory protection related errors are gracefully handled).

ioannisg, 2021-02-25:
FYI @pabigot @andyross @ceolin

jharris-intel, 2021-02-26:
This function inherently has a TOCTTOU vulnerability... so let's assume for a moment that the resulting memory isn't updated while we're accessing it.

Assuming that's the case, one approach is to use `AT S1E0R` before each read to check if the address is accessible (if it's not, `PAR_EL1` bit 0 will be set).

andyross, 2021-02-27:
To clarify: the bug here is that the user can pass an address to a syscall that causes the kernel to blow up with a SecureFault in a way that the userspace app wouldn't have been able to do itself?   That sounds like the whole mechanism has to be reworked on TrustZone platforms.   Or do you mean this is something the app could do itself, and the only problem is the docs?

As far as being a vulnerability inherently, we'll need to dig some.  The contents of the string aren't validated yet, this routine exists to copy a string out of userspace for a syscall and thus all the validation is still downstream.  And I **think** that inside a syscall the memory domain is locked and can't be changed -- it's not possible to get the kernel to fault on the subsequent memcpy() out of the region.  Maybe @dcpleung knows for sure?  This code is for sure impacted by inbound work on virtual address spaces too, about which I know almost nothing still.

dcpleung, 2021-02-27:
I think the originally assumption is that any illegal memory access would be handled by the Zephyr kernel.

Since this behavior depends on `memory_fault_recoverable()` to execute `z_arm_user_string_nlen_fixup` so `z_user_string_nlen()` can return with error, maybe this can be simulated within `arch_user_string_nlen` as @jharris-intel said? Then update the doc saying that any illegal memory access must return error instead of causing un-recoverable faults?

ioannisg, 2021-03-01:
@andyross, @dcpleung  thanks for looking at it.

> To clarify: the bug here is that the user can pass an address to a syscall that causes the kernel to blow up with a SecureFault in a way that the userspace app wouldn't have been able to do itself?

Well, yes.

> That sounds like the whole mechanism has to be reworked on TrustZone platforms.

I am not sure I agree. A user thread can, anyways, trigger a SecureFault by simply accessing memory that it is not supposed to (memory that is part of the secure execution environment). The kernel could do nothing about it. Note that other CPU fatal errors could lead to an oops() and cause user thread abort.

> Or do you mean this is something the app could do itself, and the only problem is the docs?

Yes, the app can do it itself.

What I am arguing about is that  _perhaps_ the kernel itself should never trigger such faults _directly_, i.e. by executing kernel functionality. So, then, we could honor the commitment that kernel recoverable errors are indeed recoverable. Perhaps, we could do some parameter checking in the kernel, so we do not de-reference memory that is not part of the image. 

Other than that, sure, the docs are problematic, in my opinion, since a NULL pointer passed does not lead to an exception that is recoverable.

pabigot, 2021-03-01:
> Other than that, sure, the docs are problematic, in my opinion, since a NULL pointer passed does not lead to an exception that is recoverable.

The documentation certainly has problems, but I don't believe the above is *necessarily* the underlying problem.

```
 * Any memory protection faults triggered by the examination of the string
 * will be safely handled and an error code returned.
...
 * @return undefined on error, or ...
```

The return type could be `ssize_t` so errors can be identified rather than ignored.

But If it's the exception from this that is supposed to be recoverable and so is why we aren't explicitly checking for null pointers before deferencing them in `z_vrfy_` wrappers (see history in #32367), it'd be pretty easy to just do the test explicitly in the wrappers.

andyross, 2021-03-01:
This is starting to sound bikesheddy. :)  If a userspace app is capable of generating the same TrustZone fault on some region of memory, and the only problem is that the kernel is generating a TrustZone fault **at the direction of the userspace app** by referencing the same bad address... do we really care?  I mean, was it the "kernel" that faulted just because it's validating a system call, or was this an application fault?

FWIW: The language @pabigot quoted clearly says "memory protection faults", and TrustZone is (arguably) a separate layer of protection and not part of the kernel's "memory protection" feature.

Finally: is maybe the problem with the mechanism?  I mean, copying a string out of a syscall is intended for tiny data.  I'm guessing this is done this way for performance, but have we benchmarked it vs. a simpler z_user_string_copy() which worked by manually walking the memory domain/partition info instead of relying on the CPU?  For large data, sure, I can see that it's faster to get the hardware to do the walking, but for the 1-2 pages that a given string argument will hit?  It seems dubious.

Note that the simpler version would be not only safer and easier to document, but portable too.

(And that's as much shed paint as you'll see from me, I promise.)

jharris-intel, 2021-03-01:
> I'm guessing this is done this way for performance, but have we benchmarked it vs. a simpler z_user_string_copy() which worked by manually walking the memory domain/partition info instead of relying on the CPU

BTW, I've tried this before; parsing the translation tables yourself ends up being much slower than just asking the CPU to do it for you. The core has a fair bit of hardware acceleration of translation table walking, and likely already has the relevant entries in the TLB.

It also prevents parsing vulnerabilities. Exploiting differences in two different parses of the same data is a classic attack.

> If a userspace app is capable of generating the same TrustZone fault on some region of memory, and the only problem is that the kernel is generating a TrustZone fault at the direction of the userspace app by referencing the same bad address... do we really care?

Doesn't the former (only) kill the thread, whereas the latter stops the world? Or does Zephyr know enough to be able to know (and recover from - e.g. if the actual access was inside a spinlock...) that the kernel assert was on behalf of user thread X?

closed 2021-05-15 by github-actions[bot]