zephyrproject-rtos/zephyr #32691: Function z_find_first_thread_to_unpend() missing coverage in sched.c

yingmingx, 2021-02-26:
**Describe the bug**
Function z_find_first_thread_to_unpend() missing coverage in sched.c. 

**To Reproduce**
Steps to reproduce the behavior:
1. twister -p qemu_x86 --coverage -T tests/kernel

**Expected behavior**
It's probably dead code that should be removed.

Environment (please complete the following information):

OS: ubuntu1~18.04
Toolchain zephyr-sdk-0.12.1
Commit id 5de3f850a752ec638b695c151d3de8f19503907f

andyross, 2021-03-01:
Agreed.  This looks to have been orphaned fairly recently; there was a scheduler race fix a while back (that exposed another issue, and begat the big abort rewrite and bug hunt of the past two weeks) that addressed synchronization issues by combining operations in a few spots.  This looks like a "half" of the operation that got combined.

yerabolu, 2021-03-03:
Working on the changes and will submit a PR soon.

PR #32891 (2021-03-05): kernel: sched: removing dead code
Due to the recent changes to scheduler z_find_first_thread_to_unpend
& z_remove_thread_from_ready_q are not used anymore. So removing the
dead code.

fixes: #32691
commit 4118ed1d (2021-03-05): kernel: sched: removing dead code

Due to the recent changes to scheduler z_find_first_thread_to_unpend
& z_remove_thread_from_ready_q are not used anymore. So removing the
dead code.

fixes: #32691
closed 2021-03-05 by ioannisg