zephyrproject-rtos/zephyr #32695: intel_adsp_cavs15: cannot get some testcases output

KangJianX, 2021-02-26:
**Describe the bug**
When run testcases of kernel on intel_adsp_cavs15, tests/kernel/fifo/, tests/kernel/lifo, tests/kernel/mem_heap/, tests/kernel/mbox/, tests/kernel/msgq/, tests/kernel/mem_protect/ has **no output**

**To Reproduce**
Steps to reproduce the behavior:
1. source zephyr-env.sh
2. twister -p intel_adsp_cavs15 --device-testing -T tests/kernel/ --west-flash="/home/ztest/work/zephyrproject/zephyr/boards/xtensa/intel_adsp_cavs15/tools/flash.sh,/home/ztest/work/zephyrproject/modules/audio/sof/keys/otc_private_key.pem,/home/ztest/work/zephyrproject/modules/audio/sof/zephyr/ext/rimage/config,/home/ztest/work/zephyrproject/modules/audio/sof/zephyr/ext/rimage/build/rimage" --device-serial-pty="/home/ztest/work/zephyrproject/zephyr/boards/xtensa/intel_adsp_cavs15/tools/adsplog.py,--no-history"
3. See error

**Logs and console output**
NA

**Environment (please complete the following information):**
 - OS: Fedora28
 - Toolchain: Zephyr-sdk-0.12.3
 - Commit ID: 63972d6df7121f51

KangJianX, 2021-02-26:
This issue caused by commit ID: 6b84ab38.

andyross, 2021-03-02:
This isn't happening in my tree.  Should be fixed once #32356 merges.

KangJianX, 2021-03-02:
#32356 can fix this problem.

LeiW000, 2021-03-08:
@KangJianX , my understanding is that you have verified #32356. Am I right?

KangJianX, 2021-03-08:
@LeiW000 Yes, I have verified and it work.

andyross, 2021-03-08:
Fix merged.  Close.

closed 2021-03-08 by andyross