zephyrproject-rtos/zephyr #32696: intel_adsp_cavs15: run testcases failed of tests/kernel/common

KangJianX, 2021-02-26:
**Describe the bug**
run tests/kernel/common failed

**To Reproduce**
Steps to reproduce the behavior:
1. source zephyr-env.sh
2. twister -p intel_adsp_cavs15 --device-testing -T tests/kernel/common/ --west-flash="/home/ztest/work/zephyrproject/zephyr/boards/xtensa/intel_adsp_cavs15/tools/flash.sh,/home/ztest/work/zephyrproject/modules/audio/sof/keys/otc_private_key.pem,/home/ztest/work/zephyrproject/modules/audio/sof/zephyr/ext/rimage/config,/home/ztest/work/zephyrproject/modules/audio/sof/zephyr/ext/rimage/build/rimage" --device-serial-pty="/home/ztest/work/zephyrproject/zephyr/boards/xtensa/intel_adsp_cavs15/tools/adsplog.py,--no-history"
3. See error

**Logs and console output**
............................................................................
[ADSP Device Reset]
............................................................................
[00000088] <err> os:  ** FATAL EXCEPTION
[00000088] <err> os:  ** CPU 0 EXCCAUSE 28 (load prohibited)
[00000088] <err> os:  **  PC 0xbe0096b5 VADDR 0x20000000
[00000088] <err> os:  **  PS 0x60020
[00000088] <err> os:  **    (INTLEVEL:0 EXCM: 0 UM:1 RING:0 WOE:1 OWB:0 CALLINC:2)
[00000088] <err> os:  **  A0 0xbe00f370  SP 0xbe07fe40  A2 0x20000000  A3 0x2f80749
[00000088] <err> os:  **  A4 (nil)  A5 0xffffff80  A6 0x9e020268  A7 (nil)
[00000088] <err> os:  **  A8 0x1  A9 0xbe07fe10 A10 0x9e01ff28 A11 0x9e01ff28
[00000088] <err> os:  ** A12 0x1f A13 0xbe013838 A14 0x9e020268 A15 (nil)
[00000088] <err> os:  ** LBEG 0xbe0096b5 LEND 0xbe0096bb LCOUNT 0x2780748
[00000088] <err> os:  ** SAR 0x4
[00000088] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00000088] <err> os: Current thread: 0x9e01ff28 (main)
[00000091] <err> os: Halting system

**Environment (please complete the following information):**
OS: Fedora28
Toolchain: Zephyr-sdk-0.12.3
Commit ID: 63972d6

KangJianX, 2021-02-26:
This issue caused by commit ID: 6b84ab3.

andyross, 2021-03-02:
This isn't happening in my tree.  Should be fixed once #32356 merges.

KangJianX, 2021-03-02:
#32356 the patch can fix this issue

KangJianX, 2021-03-12:
This issue has been fixed, close.

closed 2021-03-12 by KangJianX