zephyrproject-rtos/zephyr #32697: sam_e70b_xplained: running tests/kernel/timer/timer_api failed

Zhaoningx, 2021-02-26:
Describe the bug
Assertion failure at test_timeout_abs in tests/kernel/kernel/timer/timer_api/

To Reproduce
Steps to reproduce the behavior:

twister -p sam_e70b_xplained --device-testing --device-serial /dev/ttyACM1 -T tests/kernel/timer/timer_api/  -j 1 -vv

See log
Expected behavior
Testsuite should pass

Impact
Testsuite failure

Logs and console output

```
*** Booting Zephyr OS build zephyr-v2.5.0-487-g0ba94379381f  ***
Running test suite timer_api
===================================================================
START - test_time_conversions
PASS - test_time_conversions
===================================================================
START - test_timer_duration_period
PASS - test_timer_duration_period
===================================================================
START - test_timer_restart
PASS - test_timer_restart
===================================================================
START - test_timer_period_0
PASS - test_timer_period_0
===================================================================
START - test_timer_expirefn_null
PASS - test_timer_expirefn_null
===================================================================
START - test_timer_periodicity
PASS - test_timer_periodicity
===================================================================
START - test_timer_status_get
PASS - test_timer_status_get
===================================================================
START - test_timer_status_get_anytime
PASS - test_timer_status_get_anytime
===================================================================
START - test_timer_status_sync
PASS - test_timer_status_sync
===================================================================
START - test_timer_k_define
PASS - test_timer_k_define
===================================================================
START - test_timer_user_data
PASS - test_timer_user_data
===================================================================
START - test_timer_remaining
PASS - test_timer_remaining
===================================================================
START - test_timeout_abs
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/timer/timer_api/src/main.c:725: test_timeout_abs: (cap_ticks + rem_ticks + 1 == exp_ticks) || (rem_ticks + cap2_ticks + 1 == exp_ticks) || (INEXACT_MS_CONVERT && (cap_ticks + rem_ticks == exp_ticks)) || (INEXACT_MS_CONVERT && (rem_ticks + cap2_ticks == exp_ticks)) is false
FAIL - test_timeout_abs
===================================================================
START - test_sleep_abs
PASS - test_sleep_abs
===================================================================
Test suite timer_api failed.
===================================================================
PROJECT EXECUTION FAILED
```
Environment (please complete the following information):

OS: Fedora33
Toolchain: zephyr-sdk-0.12.2
Zephyr 2.5 commit ID 0ba94379381f

Zhaoningx, 2021-02-26:
Additional: this bug is also happened at **iotdk** and **frdm_k64f** platform, the error log is same as the description above.

nashif, 2021-02-26:
duplicate of #32572

closed 2021-02-26 by nashif