zephyrproject-rtos/zephyr #32702: RAM overflow with bbc:microbit for samples/bluetooth/peripheral...

Liviu112, 2021-02-26:
**Describe the bug**
Trying to build the peripheral samples the following errors occur: 
1. west build -p auto -b bbc_microbit samples/bluetooth/peripheral_hids
``` `SRAM' overflowed by 5627 bytes ```
2. west build -p auto -b bbc_microbit samples/bluetooth/peripheral_hr
``` `SRAM' overflowed by 2971 bytes ```
3. west build -p auto -b bbc_microbit samples/bluetooth/peripheral_ht
``` `SRAM' overflowed by 2971 bytes ```
4. west build -p auto -b bbc_microbit samples/bluetooth/peripheral_ots
``` `SRAM' overflowed by 4771 bytes ```
5. west build -p auto -b bbc_microbit samples/bluetooth/peripheral_sc_only
```  `SRAM' overflowed by 2803 bytes ```
6. west build -p auto -b bbc_microbit samples/bluetooth/peripheral_esp
```  `SRAM' overflowed by 1587 bytes  ```
7. west build -p auto -b bbc_microbit samples/bluetooth/peripheral_dis
```  `SRAM' overflowed by 593 bytes ```  
8. west build -p auto -b bbc_microbit samples/bluetooth/peripheral
```  `SRAM' overflowed by 6592 bytes ```  

**To Reproduce**
Steps to reproduce the behaviour:
1. zephyr 2.5
2. toolchain: zephyr 0.12.2
3. use any of the codes shown previously (1-8)

**Impact**
Not able to proceed on using the samples. 

**Environment (please complete the following information):**
 - OS: Linux 20.04
 - Toolchain: Zephyr version: 2.5.0, west 0.9

jhedberg, 2021-02-26:
This was already fixed by #32562

jhedberg, 2021-02-26:
Oh, sorry, you're referring to the peripheral samples whereas #32562 is just for mesh. This isn't strictly a bug per-se, in my opinion. It's unavoidable that many things will not fit into 16k of RAM. That said, we should probably make sure that at least one of the peripheral samples fits. @joerchan any opinion on this?

galak, 2021-03-16:
We don't consider these overflow conditions as a bug.

closed 2021-03-16 by galak

joerchan, 2021-03-17:
@jhe I think we have to let bbc_microbit and connections go, if we wanted to have one working it should have been peripheral_hr IMO, and that is almost 3k bytes we need.