zephyrproject-rtos/zephyr #32708: AARch64 SMP k_current_get performance improvements

jharris-intel, 2021-02-26:
**Is your enhancement proposal related to a problem? Please describe.**
Even after changes described in #32686, a best-case uncontended `k_yield` still takes ~826 cpu cycles. 49.8% of this time (!) is spent inside `k_current_get` or child calls thereof.

`k_current_get` is conceptually just a pointer lookup (return the address of the current thread). The current implementation is essentially:

1. (non-inlined) call to `k_current_get` (so the caller has to assume that this may trample a lot of registers).
2. (non-inlined) branch directly to `z_impl_k_current_get` (unnecessary and may cause e.g. an icache miss, but not the end of the world).
3. Lock out interrupts (inlined call to `arch_irq_lock`)
4. Grab the current core ID from `mpidr_el1`. (inlined call to `arch_curr_cpu`)
5. Calculate the base address of `_kernel.cpus`. (inlined call to `arch_curr_cpu`)
6. Do a multiply-add to find the address of `_kernel.cpus[core & 0xFF]`. (inlined call to `arch_curr_cpu`)
7. Load from offset 16 of the resulting pointer to calculate `_current_cpu->current`.
8. (non-inlined) call to `arch_irq_unlock`.

All told this takes ~28 cycles. (But there are 15 calls per `k_yield`, scattered through the callstack, which adds up very very quickly! Or about 417 cycles out of the 826 are just in calls to `k_current_get`.)

Note that the above numbers do not include second-order effects (such as surrounding code having worse register allocation due to having to do a call).

**Describe the solution you'd like**

Perhaps directly store `_current` in `tpidr_el1`, and `&_kernel.cpus[core]` in `tpidrro_el0`, or vice versa? This would reduce this from ~28 cycles to ~1, at the expense of having to update `tpidr_el1` when you'd currently update `_current`.

A local proof-of-concept dropped uncontended `k_yield` from 826 cycles to 252 cycles (!).

(Function calls on AArch64 are surprisingly expensive. Most of the additional cycles saved there are just from avoiding stacking / juggling registers around / doing redundant work across function calls, so far as I can tell.)

After this optimization, what remains is:

70 cycles in z_impl_k_yield itself
69 cycles in z_swap_next_thread
50.5 cycles in z_swap_unlocked
14 cycles in z_is_t1_higher_prio_than_t2
11=5.5x2 cycles in should_queue_thread
9.5 cycles in sys_dlist_peek_head
5 cycles in update_cache
5=2.5x2 cycles in z_is_thread_state_set.constprop.0.isra.0
4.5 cycles in sys_dlist_is_empty
4.5 cycles in z_is_idle_thread_object.isra.0
3.5 cycles in z_is_thread_prevented_from_running.isra.0
2.5 cycles in is_aborting.isra.0
2=1x2 cycles in z_is_thread_queued.isra.0
1 cycle in k_yield

...and there are still some low-hanging fruit there (e.g. I included the long-tail function calls there because several of them really should be inlined for performance - e.g. `sys_dlist_is_empty`). (Ditto, there's two sets of spinlock lock/unlock per loop still - one in `z_impl_k_yield`, one in `z_swap_unlocked`.)

**Describe alternatives you've considered**

I looked at reducing calls to `_current_cpu` / `_current` - this is non-trivial, as the calls are largely scattered around and can't easily be combined.

I looked at storing `_current` in some manner that doesn't require an interrupt mask/unmask - this is again nontrivial. The best I could do was storing `(sequence number, _current)` atomicly with a retry loop, and it wasn't any faster than the current interrupt masking approach.

**Additional context**

Ties into #32686, #32187, https://github.com/zephyrproject-rtos/zephyr/pull/32557#discussion_r583292495.

ARM (both AArch32 and AArch64 - although I don't know what ARM version introduced this) has between three and five `tpidr*` registers, intended for storing 64 bits of thread-local data (thread ID, pointer, etc):

1. `tpidr_el0`. Normally accessible (read+write) from all ELs.
2. `tpidrro_el0`. Normally readable from all ELs and writable from EL1 and above.
3. `tpidr_el1`. Normally accessible (read+write) from EL1 and above.
4. `tpidr_el2`, if you have EL2. Normally accessible (read+write) from EL2 and above.
5. `tpidr_el3`, if you have EL3. Normally accessible (read+write) from EL3 only.

Currently AArch64 uses `tpidr_el0` to store the TLS pointer, and otherwise does not use these registers. We can't rely on 4 or 5, but we can rely on 1-3 being available.

This proposal would have the following assignments:

1. `tpidr_el0` is still the TLS pointer.
2. `tpidrro_el0` stores `&_kernel.cpus[core]`.
3. `tpidr_el1` stores `_current`.

(Or swap 2/3. It doesn't matter too much.)

jharris-intel, 2021-02-26:
As an aside, take the cycle counts with a bit of a grain of salt. The total cycle count is accurate, but stalls can be delayed sometimes.

So e.g. function foo calls bar, function bar triggers a division and returns, foo tries to use the result of said division, the cycle count will sometimes show foo taking most of the cycles.

jharris-intel, 2021-02-26:
As an aside, as per usual I found an issue immediately after creating this... `-fno-inline` snuck into my local branch, which explains some of the inlining failures (although note that `ALWAYS_INLINE` overrides this).

andyross, 2021-02-26:
That sounds perfect to me, though again I try to keep my nose out of ARM work.

It's worth noting that _current is just a field on struct _cpu, so as long as you have the first it's only an extra load from L1 away to get _current (that field is hit from everywhere, so you can assume it's cached if anything is).  Dedicating the extra register might actually hurt instead of help, depending on how the pipeline handles the special registers.  I know (or vaguely remember, anyway) that on x86 a lot of MSRs are actually slower than a read from L1 because of the way they're scheduled.

Note also that since this is an MMU-capable device, another option would be to keep a per-CPU page mapped somewhere (at the cost of a little extra work at context switch time) and read it from there, something like the way the per-thread VDSO works in linux.

jharris-intel, 2021-02-26:
@andyross - 

> as long as you have the first it's only an extra load from L1 away to get _current

Good point, but somewhat misleading.

The issue is, `_current` is a field in `_current_cpu` (logically speaking). The expensive part isn't reading `_current`. It's making sure that there isn't a race condition where you (as in thread `T`) read the current cpu, then you (as in thread `T`) get swapped to a different cpu, then you (as in thread `T`) read the old cpu's `_current`.

Right now the way that's avoided is via an irq lock / unlock, which isn't super expensive... until you want to do it 10x or more per context switch. And we currently do so 15x per context switch (at least on my local branch).

Directly reading _current from somewhere means that there is no race condition there because you're not going through `_current_cpu`, so you don't need the irq lock / unlock.

> Dedicating the extra register might actually hurt instead of help, depending on how the pipeline handles the special registers

The Cortex-A53 (which is what I know) can't dual-issue a MRS/MSR, but otherwise is fairly good about reading system registers ("All system instructions are fully pipelined and most are single cycle in execution. System instructions that change processor configuration such as the SCTLR have 4 NOP cycles inserted behind to prevent a following instruction inheriting the old configuration settings."); this is definitely a concern for other A-series processors.

> Note also that since this is an MMU-capable device, another option would be to keep a per-CPU page mapped somewhere (at the cost of a little extra work at context switch time) and read it from there, something like the way the per-thread VDSO works in linux.

You don't even need extra work at context switch, although it requires a memory map set up for it. Use TTBR1 for the per-cpu page and TTBR0 for everything else (or TTBR1 for the per-cpu page and all kernel data, and TTBR0 for everything else. Either works...). Set up all CPUs with the same TTBR0 and different TTBR1s.

jharris-intel, 2021-02-26:
@carlocaione - you may be interested in this (you seem to be interested in AArch64 improvements)

jharris-intel, 2021-02-26:
@andyross -

> I try to keep my nose out of ARM work.

Fair enough. I had tagged you as "anything to do with context switching", and was mainly tagging you on these as you likely have a much better intuition for that side of things than I do. (E.g. if I'm spending a bunch of time trying to optimize something that isn't necessary in the first place.) On that note, a few questions:

1. Would it be possible/practical to avoid the multiple spinlock locks in context switching? Right now `k_yield` does lock/unlock/lock/unlock, which a) makes up a nontrivial portion of total context switch time, b) makes sched_lock even more contended, and c) in contended situations often results in additional latency when threads end up blocking on the second spin lock call.
2. Is it expected that there would be ~15 k_current_get calls in a context switch? I don't know if this is a case where AArch64's SMP implementation is slower than other architectures, my expectations as to how long a context switch should take on bare metal are skewed, etc, etc.

andyross, 2021-02-26:
So, for 1: yes, in many/most cases that are a performance path (so... probably not k_yield() specifically, but in lots of places) we can do some work to roll locks together where it makes sense.  There is likely some code size and maintainability cost, but sure.  Remember though that spinlocks are intended to be super-fast in the uncontended case.  There should be 1-3 or so instructions issued in order (and a serialization cost on big CPUs) and you should exit the loop with a successful branch predict.  Even our naive implementation based on the atomics layer seems to get this mostly right on x86 and xtensa, at least (though x86 has a lot of newer toys we don't touch -- we're just a loop around LOCK CMPXCHG).

As for 2: I'm honestly surprised it's not more.  EVERYTHING is going to have something to say about the current thread.  Is maybe something in the ARM64 implementation defeating optimization?  What I would hope to see in the generated code is that the compiler would call it once and leave it in a register.  But if it's a function call without a __pure attribute or a volatile assembly block being inlined, then that won't work.

andyross, 2021-02-26:
Hm... actually looking at it it doesn't work like I thought it did, in kernel_struct.h we have this when CONFIG_SMP:

    #define _current k_current_get()

when what I assumed we had was something more like:

    #define _current (arch_curr_cpu()->current)

Honestly that's just a bug.  It's going to force a function call, which is going to have to do the special register load every single time.  All for something we want in a register.

andyross, 2021-02-26:
And thinking more carefully, it's actually wrong to make the evaluation of _current_cpu pure, I think.  Obviously threads can migrate!  We actually enforce right now that the macro is only ever expanded when interrupts are masked (for obvious correctness reasons).  But consider code like:

 ```c
k = k_spin_lock(&lock);
printk("My CPU is %d\n", _current_cpu->id);
k_spin_unlock(&lock, k);

/* get preempted start running on a different CPU */

k = k_spin_lock(&lock);
printk("My CPU is %d\n", _current_cpu->id);
k_spin_unlock(&lock, k);
```

A compiler that tried to optimize the pure result of _current_cpu would print the same CPU ID twice, which is clearly wrong.  The spinlock calls have a memory clobber, but I don't think that's sufficient to defeat the optimization.

(Or maybe it is?  This gets to the difference between a "pure" and a "const" function that I always need to look up...)

andyross, 2021-02-26:
So I think what we really want is a definition for current more like:

```c
static ALWAYS_INLINE __attribute__(("pure")) __eval_current()
{
	return _current_cpu->current;
}

#define _current __eval_current()
```

But then we need to worry about toolchain portability and spend time staring at the generated assembly to be sure this is doing the right thing.

jharris-intel, 2021-02-26:
With inlining, after optimizations, it's 211 cycles / `k_yield`:

```
114 cycles in z_impl_k_yield
81 cycles in z_swap_next_thread
16 cycles in z_is_t1_higher_prio_than_t2
```

Or, expanding that out a little, for `z_impl_k_yield`

```asm
1.50   z_impl_k_yield:         stp	x29, lr, [sp,#-48]!
1.50   z_impl_k_yield+0x4:     mov	x29, sp
1.50   z_impl_k_yield+0x8:     stp	x19, x20, [sp,#16]
1.50   z_impl_k_yield+0xc:     stp	x21, x22, [sp,#32]
1.50   z_impl_k_yield+0x10:    mrs	x0, tpidr_el1
1.50   z_impl_k_yield+0x14:    ldrb	w0, [x0,#36]
1.50   z_impl_k_yield+0x18:    adrp	x6, .+81920 (0x10022138)
1.51   z_impl_k_yield+0x1c:    cbnz	w0, .+228 (z_impl_k_yield+0x100)
1.50   z_impl_k_yield+0x20:    mrs	x9, daif
1.50   z_impl_k_yield+0x24:    msr	daifset, #2
1.50   z_impl_k_yield+0x28:    movz	w0, #1
1.50   z_impl_k_yield+0x2c:    add	x2, x6, #0xc0c
1.50   z_impl_k_yield+0x30:    sevl
1.50   z_impl_k_yield+0x34:    wfe
1.50   z_impl_k_yield+0x38:    ldaxr	w1, [x2]
1.51   z_impl_k_yield+0x3c:    cbnz	w1, .-8 (z_impl_k_yield+0x34)
4.49   z_impl_k_yield+0x40:    stxr	w1, w0, [x2]
4.49   z_impl_k_yield+0x44:    cbnz	w1, .-16 (z_impl_k_yield+0x34)
2.34   z_impl_k_yield+0x48:    mrs	x0, tpidr_el1
2.34   z_impl_k_yield+0x4c:    ldrsb	w0, [x0,#25]
2.34   z_impl_k_yield+0x50:    tbz	w0, #31, .+36 (z_impl_k_yield+0x74)
1.71   z_impl_k_yield+0x74:    mrs	x5, tpidr_el1
1.71   z_impl_k_yield+0x78:    ldrb	w0, [x5,#25]
1.71   z_impl_k_yield+0x7c:    orr	w0, w0, #0xffffff80
1.71   z_impl_k_yield+0x80:    strb	w0, [x5,#25]
1.71   z_impl_k_yield+0x84:    mrs	x0, tpidr_el1
1.71   z_impl_k_yield+0x88:    cmp	x5, x0
1.72   z_impl_k_yield+0x8c:    b.eq	.+68 (z_impl_k_yield+0xd0)
1.66   z_impl_k_yield+0xd0:    mrs	x0, tpidr_el1
1.66   z_impl_k_yield+0xd4:    cmp	x5, x0
1.66   z_impl_k_yield+0xd8:    b.ne	.+16 (z_impl_k_yield+0xe8)
1.85   z_impl_k_yield+0xdc:    mrs	x0, tpidrro_el0
1.85   z_impl_k_yield+0xe0:    movz	w1, #1
1.85   z_impl_k_yield+0xe4:    strb	w1, [x0,#37]
1.85   z_impl_k_yield+0xe8:    mrs	x0, tpidrro_el0
1.85   z_impl_k_yield+0xec:    movz	w1, #1
1.85   z_impl_k_yield+0xf0:    strb	w1, [x0,#37]
1.85   z_impl_k_yield+0xf4:    add	x0, x6, #0xc0c
1.85   z_impl_k_yield+0xf8:    stlr	wzr, [x0]
1.85   z_impl_k_yield+0xfc:    msr	daif, x9
1.85   z_impl_k_yield+0x100:   mrs	x22, daif
1.85   z_impl_k_yield+0x104:   msr	daifset, #2
1.85   z_impl_k_yield+0x108:   mrs	x20, tpidr_el1
1.85   z_impl_k_yield+0x10c:   mrs	x0, daif
1.85   z_impl_k_yield+0x110:   msr	daifset, #2
1.85   z_impl_k_yield+0x114:   movz	w0, #1
1.85   z_impl_k_yield+0x118:   add	x21, x6, #0xc0c
1.85   z_impl_k_yield+0x11c:   sevl
1.85   z_impl_k_yield+0x120:   wfe
1.85   z_impl_k_yield+0x124:   ldaxr	w1, [x21]
1.86   z_impl_k_yield+0x128:   cbnz	w1, .-8 (z_impl_k_yield+0x120)
3.92   z_impl_k_yield+0x12c:   stxr	w1, w0, [x21]
3.92   z_impl_k_yield+0x130:   cbnz	w1, .-16 (z_impl_k_yield+0x120)
2.21   z_impl_k_yield+0x134:   bl	.-2728 (z_swap_next_thread)
0.797   z_impl_k_yield+0x138:   mov	x19, x0
0.797   z_impl_k_yield+0x13c:   cmp	x20, x0
0.797   z_impl_k_yield+0x140:   b.eq	.+140 (z_impl_k_yield+0x1cc)
2.72   z_impl_k_yield+0x190:   msr	daif, x22
2.72   z_impl_k_yield+0x194:   mrs	x0, tpidr_el1
2.72   z_impl_k_yield+0x198:   ldp	x19, x20, [sp,#16]
2.72   z_impl_k_yield+0x19c:   ldp	x21, x22, [sp,#32]
2.72   z_impl_k_yield+0x1a0:   ldp	x29, lr, [sp],#48
2.72   z_impl_k_yield+0x1a4:   ret
0.797   z_impl_k_yield+0x1cc:   stlr	wzr, [x21]
0.797   z_impl_k_yield+0x1d0:   b	.-64 (z_impl_k_yield+0x190)
```
That's `mean_cycle_count offset instruction` (note that the trace sometimes spreads out cycle counts somewhat, so e.g. don't treat this as "the entire final set of stack operations and returns took exactly the same number of instructions")

jharris-intel, 2021-02-26:
> We actually enforce right now that the macro is only ever expanded when interrupts are masked (for obvious correctness reasons).

Uh, this isn't actually the case.

We check that `_current_cpu` is expanded only when non-preemptive, but not `_current`.

See, it _is_ fine, conceptually speaking, to read `_current` even in a racy manner across context switches - because pretty much by definition `_current` will remain pointing to the current thread after we come back from the context switch.

But right now the way we do that is by having `_current` be a field of `_current_cpu`, which _isn't_ fine to read in a racy manner.

andyross, 2021-02-26:
Right.  By definition _current will never change from the perspective of generated code.  That's what it means, really :)

jharris-intel, 2021-02-26:
> There should be 1-3 or so instructions issued in order (and a serialization cost on big CPUs) and you should exit the loop with a successful branch predict. Even our naive implementation based on [...]

It's more about the expected time taken than strict # of instructions. A CISC will (almost) always take less instructions...

> we're just a loop around LOCK CMPXCHG).

What about interrupt lock / unlock?

> Remember though that spinlocks are intended to be super-fast in the uncontended case.

Sure; what's your definition of super-fast here? Do the times I measured in #32686 for the uncontended case match your expectations? (For conversion from cycles<-> time, assume ~1GHz)

And also: this is `sched_lock`, which has more than a little bit of contention...

*****

> Right. By definition _current will never change from the perspective of generated code. That's what it means, really :)

Yep, really `_current` is just a threadlocal. But unforutnately it means that doing something like your previous suggestion:

```c
static ALWAYS_INLINE __attribute__(("pure")) __eval_current()
{
	return _current_cpu->current;
}
```
is fundamentally unsafe if we could preempt.

andyross, 2021-02-26:
I went and looked it up, by the way.  In fact gcc has us covered.  "const" means a function with a return value that depends on nothing but its arguments, "pure" means a function with a return value that depends on nothing but the arguments and global program state.  So in fact we want "const" for _current, meaning it's OK to optimize that under all circumstances, and "pure" for _current_cpu, which allows the compiler to optimize it only to the next function call or memory clobber.  Which is exactly what we want.  So in principle something like this should go into thread_structs.h:

```c
/* These values are performance critical and we want the compiler to
 * be able to optimize their results, even if they are defined in
 * architecture code that might otherwise preclude that (for example
 * as C function calls, volatile __asm__ blocks, etc...).
 */

/* This function is "pure", because it can only be optimized until the
 * next context switch, so function calls and memory clobbers required
 * by locking operations will force it to be re-evaluated 
 */
static ALWAYS_INLINE __attribute__(("pure")) __eval_curr_cpu(void)
{
	return arch_curr_cpu();
}

/* The current thread is by definition unchanging from the perspective
 * of generated code, so this is "const" allowing the compiler to
 * cache its value in any manner.
 */
static ALWAYS_INLINE __attribute__(("const")) __eval_current(void)
{
	return __eval_curr_cpu()->current;
}

/* Note that _current_cpu can change as the thread migrates, so it's
 * never legal to use this result if preemption is possible.  The
 * assert is there to catch mistakes.
 */
#define _current_cpu ({ __ASSERT_NO_MSG(!z_smp_cpu_mobile()); \
			__eval_curr_cpu(); })

#define _current __eval_current()
```

I'll submit when I get a chance, but please try this out if you want and see if the generated code looks correct to you.

andyross, 2021-02-26:
And regarding the preemption-safety of _current.  You're absolutely right that in principle _current should be usable in a preemptible context.  We just don't have any such code right now[1]: everywhere that touches it is in an ISR or under a lock.  And even if you could... there's not much in the thread struct to be inspected that doesn't require synchronization.

[1] Well, the assert only exists in SMP, so it's possible there are code paths on single CPU configurations that try this I guess.  But the point is moot there as _current_cpu == 0 by definition.

jharris-intel, 2021-02-26:
> everywhere that touches it is in an ISR or under a lock.

I think I'm missing something. You've asserted this a few times, but I do not think this is the case:

See e.g. `eswifi_lock`:

```c
static inline void eswifi_lock(struct eswifi_dev *eswifi) {
	if (atomic_get(&eswifi->mutex_owner) != (atomic_t)(uintptr_t)_current) {
	[...]
}
```
Or `k_mbox_put`:

```c
int k_mbox_put(struct k_mbox *mbox, struct k_mbox_msg *tx_msg, k_timeout_t timeout) {
	tx_msg->_syncing_thread = _current;
	[...]
```

Neither of these are required to be in locked/ISR context I believe.

*****

> please try this out if you want and see if the generated code looks correct to you.

Sure, shall do.

I am mildly concerned here about the scope of const/pure. I could see a compiler writer (note: I don't think either gcc or clang do this currently) wrapping `__eval_current` with, roughly:

```c
thread_t *wrap___eval_current(void) {
    static thread_t *wrapped = __eval_current(); /* assume threadsafe static initialization */
    return wrapped;
}
```

...which is fine according to the definition, but would rather fall apart here.

carlocaione, 2021-02-27:
> This proposal would have the following assignments:
> 
>     1. `tpidr_el0` is still the TLS pointer.
>     2. `tpidrro_el0` stores `&_kernel.cpus[core]`.
>     3. `tpidr_el1` stores `_current`.

Just an heads up that `tpidrro_el0` soon will be not available anymore because we are going to use that to implement `arch_is_user_context()` as part of the userspace implementation that we are going to push next week.

jharris-intel, 2021-03-01:
Wait, how are you using it for `arch_is_user_context`? Is there a PR for this?

carlocaione, 2021-03-01:
> Wait, how are you using it for `arch_is_user_context`? Is there a PR for this?

There will be soon, hopefully today or tomorrow, for sure by the end of this week.

jharris-intel, 2021-03-01:
> There will be soon, hopefully today or tomorrow, for sure by the end of this week.

Great. I'm hoping that the use for `arch_is_user_context` can be coordinated with this. It's a major performance win.

jharris-intel, 2021-03-01:
Unfortunately, due to #32710 the above measurements are incorrect. Or rather, they were measuring something different than I had expected.

I was trying to measure `k_yield` as a quick (as in low-overhead) way to measure context switch time.

Unfortunately, due to #32710 `k_yield` wasn't actually yielding, and it took me an embarrassingly long time to realize that that was the case.

So this was measuring/optimizing the time for `k_yield` when it _didn't_ context-switch.

More numbers coming up.

carlocaione, 2021-03-01:
> Wait, how are you using it for `arch_is_user_context`? Is there a PR for this?

@jharris-intel well, no reason to wait. So the commit is a variation of this commit https://github.com/carlocaione/zephyr/commit/49727c58edb4e96ab5f67f2445e1429a8724e281 that will be part of a big PR that will be pushed soon.

Anyway the gist is that we use `tpidrro_el0` as a register to hold a value to indicate whether we are in user or kernel context. Given that we do not have access to `currentEL` from `EL0` this was an handy option to quickly determine that.

jharris-intel, 2021-03-01:
> Anyway the gist is that we use tpidrro_el0 as a register to hold a value to indicate whether we are in user or kernel context. Given that we do not have access to currentEL from EL0 this was an handy option to quickly determine that.

Hmm. So you're setting it / clearing it on every kernel-mode entry/exit then?

What's the advantage over putting this as a member of the `cpus` struct?

One possibility...

```
1. `tpidr_el0` is still the TLS pointer.
3. `tpidrro_el0` stores `_current` if in EL1, NULL if in usermode.
2. `tpidr_el1` stores `&_kernel.cpus[core]`.
```

(As you can't access `_current` in usermode anyway I believe, and on a context switch you know what you're switching to to restore `_current`...)

Then your "is in kernel context" check becomes `tpidrro_el0 != NULL`.

jharris-intel, 2021-03-01:
(You could also do something fancy like take advantage of the alignment of cpu and set/clear the low bit; I don't know if it's worth it.)