zephyrproject-rtos/zephyr #32724: qemu timer change introducing new CI failures

pabigot, 2021-02-28:
It appears #32557 introduced some timing problems in qemu x86: various tests are failing when run under heavy load.

NB: I've seen comments in other PRs saying that failures in CI that are related to deadline checks aren't due to the PR changes, but this one's mine so these are the details I can provide:

The work2 branch in #29618 is now failing check on current master, with errors like
```
START - test_fifo_cancel_wait

    Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/fifo/fifo_api/src/test_fifo_cancel.c:49: tfifo_thread_thread: (dur < 80 is false)
k_fifo_get didn't get cancelled in expected timeframe
 FAIL - test_fifo_cancel_wait
```
in tests that don't even use a work queue.

The work tests that failed pass when run individually.  When invoked in a group with:

    twister -T tests/kernel/workq/

I do get failures in work-related tests, like:

```
ERROR   - qemu_x86_64               tests/kernel/workq/work/kernel.work.api            FAILED: Failed
ERROR   - see: /mnt/nordic/zp/zephyr/twister-out/qemu_x86_64/tests/kernel/workq/work/kernel.work.api/handler.log
INFO    - Total complete:   28/ 125  22%  skipped:   17, failed:    1
ERROR   - qemu_x86_64_nokpti        tests/kernel/workq/work/kernel.work.api            FAILED: Failed
ERROR   - see: /mnt/nordic/zp/zephyr/twister-out/qemu_x86_64_nokpti/tests/kernel/workq/work/kernel.work.api/handler.log
```

which correspond to one of these failures (each is intermittent):
```
    Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/workq/work/src/main.c:1044: test_1cpu_basic_reschedule: (elapsed_ms <= max_ms is false)
long 120 > 110
    Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/workq/work/src/main.c:1308: test_1cpu_system_reschedule: (elapsed_ms <= max_ms is false)
long 230 > 110
    Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/workq/work/src/main.c:1400: test_1cpu_legacy_delayed_resubmit: (elapsed_ms <= max_ms is false)
long 120 > 110
```

Interestingly if I'm running in a file system that's on an older SATA SSD instead of a modern NVME the failures don't occur.

If I cherrypick the commit set onto mainline commits with bisect:
* 647873549334e94b8a8e13f62b9631adf04c45cf which is right before #32557: the tests pass;
* 23f699bd907fbea22ca0e50e1295bb1a94adbca9 which is the last commit in #32557: I get the above failures;
* a202670c188182e27ce15ddf871a96fd43ca381f ok 2 x
* bf99f3105fff59e17e137489c209bcb3b06ff8d9 fails
* 4ae44dd71231abb80a45e8dd1d23127c95d61929 ok 1 x then fails
* 419f37043be54bc8e7fc53974e005a3cc9f4b31d ok 3x

so the problem seems to be 4ae44dd71231abb80a45e8dd1d23127c95d61929, which makes sense since that changes how qemu deals with time.

andyross, 2021-03-01:
Is this happening in master or CI, or just your own branch?  For reference, I got through ~250  runs of "twister -p qemu_x86_64" (I left it running in a loop overnight) of the PR just before merge on my own build box with no failures at all.  Well, until it ran out of disk space, anyway.  What's the exerciser exactly?

nashif, 2021-03-01:
> Well, until it ran out of disk space, anyway.

use -M

andyross, 2021-03-01:
An option I totally remembered once I noticed the machine was out of disk space.

pabigot, 2021-03-01:
> Is this happening in master or CI, or just your own branch? For reference, I got through ~250 runs of "twister -p qemu_x86_64"

Yes, it seems to be dependent on high load, and possibly fast disk or other host variations.

The only reproducer I have is that set of commits---primarily because the new work tests have a lot of deadline checks (though note tests that don't involve work did fail in https://buildkite.com/zephyr/zephyr/builds/22361#d5471ff6-b845-4b53-b16e-5e0090836730).  Others have commented about the failure elsewhere but I don't remember where; @carlescufi confirmed this morning that he'd seen it, but thought resubmitting would make the problem disappear (it's that fragile).

jharris-intel, 2021-03-01:
FYI, I'm hitting this too locally under high concurrency.

Also, see #14173.

Does `-icount` and/or `-rtc clock=vm` help?

...though `-icount` has other issues. (`-icount` essentially tells QEMU to target a specific # of processor clock cycles / second instead of an unlimited amount, but also restricts QEMU to one host thread... which in some high-cross-cpu-traffic scenarios can seriously reduce effective throughput.)

PR #32790 (2021-03-02): Revert "drivers/timer/hpet: Extend qemu workaround"
This reverts commit 4ae44dd71231abb80a45e8dd1d23127c95d61929.

Fixes #32724.  See also: https://github.com/zephyrproject-rtos/zephyr/pull/29618#issuecomment-788950966
closed 2021-03-02 by nashif