zephyrproject-rtos/zephyr #32726: Split UART_RX_RDY into distinct callback events

Totalus, 2021-02-28:
In the current UART library implementation, the callback function is called with the `UART_RX_RDY` event type in 4 different scenarios:

- When RX timeout occured
- When provided buffer is full
- After stopping due to external event
- After uart_rx_disable()

**What is the issue with that ?**

The action to be taken on those events is not necessarely the same and being combined into one single event forces the user to implement a logic to distinguish them. In some cases it can be easy, but in some cases it becomes harder to differentiate.

**Typical use case**

I want to capture UART packets of various size. The indicator of the end of the packet is a timeout which causes a callback with UART_RX_RDY event. The issue is that I also get the same event called when the provided buffer is full even if the received packets is not entirely received yet (overlaps on the next buffer). This creates 2 distinct calls for 1 single packet (first call being the buffer is full and a second one for the packet timeout). Being the same event forces me to implement a logic to distinguish them, and also store the length so I can add it to the second event to figure out the exact packet length. And in some cases, both event are the same (a full packet was received AND it filled up the buffer), and this is only possible to catch if I know what is the packet length.

**Suggested solution**

Split the UART_RX_RDY event in more than one event to allow more granularity on the behavior for each event. Which event should have a different event type can be discussed. Here is a suggestion that seems convenient:

- The `UART_RX_RDY` event should only be called on a **timeout event**. This allows the use to distinguish each full UART packet received without confusion.

- In the case where **the provided buffer is full**, it should either be a distinct event (`UART_RX_BUFFER_FULL`) or not trigger an event at all (is there a use case for this event ?).

- On the **uart is stopped** event, the last rx data is already part of the event details, therefore the user can handle remaining data from the `UART_RX_STOPPED` event directly if needed. He will know it is likely an incomplete packet. We juste need to make sure the provided data is invalid if no data was received since last complete packet. No need to generate an `UART_RX_RDY` event.

- In the **uart was disabled** scenario the `UART_DISABLED` event could contain the last rx data like the `UART_RX_STOPPED` event or generate a call to the `UART_RX_STOPPED` event with a `UART_DISABLED` reason as `uart_rx_stop_reason`. This would allow the user to handle the remaining data.

nordic-krch, 2021-03-02:
> This allows the use to distinguish each full UART packet received without confusion.

Can you guarantee that there will be no gaps in the transfer that will trigger timeout not on packet boundary? In many cases probably yes but not always.

> In the case where the provided buffer is full, it should either be a distinct event (UART_RX_BUFFER_FULL) or not trigger an event at all (is there a use case for this event ?)

`UART_RX_RDY` can provide only one buffer and 2 buffers are involved if there was a switch in between. There is a corner case scenario when buffer ended on packet boundary. `UART_RX_BUFFER_FULL` would be generated but no timeout since there is no new data.
Note, that if you want to rely on timeouts to detect packet boundary you can use `UART_RX_RDY` which contains `offset` and `len`. I you know the size of the provided buffer then if `offset + len` is equal to the buffer size then event occured on buffer full. You can start the timer and if it timeouts before next `UART_RX_RDY` it means that it was also packet boundary. If new data arrives before the timeout then timer can be stopped. I don't like the alternative with `UART_RX_BUFFER_FULL` since there would need to be `UART_RX_RDY` with 0 len to indicated timeout.

Regarding `UART_RX_RDY` after disabling, as user explicitly called `uart_rx_disable` then state is known and controlled by the user and data can be discarded or not. 

Regarding content of rx data in `UART_RX_STOPPED` event: at least in nordic implementation this data is not set at all and valid data is sent in `UART_RX_RDY` occuring after `UART_RX_STOPPED`. It seems like a bug then. Anyway, even in nordic implementation, scenario can be distinguished, first `UART_RX_STOPPED` event occurs, followed by `UART_RX_RDY` (if there was any data received) and finally `UART_RX_DISABLED`.

Totalus, 2021-03-02:
> 
> 
> > This allows the use to distinguish each full UART packet received without confusion.
> 
> Can you guarantee that there will be no gaps in the transfer that will trigger timeout not on packet boundary? In many cases probably yes but not always.

The only case I can see where there would be a timeout not on a packet boundary is if the remote host does not send a full packet in one chunk, which could happen for two reasons:

- The remote host UART driver is not implemented properly, you can't do anything about (not in the local host implementation).
- The flow control is active and the communication is stopped because the UART fifo is full. In this case, the timer should probably be paused until the transmission can continue to avoid triggering a timeout before the end of the packet.

In any case the use of this type of implementation assumes that both hosts communicating are aware that packets are delimited by timeouts and therefore ensure that transmission does not have any gap and a minimal idle time must be inserted in between each packet to ensure a timeout in the remote host.

> `UART_RX_RDY` can provide only one buffer and 2 buffers are involved if there was a switch in between. There is a corner case scenario when buffer ended on packet boundary. `UART_RX_BUFFER_FULL` would be generated but no timeout since there is no new data.

I don't see why this corner case scenario would not trigger a timeout. The timeout is calculated between two consecutive received bytes, wether the buffer fills up or not. In this case, we would first have a `UART_RX_BUFFER_FULL` event since the buffer filled, and then a `UART_RX_RDY` event since there was not new uart bytes received for the given timeout period.

> Note, that if you want to rely on timeouts to detect packet boundary you can use `UART_RX_RDY` which contains `offset` and `len`. I you know the size of the provided buffer then if `offset + len` is [...]

There are workarounds, but what is the point of integrating a packet timeout in the library if you need to implement your own code and timeout on top of it to properly detect packets ? Also how do you do if you do not know the lenght of the packet ?

> I don't like the alternative with `UART_RX_BUFFER_FULL` since there would need to be `UART_RX_RDY` with 0 len to indicated timeout.

Timeout should occur only after new bytes of data were received after the last `UART_RX_RDY` event has been emitted. In between the last call to `UART_RX_RDY` event and the first next UART byte received, the timer should be stopped. Therefore there should never be any `UART_RX_RDY` with lenght 0. I think this is already the case.

> Regarding `UART_RX_RDY` after disabling, as user explicitly called `uart_rx_disable` then state is known and controlled by the user and data can be discarded or not.

That is true. Having the information in the event just makes it easier to handler and avoid storing the state, but maybe it is not possible (ex: the data may not yet be all written to the RAM buffer).

> Regarding content of rx data in `UART_RX_STOPPED` event: at least in nordic implementation this data is not set at all and valid data is sent in `UART_RX_RDY` occuring after `UART_RX_STOPPED`. It [...]

Yes, the stop event is the easiest to distinguish. It is possible that the data is not available in RAM when the `UART_RX_STOPPED` event is generated, thus having to call a `UART_RX_RDY` after when the data is ready to process. I'm not sure about the internal details.