zephyrproject-rtos/zephyr #32733: RS-485 support

manoj153, 2021-03-01:
is there a special binding exclusive for RS-485?

I see Linux has support for it.
https://www.kernel.org/doc/html/latest/driver-api/serial/serial-rs485.html

How is this interface being or suppose to handle in zephyr from device tree, read-write API and binding perspective?

mbolivar-nordic, 2021-03-01:
Hi there. Zephyr doesn't have an equivalent devicetree binding or support as far as I can tell.

jfischer-no, 2021-03-03:
AFAIK we do not have explicit support for this, even though some controllers have RS-485 mode. For MODBUS RTU support I have added [optional GPIOs](https://github.com/zephyrproject-rtos/zephyr/pull/26516/commits/877366301920bedbe2d65a916a03587d12acfc63) to control RS-485 transceiver.

manoj153, 2021-03-04:
Guys what is the thought if I contribute a generic-rs_485 binding for zephyr. 
A lot of industrial devices uses rs_485 and they might not necessarily leverage Modbus. FME, tons of vendor releases custom stream byte protocol. This planned binding should help out?

jfischer-no, 2021-03-04:
> Guys what is the thought if I contribute a generic-rs_485 binding for zephyr.
> A lot of industrial devices uses rs_485 and they might not necessarily leverage Modbus. FME, tons of vendor releases custom stream byte protocol. This planned binding should help out?

I do not see how that could help currently. What do you want to describe with a generic-rx_485 binding, non-existing RS-485 support for optional feature of an UART controller?

manoj153, 2021-03-22:
May need to set the registers for the SOC to generate the DE signal automatically as an alternative to IO toggling in the code

Yours is exclusive to Modbus, and I suspect many users may need a custom one to support non-modbus compliant devices that use RS-485 bus.

I can think we could have a generic sample as well in this case.

manoj153, 2021-03-24:
@erwango

closed 2021-06-07 by github-actions[bot]

CraigBurden, 2022-10-04:
@manoj153 I would certainly be interested in that. My product uses a proprietary protocol over RS485 and so I am in the market for a generic driver that I can build on top of.

@jfischer-no I guess it could be an extension to the UART controller since all it would require is an addition to the existing flow control logic.

Any contribution made should be coupled with a refactor of the Modbus RTU implementation to use the new functionality.
I will build this at some stage so if it hasn't been upstreamed by then I will do so

wniva, 2024-04-02:
I would also be interested in such a solution (via UART + GPIO's). I use Nordic nRF52832/nRF52840.