zephyrproject-rtos/zephyr #32756: Enable mcumgr shell management to send responses to UART other than assigned to shell

de-nordic, 2021-03-01:
Currently it is possible to assign different UART devices to mcumgr and shell/console, for example via chosen `zephyr,uart-mcumgr ` and `zephyr,shell-uart` respectively. 
This allows in most cases to operate mcumgr, for example for uploads/downloads,  and have shell/console available on other uart without interruptions from mcumgr.
This does not work when mcumgr is used with shell management: all shell requests are given response with error 8 (not supported).
When both, the mcumgr and the shell, are set to the same uart this works fine.

**Describe the solution you'd like**
To be able to used mcumgr shell management on one UART and have shell connected on the other UART.

nordicjm, 2022-08-25:
Cannot replicate this issue, have enabled uart1 for a board and changed the chosen nodes to:
```
                zephyr,console = &uart0;
                zephyr,shell-uart = &uart1;
```
uart1 is actually not connected to an external UART so I cannot monitor it, but with mcumgr configured to use UART instead of shell, I can send shell commands and get responses - sending `kernel stacks`:
```
Op:  3
Flags:  0
Length:  320
Group:  9
Sequence:  1
ID:  0
0 true
0 true
Text:  "o"
Text:  "\r\n0x20000c48 sysworkq   (real size 2304):\tunused 1272\tusage 1032 / 2304 (44 %)\r\n0x200009e8 shell_dummy (real size 2048):\tunused 1668\tusage 380 / 2048 (18 %)\r\n0x20000930 shell_uart (real size 2048):\tunused 1668\tusage 380 / 2048 (18 %)\r\n0x20000878 logging    (real size 768):\tunused 232\tusage 536 / 76"
Text:  "rc"
number:  0
```
Sending `kernel version`:
```
Op:  3
Flags:  0
Length:  45
Group:  9
Sequence:  1
ID:  0
0 true
0 true
Text:  "o"
Text:  "\r\nZephyr version 3.1.99\r\n"
Text:  "rc"
number:  0
```
Getting task list (not a shell command):
```
Op:  1
Flags:  0
Length:  195
Group:  0
Sequence:  1
ID:  2
0 true
0 true
Text:  "tasks"
0 true
Text:  "sysworkq"
0 true
Text:  "prio"
number:  4294967295
Text:  "tid"
number:  0
Text:  "state"
number:  128
Text:  "shell_dummy"
0 true
Text:  "prio"
number:  14
Text:  "tid"
number:  1
Text:  "state"
number:  2
Text:  "shell_uart"
0 true
Text:  "prio"
number:  14
Text:  "tid"
number:  2
Text:  "state"
number:  2
Text:  "logging"
0 true
Text:  "prio"
number:  14
Text:  "tid"
number:  3
Text:  "state"
number:  2
Text:  "idle"
0 true
Text:  "prio"
number:  15
Text:  "tid"
number:  4
Text:  "state"
number:  0
Text:  "main"
0 true
Text:  "prio"
number:  0
Text:  "tid"
number:  5
Text:  "state"
number:  16
```

nordicjm, 2022-10-04:
Cannot replicate issue with zephyr 3.2, closing.

closed 2022-10-04 by nordicjm