zephyrproject-rtos/zephyr #32771: STM32 with Ethernet crashes when receiving packets early

jonasn-ge, 2021-03-02:
The STM32 Ethernet driver can receive data before the interface is ready to handle it, leading to a crash.

The problem can typically be reproduced within a couple of tries if the network is filled with traffic, I do that by pinging the IPv4 broadcast address with a 0.001 s interval, when starting the STM32 board.

Different versions have crashed in different ways, but in stock 2.5.0 this is what I get before even main() is called. Asserts enabled:

```
ASSERTION FAIL [net_if_get_link_addr(iface)->addr != ((void *)0)] @ WEST_TOPDIR/oc_zephyr/subsys/net/ip/net_if.c:3468
[00:00:04.368,000] <err> eth_stm32_hal: Failed to enqueue frame into RX queue: -62
[00:00:04.368,000] <err> eth_stm32_hal: Failed to enqueue frame into RX queue: -62
[00:00:04.368,000] <err> eth_stm32_hal: Failed to enqueue frame into RX queue: -62
[00:00:04.368,000] <err> eth_stm32_hal: Failed to enqueue frame into RX queue: -62
[00:00:04.379,000] <err> os: r0/a1:  0x00000004  r1/a2:  0x00000d8c  r2/a3:  0xffffffbf
[00:00:04.379,000] <err> os: r3/a4:  0x0802f52d r12/ip:  0x00000000 r14/lr:  0x0803317d
[00:00:04.379,000] <err> os:  xpsr:  0x41000000
[00:00:04.379,000] <err> os: Faulting instruction address (r15/pc): 0x0804761c
[00:00:04.379,000] <err> os: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
[00:00:04.379,000] <err> os: Current thread: 0x20004f98 (sysworkq)
[00:00:04.457,000] <err> os: Halting system
```
The above assertion indicates that the net_if_set_link_addr call could benefit from happening earlier in eth_iface_init() in eth_stm32_hal.c. But it's still crashing after that:

```
[00:00:04.418,000] <err> os: ***** USAGE FAULT *****
[00:00:04.418,000] <err> os:   Illegal use of the EPSR
[00:00:04.418,000] <err> os: r0/a1:  0x20004448  r1/a2:  0x20004ec8  r2/a3:  0x20004ec8
[00:00:04.418,000] <err> os: r3/a4:  0x00000000 r12/ip:  0x00000000 r14/lr:  0x080462df
[00:00:04.418,000] <err> os:  xpsr:  0x6000000f
[00:00:04.418,000] <err> os: Faulting instruction address (r15/pc): 0x00000000
[00:00:04.418,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:04.418,000] <err> os: Fault during interrupt handling
```
The above exception appears to be because net_if_up is called (in k_sys_work_q?) before net_if_init has finished executing.

A bit of k_sleep before the first call to net_eth_carrier_on in rx_thread() takes care of that, and another k_sleep after the call takes care of the "Failed to enqueue frame into RX queue" errors. But that's hardly an elegant solution.

**Environment:**
 - Zephyr 2.5.0 (although it's been happening since at least 2.0)
 - STM32F429

erwango, 2021-03-02:
@jonasn-ge Thanks for reporting this issue and the details provided so far.
To fasten analysis and eventually bug fixing, could you share a sample to reproduce the issue easily (or modification of an in tree sample)?

jonasn-ge, 2021-03-02:
@erwango I will work on a sample.

jonasn-ge, 2021-03-02:
It turns out that another ingredient is needed to get the crash: we need to disable auto-negotiation. That probably has the effect of making the timing just right (or wrong).

Attached is a minimal reproducer, including a "hack" to disable auto-negotiation. I have tried it on our custom board, unfortunately I don't have a Nucleo board on hand to test with. When all is well the program does nothing, but when bombarded with Ethernet frames it halts in the assert in net_if_up().
[reproduce-32771.zip](https://github.com/zephyrproject-rtos/zephyr/files/6069542/reproduce-32771.zip)

FRASTM, 2021-03-31:
running the this sample on a nucleo_f767zi, gives : 
```
*** Booting Zephyr OS build zephyr-v2.5.0-1836-g61aab7519e9c  ***               
[00:00:02.628,000] <inf> main: Hello                                            
[00:00:02.895,000] <err> eth_stm32_hal: Failed to enqueue frame into RX queue: 2
[00:00:00.000,000] <err> main: Missing Ethernet device                          
*** Booting Zephyr OS build zephyr-v2.5.0-1836-g61aab7519e9c  ***               
[00:00:03.278,000] <inf> main: Hello                                            
```

chrta, 2021-04-06:
@jonasn-ge Do you have the MPU enabled? I am currently facing a similar problem on the nucleo_f429zi with zephyr 2.5.0. I get MPU faults (Instruction Access Violation) during ethernet initialization for certain applications (not all), but i am still in the process to identify the reason or an reproducer.

jonasn-ge, 2021-04-07:
@FRASTM: Yes, I have noticed that the reproducer code has stopped working against Zephyr master. It appears that it's no longer possible to get a device pointer before the device is initialized, or something. What still works is modifying the eth0_data struct in eth_stm32_hal.c with the new auto-negotiation settings. Then the reproducer main.c can be essentially empty.

jonasn-ge, 2021-04-07:
> Do you have the MPU enabled?
Yes, MPU is enabled. Though I mostly experienced the code trying to jump to a NULL pointer and the MPU doesn't protect against that currently (there is an open PR for that, last time i looked).

erwango, 2021-04-07:
> @FRASTM: Yes, I have noticed that the reproducer code has stopped working against Zephyr master. It appears that it's no longer possible to get a device pointer before the device is initialized, or something. 

Exact. 

> Though I mostly experienced the code trying to jump to a NULL pointer and the MPU doesn't protect against that currently (there is an open PR for that, last time i looked).

If this is `CORTEX_M_DEBUG_NULL_POINTER_EXCEPTION_DETECTION` it has been merged.
You can enable it using `CONFIG_CORTEX_M_DEBUG_NULL_POINTER_EXCEPTION_DETECTION_MPU=y`

FRASTM, 2021-04-13:
@jonasn-ge could you please check the CONFIG_CORTEX_M_DEBUG_NULL_POINTER_EXCEPTION_DETECTION_MPU ( force setting to 'y' in the prj.conf) and run the test

jonasn-ge, 2021-04-15:
@FRASTM I have run with current Zephyr master (zephyr-v2.5.0-2255-g0dc7b9ee46), patched to disable auto-negotiation in eth_stm32_hal.c. The behaviour is the same as I've seen before when I start filling the network with traffic ("ping 10.10.10.255 -i 0.005 -A" from a Busybox shell work well for me).

With CONFIG_ASSERT=y:
```
*** Booting Zephyr OS build zephyr-v2.5.0-2255-g0dc7b9ee460c  ***
[00:00:04.369,000] <inf> main: Hello
ASSERTION FAIL [handler != 0] @ WEST_TOPDIR/zephyr/kernel/work.c:669
[00:00:04.364,000] <err> eth_stm32_hal: Failed to enqueue frame into RX queue: -62
[00:00:04.370,000] <err> os: r0/a1:  0x00000004  r1/a2:  0x0000029d  r2/a3:  0xffffffbf
[00:00:04.370,000] <err> os: r3/a4:  0x08001d15 r12/ip:  0xa0000000 r14/lr:  0x0800b5b5
[00:00:04.370,000] <err> os:  xpsr:  0x41000000
[00:00:04.370,000] <err> os: Faulting instruction address (r15/pc): 0x0800cac0
[00:00:04.370,000] <err> os: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
[00:00:04.370,000] <err> os: Current thread: 0x20001488 (unknown)
[00:00:04.423,000] <err> os: Halting system
```

With CONFIG_CORTEX_M_DEBUG_NULL_POINTER_EXCEPTION_DETECTION_MPU=y instead:
```
[00:00:04.364,000] <err> eth_stm32_hal: Failed to enqueue frame into RX queue: -62
[00:00:04.364,000] <err> os: ***** MPU FAULT *****
[00:00:04.364,000] <err> os:   Instruction Access Violation
[00:00:04.364,000] <err> os: r0/a1:  0x2000165c  r1/a2:  0x00000000  r2/a3:  0x00000001
[00:00:04.364,000] <err> os: r3/a4:  0x00000001 r12/ip:  0x00000000 r14/lr:  0x080085d7
[00:00:04.364,000] <err> os:  xpsr:  0x60000000
[00:00:04.364,000] <err> os: Faulting instruction address (r15/pc): 0x00000000
[00:00:04.364,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:04.364,000] <err> os: Current thread: 0x20001478 (unknown)
[00:00:04.429,000] <err> os: Halting system
```

LR 0x080085d6 points into work_queue_main(), a couple of lines after the __ASSERT_NO_MSG(handler != 0) that fails if asserts are enabled.

FRASTM, 2021-04-21:
@jonasn-ge, could you please describe the exact test environment (target, ethernet connection to/from, test appli, command, etc) all the setup to reproduce the issue.

ABOSTM, 2021-05-12:
@jonasn-ge, I could not exactly reproduce this issue in the same condition than you,
but I succeed to reproduce it with a step by step debugging (probably giving more time to the ping event to arrive at the wrong place).
As I could see, problem is due to the fact that IRQ is enabled (in `eth_iface_init()` thanks to `DEV_CFG(dev)->config_func()` ) 
before iface address is set `net_if_set_link_addr()`. and thus IRQ linked to ping is handled whereas init is not finished.

I proposed a fix #35218 for that by moving IRQ enable at the far end of the `eth_iface_init()` .
It seems working in my setup: going step by step I could not get your original ASSERT error.
But I face another issue (Usage Fault) ... which I hope is linked to step by step execution.
So @jonasn-ge could you test this fix with your original setup
and tell me if you reproduce the original issue or if you face the usage fault issue ?

@jukkar, 
I am not that expert in ethernet, so your review and comment would be very welcome,
specially around potential side effect. Thanks  :-)

jonasn-ge, 2021-05-14:
@ABOSTM thank you for looking into this. Yes, the patch in #35218 helps.

With v2.6.0-rc1-59-g8e69dafccf25 I typically get a crash wtih `ASSERTION FAIL [handler != ((void *)0)] @ WEST_TOPDIR/zephyr/kernel/work.c:709` (or a USAGE FAULT if I disable asserts)
With v2.6.0-rc1-60-gc31c4f9413 (@ABOSTM 's patch) I can't reproduce any crash.

I still get a bunch of `<err> eth_stm32_hal: Failed to enqueue frame into RX queue: -62` in the log shortly after start, but they don't seem to actually be a problem.

My hardware setup consists of my STM32-based board connected to a cheap Netgear GS105 switch which in turn connects to my network and PC. The PC is sending broadcast packets with "sudo ping -b 10.10.20.255 -i 0.01". Having the The GS105 switch in front of the board makes a difference, perhaps it's quick to start sending packets when the port comes up.

I also use the following patch to disable auto-negotiation:
```
diff --git a/drivers/ethernet/eth_stm32_hal.c b/drivers/ethernet/eth_stm32_hal.c
index a36ca61a68..4ff9bd111e 100644
--- a/drivers/ethernet/eth_stm32_hal.c
+++ b/drivers/ethernet/eth_stm32_hal.c
@@ -897,6 +897,9 @@ static struct eth_stm32_hal_dev_data eth0_data = {
 #else
                        .MediaInterface = ETH_MEDIA_INTERFACE_RMII,
 #endif
+                       .AutoNegotiation = ETH_AUTONEGOTIATION_DISABLE,
+                       .Speed = ETH_SPEED_100M,
+                       .DuplexMode = ETH_MODE_FULLDUPLEX,
                },
        },
        .mac_addr = {
```

jukkar, 2021-05-14:
> I still get a bunch of `<err> eth_stm32_hal: Failed to enqueue frame into RX queue: -62` in the log shortly after start, but they don't seem to actually be a problem.

The errno 62 is ENETDOWN, which means that from upper network stack point of view, the network interface is still down and packets cannot be accepted yet. Probably we could check this ENETDOWN value and not print anything in this case as this is kind of normal situation.

mrfuchs, 2021-05-17:
I'm experiencing a similar behavior with my board (custom, STM32F437VI SOC) and Zephyr 2.5. Unfortunately, neither #35218, nor disabling auto-negotiation helps with my setup.

However, I noticed that breaking at `HAL_Delay(PHY_RESET_DELAY);` and jumping above the delay with the debugger helps in my case. Interestingly, that's exactly the line before auto-negotiation is set up:

https://github.com/zephyrproject-rtos/hal_stm32/blob/f8ff8d25aa0a9e65948040c7b47ec67f3fa300df/stm32cube/stm32f4xx/drivers/src/stm32f4xx_hal_eth.c#L340-L343

So, my issue may be related to the one reported here.

closed 2021-05-17 by nashif

mrfuchs, 2021-05-17:
Combining all three patches makes it work for me:

1. Apply #35218
2. Apply patch re. auto-negotiation disable from https://github.com/zephyrproject-rtos/zephyr/issues/32771#issuecomment-841075466
3. Comment the calls to `HAL_Delay()` in `HAL_ETH_Init()`:

```diff
 stm32cube/stm32f4xx/drivers/src/stm32f4xx_hal_eth.c | 4 ++--
 1 file changed, 2 insertions(+), 2 deletions(-)

diff --git a/stm32cube/stm32f4xx/drivers/src/stm32f4xx_hal_eth.c b/stm32cube/stm32f4xx/drivers/src/stm32f4xx_hal_eth.c
index 2b6567b1..8fee17c7 100644
--- a/stm32cube/stm32f4xx/drivers/src/stm32f4xx_hal_eth.c
+++ b/stm32cube/stm32f4xx/drivers/src/stm32f4xx_hal_eth.c
@@ -338,7 +338,7 @@ HAL_StatusTypeDef HAL_ETH_Init(ETH_HandleTypeDef *heth)
   }
 
   /* Delay to assure PHY reset */
-  HAL_Delay(PHY_RESET_DELAY);
+//  HAL_Delay(PHY_RESET_DELAY);
 
   if((heth->Init).AutoNegotiation != ETH_AUTONEGOTIATION_DISABLE)
   {
@@ -475,7 +475,7 @@ HAL_StatusTypeDef HAL_ETH_Init(ETH_HandleTypeDef *heth)
     }
 
     /* Delay to assure PHY configuration */
-    HAL_Delay(PHY_CONFIG_DELAY);
+//    HAL_Delay(PHY_CONFIG_DELAY);
   }
 
   /* Config MAC and DMA */

```

Otherwise the system crashes/reboots here (with the stack trace below):
https://github.com/zephyrproject-rtos/zephyr/blob/21d1ad3762302b3e461953df59430c77e0709274/arch/arm/core/aarch32/swap.c#L41

```
Thread #1 57005 (Suspended : Breakpoint)	
	arch_swap() at swap.c:41 0x801e7a6	
	z_swap_irqlock() at kswap.h:162 0x8034074	
	z_swap() at kswap.h:173 0x8034074	
	z_tick_sleep() at sched.c:1’268 0x8034074	
	z_tick_sleep() at sched.c:1’235 0x8034074	
	z_impl_k_sleep() at sched.c:1’296 0x80340b4	
	k_sleep() at kernel.h:86 0x80398f8	
	HAL_ETH_Init() at stm32f4xx_hal_eth.c:341 0x80292f4	
	eth_initialize() at eth_stm32_hal.c:686 0x801e38e	
	z_sys_init_run_level() at device.c:61 0x80334ba	
	bg_thread_main() at init.c:149 0x8033696	
	z_thread_entry() at thread_entry.c:29 0x8038e88	
	arch_switch_to_main_thread() at thread.c:526 0x801e9a4	
	0xaaaaaaaa
```

erwango, 2021-05-17:
@mrfuchs Thanks for these inputs, but since this ticket is now closed (as #35218 was merged). Can you open a new one ?

PR #35218 (2021-05-17): drivers: ethernet: eth_stm32_hal.c: enable IRQ at the end of iface init
drivers: ethernet: eth_stm32_hal.c: enable IRQ at the end of iface init

This avoid IRQ to be handle before iface init is finished
(especially before iface address is set)
Fixes #32771
commit d915eb82 (2021-05-17): drivers: ethernet: stm32: enable IRQ at the end of iface init

This avoid IRQ to be handle before iface init is finished
(especially before iface address is set)
Fixes #32771