zephyrproject-rtos/zephyr #32789: USB DFU support for STM32F1 series

ycsin, 2021-03-02:
**Is this request related to a missing driver support for a particular hardware platform, SoC or board? Please describe.**
Currently the `samples/subsys/usb/dfu` samples can't be compile for STM32F1 series SoC. Trying to compile will result in an error due to Kconfig warnings, as the F1 series is missing the MPU Kconfigs required by `USB_DFU_CLASS`.

```
warning: MPU_ALLOW_FLASH_WRITE (defined at C:/Users/yongc/zephyrproject_master/zephyr/arch/arm\core/aarch32\cortex_m\mpu/Kconfig:78) has direct dependencies ARM_MPU && CPU_HAS_MPU && CPU_CORTEX_M && !ARM64 && ARM with value n, but is currently being y-selected by the following symbols:
 - USB_DFU_CLASS (defined at subsys/usb/class/dfu/Kconfig:4), with value y, direct dependencies IMG_MANAGER && USB_DEVICE_STACK (value: y), and select condition IMG_MANAGER && USB_DEVICE_STACK (value: y)Parsing C:/Users/yongc/zephyrproject_master/projects/dfu/Kconfig
Loaded configuration 'C:/Users/yongc/GitLab/boards/boards/arm/ivm22_f103re/ivm22_f103re_defconfig'
Merged configuration 'C:/Users/yongc/zephyrproject_master/projects/dfu/prj.conf'
error: Aborting due to Kconfig warnings
```

**To reproduce**
Compile the sample for `waveshare_open103z`, which has usb defined in its dts.

**Additional context**
My STM32F1 custom board with custom firmware has USB DFU working using samples from ST. I'm guessing that USB DFU on Zephyr should be able to work without MPU?

ycsin, 2021-03-06:
The sample can be conpiled if `MPU_ALLOW_FLASH_WRITE ` is removed from USB DFU class's Kconfig. But dfu-util is not able to open the device after pluggin it into a Windows 10 machine.

ycsin, 2021-03-08:
After configuring these:

- `CONFIG_USB_DEVICE_VID=0x0483`
- `CONFIG_USB_DEVICE_PID=0xDF11`

It is recognized as "STM Device in DFU Mode" in Windows 10's device manager:
[image]

According to STM32 DFU example code and change `usb_dfu.h`'s `DFU_VERSION` to `0x011A` the device is now recognized by ST's DfuSe Demo software. But it says "Unable to find or decode device mapping... Bad Firmware"

[image]

I guess it is missing something like this?

```c
uint8_t DFU_StringInterface0[DFU_SIZ_STRING_INTERFACE0] =
  {
    DFU_SIZ_STRING_INTERFACE0,
    0x03,
    // Interface 0: "@Internal Flash   /0x08000000/06*002Ka,250*002Kg"
    '@', 0, 'I', 0, 'n', 0, 't', 0, 'e', 0, 'r', 0, 'n', 0, 'a', 0, 'l', 0,  /* 18 */
    ' ', 0, 'F', 0, 'l', 0, 'a', 0, 's', 0, 'h', 0, ' ', 0, ' ', 0, /* 16 */

    '/', 0, '0', 0, 'x', 0, '0', 0, '8', 0, '0', 0, '0', 0, '0', 0, '0', 0, '0', 0, '0', 0, /* 22 */

    '/', 0, '0', 0, '6', 0, '*', 0, '0', 0, '0', 0, '2', 0, 'K', 0, 'a', 0, /* 18 */
    ',', 0, '2', 0, '5', 0, '0', 0, '*', 0, '0', 0, '0', 0, '2', 0, 'K', 0, 'g', 0, /* 20 */
  };
```

ycsin, 2021-03-08:
This is the debug log output:
```
--- 1 messages dropped ---
[00:08:41.087,000] <dbg> usb_device.usb_handle_control_transfer: ep 0x00, status 0x00
--- 9 messages dropped ---
[00:08:41.087,000] <dbg> usb_dc_stm32.usb_dc_ep_read_wait: ep 0x00, 8 bytes, 0+8, 0x200027d8
[00:08:41.087,000] <dbg> usb_dc_stm32.usb_dc_ep_start_read: ep 0x00, len 64
[00:08:41.087,000] <dbg> usb_device.custom_handler: bRequest 0x06, wIndex 0x0000
[00:08:41.087,000] <dbg> usb_device.usb_handle_std_device_req: REQ_GET_DESCRIPTOR
[00:08:41.087,000] <dbg> usb_dc_stm32.usb_dc_ep_write: ep 0x80, len 27
[00:08:41.087,000] <dbg> usb_dc_stm32.usb_dc_ep_start_read: ep 0x80, len 0
[00:08:41.088,000] <dbg> usb_dc_stm32.HAL_PCD_DataInStageCallback: epnum 0x00
[00:08:41.088,000] <dbg> usb_device.usb_handle_control_transfer: ep 0x80, status 0x02
[00:08:41.090,000] <dbg> usb_dc_stm32.HAL_PCD_SetupStageCallback:
[00:08:41.090,000] <dbg> usb_device.usb_handle_control_transfer: ep 0x00, status 0x00
[00:08:41.090,000] <dbg> usb_dc_stm32.usb_dc_ep_read_wait: ep 0x00, 8 bytes, 0+8, 0x200027d8
[00:08:41.090,000] <dbg> usb_dc_stm32.usb_dc_ep_start_read: ep 0x00, len 64
[00:08:41.090,000] <dbg> usb_device.custom_handler: bRequest 0x06, wIndex 0x0000
[00:08:41.090,000] <dbg> usb_device.usb_handle_std_device_req: REQ_GET_DESCRIPTOR
[00:08:41.090,000] <dbg> usb_dc_stm32.usb_dc_ep_write: ep 0x80, len 9
[00:08:41.090,000] <dbg> usb_dc_stm32.usb_dc_ep_start_read: ep 0x80, len 0
[00:08:41.091,000] <dbg> usb_dc_stm32.HAL_PCD_DataInStageCallback: epnum 0x00
[00:08:41.091,000] <dbg> usb_device.usb_handle_control_transfer: ep 0x80, status 0x02
[00:08:41.093,000] <dbg> usb_dc_stm32.HAL_PCD_SetupStageCallback:
[00:08:41.093,000] <dbg> usb_device.usb_handle_control_transfer: ep 0x00, status 0x00
[00:08:41.093,000] <dbg> usb_dc_stm32.usb_dc_ep_read_wait: ep 0x00, 8 bytes, 0+8, 0x200027d8
[00:08:41.093,000] <dbg> usb_dc_stm32.usb_dc_ep_start_read: ep 0x00, len 64
[00:08:41.093,000] <dbg> usb_device.custom_handler: bRequest 0x06, wIndex 0x0000
[00:08:41.093,000] <dbg> usb_device.usb_handle_std_device_req: REQ_GET_DESCRIPTOR
[00:08:41.093,000] <dbg> usb_dc_stm32.usb_dc_ep_write: ep 0x80, len 27
[00:08:41.093,000] <dbg> usb_dc_stm32.usb_dc_ep_start_read: ep 0x80, len 0
[00:08:41.094,000] <dbg> usb_dc_stm32.HAL_PCD_DataInStageCallback: epnum 0x00
[00:08:41.094,000] <dbg> usb_device.usb_handle_control_transfer: ep 0x80, status 0x02
```

erwango, 2021-04-16:
To sum up the question: Could USB_DFU be supported w/o MPU support ?

@jfischer-no any insight ?

jfischer-no, 2021-04-16:
> To sum up the question: Could USB_DFU be supported w/o MPU support ?
> 
> @jfischer-no any insight ?

Yes. But I do not have an overview of what MPU_ALLOW_FLASH_WRITE has for hidden effects.
I guess `select MPU_ALLOW_FLASH_WRITE if ARM_MPU` or `imply MPU_ALLOW_FLASH_WRITE` could resolve this issue.

```diff c
diff --git a/subsys/usb/class/dfu/Kconfig b/subsys/usb/class/dfu/Kconfig
index 11a3fdacf2..f90e8ec348 100644
--- a/subsys/usb/class/dfu/Kconfig
+++ b/subsys/usb/class/dfu/Kconfig
@@ -3,7 +3,7 @@
 
 menuconfig USB_DFU_CLASS
        bool "USB DFU Class Driver"
-       select MPU_ALLOW_FLASH_WRITE
+       select MPU_ALLOW_FLASH_WRITE if ARM_MPU
        select POLL
        depends on IMG_MANAGER
        select IMG_ERASE_PROGRESSIVELY if SOC_FLASH_NRF
```

erwango, 2021-04-16:
> Yes. But I do not have an overview of what MPU_ALLOW_FLASH_WRITE has for hidden effects.
I guess select MPU_ALLOW_FLASH_WRITE if ARM_MPU or imply MPU_ALLOW_FLASH_WRITE could resolve this issue.

Great. @ycsin  Can you have a try and report back ?

ycsin, 2021-04-16:
> > Yes. But I do not have an overview of what MPU_ALLOW_FLASH_WRITE has for hidden effects.
> > I guess select MPU_ALLOW_FLASH_WRITE if ARM_MPU or imply MPU_ALLOW_FLASH_WRITE could resolve this issue.
> 
> Great. @ycsin Can you have a try and report back ?

Is this patch different from manually removing `MPU_ALLOW_FLASH_WRITE` in terms of the functionality? I think the ST's dfuse software and the dfu-util still won't work.

erwango, 2021-05-28:
@ycsin Do you have a new status on this point ?

ycsin, 2021-05-28:
@erwango Nope.. This has became a 'good to have' feature in my current priority list (for F103) and I'm currently migrating from F103 to G0B1.

erwango, 2021-05-28:
@ycsin Ok, when you have time please report back of the proposed solution is working.

closed 2021-08-11 by github-actions[bot]