zephyrproject-rtos/zephyr #32798: west flash fails for reel board

kydos, 2021-03-02:
I have a brand new reel-board from Phytec, but apparently I can't manage to flash it. The problem is simple to reproduce on a clean installation of zephyr on Ubuntu 20.10 (groovy). I can flash no issue an STM32 Discovery and an Olimex Board. But no chance with the reel-board, which is the board displayed on your documentation.
Below I describe what I did using as target reel_board, the same error occurs when using reel_board2.

Thanks in advance for any insights.

```
$ west build -p auto -b reel_board samples/basic/blinky
$ west flash
- west flash: rebuilding
[1/2] Linking C executable zephyr/zephyr.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:       19584 B         1 MB      1.87%
            SRAM:        4392 B       256 KB      1.68%
        IDT_LIST:          0 GB         2 KB      0.00%

[1/2] cd /home/kydos/zephyrproject/zephyr/build/zephyr/cmake/flash && /usr/bin/cmake -E echo

-- west flash: using runner pyocd
-- runners.pyocd: Flashing file: /home/kydos/zephyrproject/zephyr/build/zephyr/zephyr.hex
0000429:CRITICAL:__main__:No ACK received
Traceback (most recent call last):
  File "/home/kydos/.local/lib/python3.8/site-packages/pyocd/probe/cmsis_dap_probe.py", line 250, in read_dp
    result = self._link.read_reg(reg_id, now=now)
  File "/home/kydos/.local/lib/python3.8/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 853, in read_reg
    return read_reg_cb()
  File "/home/kydos/.local/lib/python3.8/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 847, in read_reg_cb
    res = transfer.get_result()
  File "/home/kydos/.local/lib/python3.8/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 144, in get_result
    self.daplink.flush()
  File "/home/kydos/.local/lib/python3.8/site-packages/pyocd/utility/concurrency.py", line 28, in _locking
    return func(self, *args, **kwargs)
  File "/home/kydos/.local/lib/python3.8/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 672, in flush
    self._read_packet()
  File "/home/kydos/.local/lib/python3.8/site-packages/pyocd/utility/concurrency.py", line 28, in _locking
    return func(self, *args, **kwargs)
  File "/home/kydos/.local/lib/python3.8/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 932, in _read_packet
    decoded_data = cmd.decode_data(raw_data)
  File "/home/kydos/.local/lib/python3.8/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 450, in decode_data
    data = self._decode_transfer_block_data(data)
  File "/home/kydos/.local/lib/python3.8/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 419, in _decode_transfer_block_data
    self._check_response(data[3])
  File "/home/kydos/.local/lib/python3.8/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 342, in _check_response
    raise DAPAccessIntf.TransferError("No ACK received")
pyocd.probe.pydapaccess.dap_access_api.DAPAccessIntf.TransferError: No ACK received

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/kydos/.local/lib/python3.8/site-packages/pyocd/__main__.py", line 401, in run
    self._COMMANDS[self._args.cmd](self)
  File "/home/kydos/.local/lib/python3.8/site-packages/pyocd/__main__.py", line 571, in do_flash
    with session:
  File "/home/kydos/.local/lib/python3.8/site-packages/pyocd/core/session.py", line 343, in __enter__
    self.open()
  File "/home/kydos/.local/lib/python3.8/site-packages/pyocd/core/session.py", line 461, in open
    self._board.init()
  File "/home/kydos/.local/lib/python3.8/site-packages/pyocd/board/board.py", line 85, in init
    self.target.init()
  File "/home/kydos/.local/lib/python3.8/site-packages/pyocd/core/soc_target.py", line 117, in init
    seq.invoke()
  File "/home/kydos/.local/lib/python3.8/site-packages/pyocd/utility/sequencer.py", line 219, in invoke
    resultSequence.invoke()
  File "/home/kydos/.local/lib/python3.8/site-packages/pyocd/utility/sequencer.py", line 214, in invoke
    resultSequence = call()
  File "/home/kydos/.local/lib/python3.8/site-packages/pyocd/coresight/dap.py", line 374, in _connect
    connector.connect(self._protocol)
  File "/home/kydos/.local/lib/python3.8/site-packages/pyocd/coresight/dap.py", line 166, in connect
    self._connect_dp(protocol)
  File "/home/kydos/.local/lib/python3.8/site-packages/pyocd/coresight/dap.py", line 206, in _connect_dp
    self._idr = self.read_idr()
  File "/home/kydos/.local/lib/python3.8/site-packages/pyocd/coresight/dap.py", line 233, in read_idr
    dpidr = self._probe.read_dp(DP_IDR, now=True)
  File "/home/kydos/.local/lib/python3.8/site-packages/pyocd/probe/cmsis_dap_probe.py", line 252, in read_dp
    six.raise_from(self._convert_exception(error), error)
  File "<string>", line 3, in raise_from
pyocd.core.exceptions.TransferError: No ACK received
FATAL ERROR: command exited with status 1: pyocd flash -e sector -t nrf52840 -f 4000000 /home/kydos/zephyrproject/zephyr/build/zephyr/zephyr.hex

jfischer-no, 2021-03-04:
@jremmert-phytec-iot :arrow_up:

jfischer-no, 2021-03-04:
It is not Zephyr issue. Either you did not turn on your board or you use the wrong USB connector or the board is broken.

kydos, 2021-03-04:
Thanks for your swift reply @jfischer-no, we've just solved the problem and as you point out it was  not depending on zephyr. We are porting https://github.com/eclipse-zenoh/zenoh-pico on zephyr, will inform the community once available. Thanks again for your help and kudos for the great work on Zephyr.

closed 2021-03-04 by kydos