zephyrproject-rtos/zephyr #32800: Race conditions with setting thread attributes after `z_ready_thread`?

jharris-intel, 2021-03-02:
While trying to optimize `k_sem`, I realized that different portions of the codebase are inconsistent in how they handle `z_ready_thread` and `z_reschedule`.

In particular, with the following pattern:

```c
thread = z_unpend_first_thread(...)
if (thread) {
	z_ready_thread(thread);
	[...] /* HERE IS WHERE I'M INTERESTED IN */
	z_reschedule(...)
```

Can a thread be run after the `z_ready_thread` call but before the `z_reschedule` call? (I'm thinking particularly of SMP here.)

If it cannot, there are a couple of optimizations that can be done here.

If it can, several portions of the codebase seem to need fixing, for instance `pthread_mutex_unlock`:

```c
		thread = z_unpend_first_thread(&m->wait_q);
		if (thread) {
			m->owner = (pthread_t)thread;
			m->lock_count++;
			z_ready_thread(thread);
			arch_thread_return_value_set(thread, 0);
			z_reschedule_irqlock(key);
```
If the thread can run after `z_ready_thread` but before `z_reschedule_irqlock`, then it can potentially see a stale return value, or have a stale return value set.

Ditto `z_impl_k_stack_push`:

```c
	first_pending_thread = z_unpend_first_thread(&stack->wait_q);

	if (first_pending_thread != NULL) {
		z_ready_thread(first_pending_thread);

		z_thread_return_value_set_with_data(first_pending_thread,
						   0, (void *)data);
		z_reschedule(&stack->lock, key);
```

Ditto `z_impl_k_futex_wake`:

```c
	do {
		thread = z_unpend_first_thread(&futex_data->wait_q);
		if (thread) {
			z_ready_thread(thread);
			arch_thread_return_value_set(thread, 0);
			woken++;
		}
	} while (thread && wake_all);

	z_reschedule(&futex_data->lock, key);
```

jharris-intel, 2021-03-02:
@andyross - I'd be interested in your thoughts on this, as this is "gnarly scheduler stuff" :-)

andyross, 2021-03-02:
Yes, once read_thread() returns the thread is in the queue and available to run at any time by this CPU or another.  The reschedule() call is just a cooperative point at which you can release a lock and allow a context switch.  And indeed, those spots in the IPC layer look wrong to me too.

(The posix layer using legacy irqlocks is something that should be fixed too, I think I might have mentioned that at one point)

jharris-intel, 2021-03-02:
Alright, that's what I thought; thanks for the doublecheck. I'll do up a PR to fix those cases then...

PR #32802 (2021-03-03): kernel: fix race conditions with z_ready_thread
Several internal APIs wrote thread attributes (return value, mainly)
_after_ calling `z_ready_thread`. This is unsafe, at least in SMP,
because another core could have already picked up and run the thread.

Fixes #32800.
commit c7bb423f (2021-03-03): kernel: fix race conditions with z_ready_thread

Several internal APIs wrote thread attributes (return value, mainly)
_after_ calling `z_ready_thread`. This is unsafe, at least in SMP,
because another core could have already picked up and run the thread.

Fixes #32800.
closed 2021-03-03 by nashif