zephyrproject-rtos/zephyr #32809: Fail to build ARC zephyr with MetaWare toolchain

qipengzha, 2021-03-03:
**Describe the bug**
I got build failure when try to build arc nsim_em board with MetaWare toolchain which installed by mw_devkit_arc_Q_2020_06_linux_x64_install.bin
Could you help check where is wrong ?  below is the detail log.
Thanks.

**To Reproduce**
Steps to reproduce the behavior:

mike@mikezha2$ echo $ZEPHYR_TOOLCHAIN_VARIANT
arcmwdt
 
mike@mikezha2$ echo $ARCMWDT_TOOLCHAIN_PATH
/home/mike/metaware/ARC/
 
mike@mikezha2$ west build -b nsim_em samples/hello_world/
-- west build: generating a build system
Including boilerplate (Zephyr base): /home/mike/work/mcf/zephyr/zephyrproject/zephyr/cmake/app/boilerplate.cmake
-- Application: /home/mike/work/mcf/zephyr/zephyrproject/zephyr/samples/hello_world
-- Zephyr version: 2.4.99 (/home/mike/work/mcf/zephyr/zephyrproject/zephyr)
-- Found Python3: /usr/bin/python3.6 (found suitable exact version "3.6.9") found components:  Interpreter
-- Found west (found suitable version "0.8.0", minimum required is "0.7.1")
-- Board: nsim_em
-- Cache files will be written to: /home/mike/.cache/zephyr
-- Found toolchain: zephyr (/opt/zephyr-sdk)
-- Found dtc: /opt/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: /home/mike/work/mcf/zephyr/zephyrproject/zephyr/boards/arc/nsim/nsim_em.dts
-- Generated zephyr.dts: /home/mike/work/mcf/zephyr/zephyrproject/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /home/mike/work/mcf/zephyr/zephyrproject/zephyr/build/zephyr/include/generated/devicetree_unfixed.h
Parsing /home/mike/work/mcf/zephyr/zephyrproject/zephyr/Kconfig
Loaded configuration '/home/mike/work/mcf/zephyr/zephyrproject/zephyr/boards/arc/nsim/nsim_em_defconfig'
Merged configuration '/home/mike/work/mcf/zephyr/zephyrproject/zephyr/samples/hello_world/prj.conf'
Configuration saved to '/home/mike/work/mcf/zephyr/zephyrproject/zephyr/build/zephyr/.config'
Kconfig header saved to '/home/mike/work/mcf/zephyr/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
-- The ASM compiler identification is unknown
-- Found assembler: /home/mike/metaware/ARC/MetaWare/arc/bin/ccac
-- Warning: Did not find file Compiler/-ASM
CMake Error at ../../cmake/extensions.cmake:1571 (message):
  Assertion failed: The toolchain is unable to build a dummy C file.  See
  CMakeError.log.
Call Stack (most recent call first):
  ../../CMakeLists.txt:42 (assert)
 
 
-- Configuring incomplete, errors occurred!
See also "/home/mike/work/mcf/zephyr/zephyrproject/zephyr/build/CMakeFiles/CMakeOutput.log".
See also "/home/mike/work/mcf/zephyr/zephyrproject/zephyr/build/CMakeFiles/CMakeError.log".
FATAL ERROR: command exited with status 1: /opt/cmake-3.14.2-Linux-x86_64/bin/cmake -DWEST_PYTHON=/usr/local/bin/python3.6 -B/home/mike/work/mcf/zephyr/zephyrproject/zephyr/build -S/home/mike/work/mcf/zephyr/zephyrproject/zephyr/samples/hello_world -GNinja -DBOARD=nsim_em
 
mike@mikezha2$ ccac
DesignWare ARC C/C++ Compiler Q-2020.06 (build 005) (LLVM 9.0.1)
(c) Copyright 2013-2020 Synopsys Corporation
No files specified. Specify -help for usage information
 
mike@mikezha2$ whereis ccac
ccac: /home/mike/metaware/ARC/MetaWare/arc/bin/ccac
 
mike@mikezha2$ ls /home/mike/metaware/ARC/MetaWare/arc/bin/
a4deadbeef.out  ces.css             empty.av2.be     ldarc                     libdpfpfast_arc600.so  libmwextxmac.so           libtermide.so          mdb.cnf      sc.cnf
aoma            cgen.js             empty_trace.bin  libabridge.so             libdpfpfast_arc700.so  libmwextxy3.so            libtermsim.so          mrg@l386com  sc.jar
aomc            clangac             FileCheck        libacsi.so                libdsp_packa.so        libmwextxy4.so            libtrace_as_target.so  nmac         scopts.hlp
aomt            cld                 findhelp         libapidglnt.so            libflash.so            libmwextxy.so             libuix.so              nmarc        sizeac
arac            cond_defs_hs.txt    glprop.so        libapidglnt_umr.so        libfreertos.so         libmwextzn.so             libvperip.so           opcpre.pt    sizearc
ararc           cond_defs.txt       gpio_drivers     libapizebu.so             libhwcounters.so       libmwmemmod.so            libvrd.so              opc.pt       stripac
arcdsp.s        cond_defs_v2.txt    hcac             libarcchan.so             libicm.so              libmwxmacd.so             libxcam_counters.so    optac        striparc
arcexlib.s      deadbeef.out        hcac.cnf         libarccli.so              libmmud.so             libmwxydisp.so            libxmldisp.so          perfhtml     tcfgen
arcsimd.s       demo                hcac.hlp         libarchand.so             libmmudv3.so           libparc.so                llcac                  pif          tcftool
arcsrv          dom4j.jar           hcac@l3862com    libarchw.so               libmmudv4.so           libpdr.so                 lldac                  plastic.jar  termsimsrvr
arcvdsp3.s      dom4j_license.html  hcansip.pt       libarcmcd.so              libmpfifo.so           libplotter.so             llvm-asac              plotter      trace_to_replay
arcvdsp4.s      dummy.o             hcansi.pt        libarcmmu.so              libmwcached.so         libprofmm.so              llvm-cov               profac       unmangle
arcvdsp.s       dummy.s             hcarc            libarcmmuv3.so            libmwextcrc.so         libprof.so                llvm-linkac            profarc      vinstr.s
asac            eccgen              hcarc.cnf        libarc-verilator.so       libmwextdvbf.so        librascalint.so           llvm-profdata          rig          vraptor.s
asarc           elf2bin             hcarc.hlp        libarc-vtoc.so            libmwextea.so          libscdrvext.so            mcc                    runac        xferac
bcr.mdb         elf2hex             hc@l3861com      libas221.so               libmwextlib.so         libsimd.so                mcc.cnf                runarc       xferarc
blast           elfdump             hcopts.hlp       libbridge.so              libmwextmmac.so        libsmp.so                 mcc.hlp                scac         xvcg
blastarc        elfdumpac           inl@l386com      libcfread.so              libmwextmul.so         libspfpcompact_arc600.so  mdb                    scac.cnf
bplist          elfdumparc          ip_library       libcfwrite.so             libmwextslide.so       libspfpcompact_arc700.so  mdb32                  scanmp
bpmerge         empty.ac            jligen           libcrout.so               libmwexttimer.so       libspfpfast_arc600.so     mdb32.cnf              ScanMP
buildlib        empty.ac.be         js.jar           libdpfpcompact_arc600.so  libmwextv2.so          libspfpfast_arc700.so     mdb64                  scarc
ccac            empty.av2           ldac             libdpfpcompact_arc700.so  libmwextxmac3.so       libtadsc.so               mdb64.cnf              scarc.cnf
 

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**
Add any other context about the problem here.

nashif, 2021-03-03:
@abrodkin  @ruuddw FYI

nashif, 2021-03-03:
@qipengzha  can you please provide the contents of `/build/CMakeFiles/CMakeOutput.log` and `build/CMakeFiles/CMakeError.log`.

qipengzha, 2021-03-04:
mike@mikezha2$ ccac test.c
[ccac] (Warning)
****************************************************************
* Warning: building for the ARC EM processor with no extension *
* options may result in inferior code size and performance.    *
*                                                              *
* You may wish to try the -emsize or -emspeed options, or see  *
* the MetaWare Development Toolkit Getting Started guide.      *
*                                                              *
* To compile for a base-case EM processor and eliminate this   *
* warning use the -Xbasecase option.                           *
****************************************************************

DesignWare ARC C/C++ Compiler Q-2020.06 (build 005) (LLVM 9.0.1)
(c) Copyright 2013-2020 Synopsys Corporation
MetaWare ARC Assembler Q-2020.06 (build 005)
(c) Copyright 1996-2020, Synopsys, Inc.
MetaWare Linker (ELF/ARCompact2) Q-2020.06 (build 005)
(c) Copyright 1995-2020, Synopsys, Inc.
Unresolved Symbol:                      Referenced from:
------------------                      ----------------
main                                     /home/mike/metaware/ARC/MetaWare/arc/lib/av2em/le/crt1.o

qipengzha, 2021-03-04:
[CMakeOutput.log](https://github.com/zephyrproject-rtos/zephyr/files/6080431/CMakeOutput.log)
[CMakeError.log](https://github.com/zephyrproject-rtos/zephyr/files/6080432/CMakeError.log)

evgeniy-paltsev, 2021-03-04:
Hi @qipengzha, could you please specify the Zephyr version you are using (commit SHA or Version used)?

nashif, 2021-03-04:
> Hi @qipengzha, could you please specify the Zephyr version you are using (commit SHA or Version used)?

in the log above it says
```
Zephyr version: 2.4.99 
```

ruuddw, 2021-03-04:
Thanks for the ccac test.c test. That rules out license issues.

evgeniy-paltsev, 2021-03-04:
> Zephyr version: 2.4.99 

As I understand that number cover almost all 2.5 development process - from the `2.4.0` release to the `2.5.0-rc1`.
Given that it would be more useful to know exact commit which was used for building.

evgeniy-paltsev, 2021-03-04:
PS: 
I can't reproduce it when building Zephyr from 1463596205 (one commit before `2.5.0-rc1` - so we get `zephyr version: 2.4.99`) with 2020.06 MWDT.

```
$ west build -b nsim_em samples/hello_world/
-- west build: generating a build system
Including boilerplate (Zephyr base): /home/paltsev/CodeData/zephyr/zephyr-arc-64/zephyr/cmake/app/boilerplate.cmake
-- Application: /home/paltsev/CodeData/zephyr/zephyr-arc-64/zephyr/samples/hello_world
-- Zephyr version: 2.4.99 (/home/paltsev/CodeData/zephyr/zephyr-arc-64/zephyr)
-- Found Python3: /usr/bin/python3.7 (found suitable exact version "3.7.7") found components: Interpreter 
-- Found west (found suitable version "0.7.2", minimum required is "0.7.1")
-- Board: nsim_em
-- Cache files will be written to: /home/paltsev/.cache/zephyr
-- Found dtc: /usr/bin/dtc (found suitable version "1.4.7", minimum required is "1.4.6")
-- Found BOARD.dts: /home/paltsev/CodeData/zephyr/zephyr-arc-64/zephyr/boards/arc/nsim/nsim_em.dts
-- Generated zephyr.dts: /home/paltsev/CodeData/zephyr/zephyr-arc-64/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /home/paltsev/CodeData/zephyr/zephyr-arc-64/zephyr/build/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: /home/paltsev/CodeData/zephyr/zephyr-arc-64/zephyr/build/zephyr/include/generated/device_extern.h
Parsing /home/paltsev/CodeData/zephyr/zephyr-arc-64/zephyr/Kconfig
Loaded configuration '/home/paltsev/CodeData/zephyr/zephyr-arc-64/zephyr/boards/arc/nsim/nsim_em_defconfig'
Merged configuration '/home/paltsev/CodeData/zephyr/zephyr-arc-64/zephyr/samples/hello_world/prj.conf'
Configuration saved to '/home/paltsev/CodeData/zephyr/zephyr-arc-64/zephyr/build/zephyr/.config'
Kconfig header saved to '/home/paltsev/CodeData/zephyr/zephyr-arc-64/zephyr/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is Clang 9.0.1
-- The CXX compiler identification is Clang 9.0.1
-- The ASM compiler identification is unknown
-- Found assembler: /home/paltsev/CodeData/debug/metaware/metaware_2020_06/MetaWare/arc/bin/ccac
-- Warning: Did not find file Compiler/-ASM
CMake Warning at ../../cmake/linker/arcmwdt/target.cmake:137 (message):
  MWDT toolchain does not support

             CONFIG_LINKER_ORPHAN_SECTION_WARN
Call Stack (most recent call first):
  ../../CMakeLists.txt:242 (toolchain_ld_baremetal)

-- Configuring done
-- Generating done
-- Build files have been written to: /home/paltsev/CodeData/zephyr/zephyr-arc-64/zephyr/build
-- west build: building application
[1/119] Preparing syscall dependency handling

[116/119] Generating isr_tables.c
MetaWare Hex File Generator Q-2020.06 (build 005)
(c) Copyright 1995-2020, Synopsys, Inc.
[elf2hex] (Warning) Emitting padding from 0x29e3 to 0x80001920
[elf2hex] FILL LIMIT EXCEEDED BETWEEN ADDRESS 0x29e3 and 0x80001920

[elf2hex] (Warning) Emitting padding from 0x800019a8 to 0xfffff800
[elf2hex] FILL LIMIT EXCEEDED BETWEEN ADDRESS 0x800019a8 and 0xfffff800

[119/119] Linking C executable zephyr/zephyr.elf
text	data	bss	dec	hex	filename
291	10568	0	10859	2a6b	zephyr.elf
```

nashif, 2021-03-04:
> As I understand that number cover almost all 2.5 development process - from the `2.4.0` release to the `2.5.0-rc1`.
> Given that it would be more useful to know exact commit which was used for building.

yes, you are correct. @qipengzha  maybe you should provide the commit you are using.

qipengzha, 2021-03-05:
The HEAD commit is 19c2f784959447b3e68a318cfcfdaba29c83abf0, zephyr version is 2.4.99
thanks.

nashif, 2021-03-22:
@evgeniy-paltsev any updates? Do you need any additional information?

ruuddw, 2021-03-25:
It looks the problem is even before starting to compile Zephyr, in the cmake configuration/discovery of the tools. More details might show in the cmake output log, that give a hint on what's wrong?

nashif, 2021-04-07:
the log files are already attach in https://github.com/zephyrproject-rtos/zephyr/issues/32809#issuecomment-790232747

ruuddw, 2021-04-07:
Hmm, it doesn't make sense to me. The CmakeOutput.log says the C and C++ compilers are succesfully recognized as CLang, but the west log says "-- The C compiler identification is unknown". Comparing to Evgeniy's log, there C compiler is recognized "-- The C compiler identification is Clang 9.0.1". I'm using a later Metaware version and see Clang 10.0.1 being recognized.

IRISZZW, 2021-04-20:
@qipengzha I notice that you use a old version cmake(14.2), Can you upgrade it to the newest version and try again. this may help.

nashif, 2021-04-20:
btw, one issue debugging this type of problem is the cache cmake creates, so sometimes we do not get all the errors because of that, it is advised to remove all cache files and then provide the files. The file to remove is `~/.cache/zephyr/ToolchainCapabilityDatabase/` on Linux.

@tejlmand  we need a better way to disable cache for such cases...

evgeniy-paltsev, 2021-04-21:
Just for a record - I've tested build from exact same commit (19c2f78) and with exact same MWDT version (2020.06) and I can't reproduce this. 
Given the old cmake usage and strange symptoms in logs (incorrect compiler detection) probably it worth to try it with newer cmake (I've tested it on 3.17.2) and also remove cmake cache as nashif proposed.

tejlmand, 2021-04-22:
> @qipengzha I notice that you use a old version cmake(14.2), Can you upgrade it to the newest version and try again. this may help.

This would be a very good idea, as newer CMake's has better handling of toolchains.
There is also a possibility that your Toolchain capabilities db might have been setup at an earlier stage where other things could be wrong (like license check), in which case certain flags could be marked as not working.
So it could also be a good idea to clean the db.
To clear it, run: `rm -rf ~/.cache/zephyr/ToolchainCapabilityDatabase/`

qipengzha, 2021-04-23:
Great thanks all your support.
it works now after try to get log or clean cmake caches,  i'm not very sure what exactly make the change.
will try some more test by change build env.

arcmwdt
/home/mike/metaware/ARC/
-- west build: generating a build system
Including boilerplate (Zephyr base): /home/mike/work/mcf/zephyr/zephyrproject/zephyr/cmake/app/boilerplate.cmake
-- Application: /home/mike/work/mcf/zephyr/zephyrproject/zephyr/samples/hello_world
-- Zephyr version: 2.4.99 (/home/mike/work/mcf/zephyr/zephyrproject/zephyr)
-- Found Python3: /usr/bin/python3.6 (found suitable exact version "3.6.9") found components:  Interpreter
-- Found west (found suitable version "0.8.0", minimum required is "0.7.1")
-- Board: nsim_em
-- Cache files will be written to: /home/mike/.cache/zephyr
-- Found toolchain: zephyr (/opt/zephyr-sdk)
-- Found dtc: /opt/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: /home/mike/work/mcf/zephyr/zephyrproject/zephyr/boards/arc/nsim/nsim_em.dts
-- Generated zephyr.dts: /home/mike/work/mcf/zephyr/zephyrproject/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /home/mike/work/mcf/zephyr/zephyrproject/zephyr/build/zephyr/include/generated/devicetree_unfixed.h
Parsing /home/mike/work/mcf/zephyr/zephyrproject/zephyr/Kconfig
Loaded configuration '/home/mike/work/mcf/zephyr/zephyrproject/zephyr/boards/arc/nsim/nsim_em_defconfig'
Merged configuration '/home/mike/work/mcf/zephyr/zephyrproject/zephyr/samples/hello_world/prj.conf'
Configuration saved to '/home/mike/work/mcf/zephyr/zephyrproject/zephyr/build/zephyr/.config'
Kconfig header saved to '/home/mike/work/mcf/zephyr/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is Clang 9.0.1
-- The CXX compiler identification is Clang 9.0.1
-- The ASM compiler identification is unknown
-- Found assembler: /home/mike/metaware/ARC/MetaWare/arc/bin/ccac
-- Warning: Did not find file Compiler/-ASM
CMake Warning at ../../cmake/linker/arcmwdt/target.cmake:138 (message):
  MWDT toolchain does not support

             CONFIG_LINKER_ORPHAN_SECTION_WARN
Call Stack (most recent call first):
  ../../CMakeLists.txt:242 (toolchain_ld_baremetal)

-- Configuring done
-- Generating done
-- Build files have been written to: /home/mike/work/mcf/zephyr/zephyrproject/zephyr/build
-- west build: building application
[1/118] Preparing syscall dependency handling

[2/118] Generating misc/generated/syscalls_subdirs.trigger
[3/118] Generating misc/generated/syscalls.json, misc/generated/struct_tags.json
[4/118] Generating include/generated/syscall_dispatch.c, include/generated/syscall_list.h
[5/118] Generating include/generated/kobj-types-enum.h, include/generated/otype-to-str.h
[6/118] Generating include/generated/driver-validation.h
[7/118] Building C object zephyr/CMakeFiles/offsets.dir/arch/arc/core/offsets/offsets.c.obj
[8/118] Generating include/generated/offsets.h
[9/118] Building C object zephyr/lib/libc/minimal/CMakeFiles/lib__libc__minimal.dir/source/string/strncasecmp.c.obj
[10/118] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/cbprintf.c.obj
[11/118] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/crc16_sw.c.obj
[12/118] Building C object CMakeFiles/app.dir/src/main.c.obj

...
[83/118] Building C object zephyr/kernel/CMakeFiles/kernel.dir/fatal.c.obj
[84/118] Building C object zephyr/kernel/CMakeFiles/kernel.dir/idle.c.obj
[85/118] Building C object zephyr/kernel/CMakeFiles/kernel.dir/init.c.obj
[86/118] Linking C static library zephyr/lib/libc/minimal/liblib__libc__minimal.a
[87/118] Linking C static library zephyr/drivers/serial/libdrivers__serial.a
[88/118] Building C object zephyr/kernel/CMakeFiles/kernel.dir/kheap.c.obj
[89/118] Building C object zephyr/arch/arch/arc/core/mpu/CMakeFiles/arch__arc__core__mpu.dir/arc_mpu.c.obj
[90/118] Building C object zephyr/kernel/CMakeFiles/kernel.dir/mem_slab.c.obj
[91/118] Building C object zephyr/kernel/CMakeFiles/kernel.dir/msg_q.c.obj
[92/118] Building C object zephyr/kernel/CMakeFiles/kernel.dir/mutex.c.obj
[93/118] Building C object zephyr/kernel/CMakeFiles/kernel.dir/pipes.c.obj
[94/118] Linking C static library zephyr/arch/arch/arc/core/mpu/libarch__arc__core__mpu.a
[95/118] Building C object zephyr/kernel/CMakeFiles/kernel.dir/queue.c.obj
[96/118] Building C object zephyr/kernel/CMakeFiles/kernel.dir/stack.c.obj
[97/118] Building C object zephyr/kernel/CMakeFiles/kernel.dir/sched.c.obj
[... 6 lines cut ...]
[104/118] Building C object zephyr/kernel/CMakeFiles/kernel.dir/timeout.c.obj
[105/118] Building C object zephyr/kernel/CMakeFiles/kernel.dir/atomic_c.c.obj
[106/118] Building C object zephyr/kernel/CMakeFiles/kernel.dir/timer.c.obj
[107/118] Building C object zephyr/kernel/CMakeFiles/kernel.dir/mailbox.c.obj
[108/118] Building C object zephyr/kernel/CMakeFiles/kernel.dir/sem.c.obj
[109/118] Building C object zephyr/arch/arch/arc/core/CMakeFiles/arch__arc__core.dir/vector_table.c.obj
[110/118] Linking C static library zephyr/arch/arch/arc/core/libarch__arc__core.a
[111/118] Building C object zephyr/kernel/CMakeFiles/kernel.dir/mempool.c.obj
[112/118] Linking C static library zephyr/kernel/libkernel.a
[113/118] Linking C executable zephyr/zephyr_prebuilt.elf
[114/118] Generating linker_pass_final.cmd
[115/118] Generating isr_tables.c
MetaWare Hex File Generator Q-2020.06 (build 005)
(c) Copyright 1995-2020, Synopsys, Inc.
[... cut ...]
[elf2hex] FILL LIMIT EXCEEDED BETWEEN ADDRESS 0x2a52 and 0x80001920

[elf2hex] (Warning) Emitting padding from 0x800019a8 to 0xfffff800
[elf2hex] FILL LIMIT EXCEEDED BETWEEN ADDRESS 0x800019a8 and 0xfffff800

[116/118] Building C object zephyr/CMakeFiles/zephyr_final.dir/misc/empty_file.c.obj
[117/118] Building C object zephyr/CMakeFiles/zephyr_final.dir/isr_tables.c.obj
[118/118] Linking C executable zephyr/zephyr.elf
text    data    bss     dec     hex     filename
394     10576   0       10970   2ada    zephyr.elf

ruuddw, 2021-04-23:
Thanks @qipengzha for confirming that clearing up your environment solved the problem. I close this issue, feel free to reopen if you run into the problem again - or create a new issue if it is a different problem.

closed 2021-04-23 by ruuddw

qipengzha, 2021-04-25:
Hi 
it works with zephyr 2.4.99. but got some errors(see below error log) related to link script on zephyr 2.5.0, I just WA by add some macros.

-- The C compiler identification is Clang 9.0.1
-- The CXX compiler identification is Clang 9.0.1
-- The ASM compiler identification is unknown
-- Found assembler: /home/mike/metaware/ARC/MetaWare/arc/bin/ccac
-- Warning: Did not find file Compiler/-ASM
CMake Warning at ../../cmake/linker/arcmwdt/target.cmake:137 (message):
  MWDT toolchain does not support

             CONFIG_LINKER_ORPHAN_SECTION_WARN
Call Stack (most recent call first):
  ../../CMakeLists.txt:242 (toolchain_ld_baremetal)

-- Configuring done
-- Generating done
-- Build files have been written to: /home/mike/work/mcf/code/fpga/zephyr/build
-- west build: building application
[1/124] Preparing syscall dependency handling

[119/124] Linking C executable zephyr/zephyr_prebuilt.elf
FAILED: zephyr/zephyr_prebuilt.elf zephyr/zephyr_prebuilt.map
: && ccache /home/mike/metaware/ARC/MetaWare/arc/bin/ccac    zephyr/CMakeFiles/zephyr_prebuilt.dir/misc/empty_file.c.obj  -o zephyr/zephyr_prebuilt.elf  -Wl,-T/home/mike/work/mcf/code/fpga/zephyr/build/zephyr/linker.cmd -Wl,--gc-sections -Wl,--entry=__start -Wl,--Map=/home/mike/work/mcf/code/fpga/zephyr/build/zephyr/zephyr_prebuilt.map -Wl,--whole-archive app/libapp.a zephyr/libzephyr.a zephyr/arch/common/libarch__common.a zephyr/arch/arch/arc/core/libarch__arc__core.a zephyr/arch/arch/arc/core/mpu/libarch__arc__core__mpu.a zephyr/lib/libc/minimal/liblib__libc__minimal.a zephyr/lib/posix/liblib__posix.a zephyr/drivers/serial/libdrivers__serial.a modules/intel/lib..__modules__hal__intel__zephyr.a -Wl,--no-whole-archive zephyr/kernel/libkernel.a zephyr/CMakeFiles/offsets.dir/arch/arc/core/offsets/offsets.c.obj -L/home/mike/work/mcf/code/fpga/zephyr/build/zephyr zephyr/arch/common/libisr_tables.a -Wl,--sort-section=alignment -Wl,-u_OffsetAbsSyms -Wl,-u_ConfigAbsSyms -Hlld -Hnosdata -Hnocrt -Xtimer0 -Hnoxcheck -Hnocplus -Hcl -Hheap=0 -Hnoivt && cd /home/mike/work/mcf/code/fpga/zephyr/build/zephyr && /opt/cmake-3.14.2-Linux-x86_64/bin/cmake -E echo
lldac: error: /home/mike/work/mcf/code/fpga/zephyr/build/zephyr/linker.cmd:200: ) expected, but got ,
>>>  } GROUP_ROM_LINK_IN(DCCM, ICCM)