zephyrproject-rtos/zephyr #32817: Supporting fedora in the getting started docs

HannanSolo, 2021-03-03:
It seems when I went through the getting started docs for 2.5.99 the only OS support for linux is ubuntu. Yet the 1.1.0 documentation did support fedora.

Just wanted to ask if its possible to use fedora for my Zephyr and if so add it to documentation.

galak, 2021-03-03:
> It seems when I went through the getting started docs for 2.5.99 the only OS support for linux is ubuntu. Yet the 1.1.0 documentation did support fedora.
> 
> Just wanted to ask if its possible to use fedora for my Zephyr and if so add it to documentation.

Its completely possible to use Zephyr on fedora, I do it daily.

If you look at the docs:

https://docs.zephyrproject.org/latest/getting_started/installation_linux.html#install-linux-host-dependencies

There should be a tab for fedora.

carlescufi, 2021-03-03:
Closing since this is already covered in the Getting Started Guide.

closed 2021-03-03 by carlescufi

HannanSolo, 2021-03-03:
Wow! This is great , thank you so much. I was following these docs https://docs.zephyrproject.org/latest/getting_started/index.html
wonder why its not listed there.

galak, 2021-03-03:
> Wow! This is great , thank you so much. I was following these docs https://docs.zephyrproject.org/latest/getting_started/index.html
> wonder why its not listed there.

There is a reference about other distros and a link in the docs.

`(instructions for other Linux distributions are discussed in Install Linux Host Dependencies)`