zephyrproject-rtos/zephyr #32818: Function z_swap_next_thread() missing coverage in sched.c

yingmingx, 2021-03-03:
Describe the bug
Function z_swap_next_thread() missing coverage in sched.c.

struct k_thread *z_swap_next_thread(void)
{
#ifdef CONFIG_SMP
	return next_up();
#else
	return _kernel.ready_q.cache;
#endif
}

When using qemu_x86, this line "return _kernel.ready_q.cache;" cannot be coveraged.  Function z_swap_next_thread only used in do_swap() which is called when enable CONFIG_USE_SWITCH.
 
To Reproduce
Steps to reproduce the behavior:
twister -p qemu_x86 --coverage -T tests/kernel

Expected behavior
It's probably dead code that should be removed.

Environment (please complete the following information):
OS: ubuntu1~18.04
Toolchain zephyr-sdk-0.12.1

nashif, 2021-03-03:
You should try qemu_x86_64

yingmingx, 2021-03-03:
> You should try qemu_x86_64

I've tried qemu_x86_64. The branch "return next_up();" was covered.  
My concern is branch "return _kernel.ready_q.cache;" is not covered.

nashif, 2021-03-03:
> My concern is branch "return _kernel.ready_q.cache;" is not covered.

also when you disable SMP?

nashif, 2021-03-03:
> Firstly, I saw our module coverage report which run command "twister -p qemu_x86 --coverage -T module-list -x=CONFIG_ASSERT=n -C". The else branch "return _kernel.ready_q.cache;" is not covered."
> Then, I run "twister -p qemu_x86_x64 --coverage -T tests/kernel/sched" to test SMP situation. The if branch "return next_up();" is covered.
> So I think the else branch is redundant. I'm not sure these tests are enough.

you should not edit comments, add your comment below.

nashif, 2021-03-03:
> hen, I run "twister -p qemu_x86_x64 --coverage -T tests/kernel/sched" to test SMP situation. The if branch "return next_up();" is covered.

you need to disable SMP to get to this, qemu_x86 and qemu_x86_64 use swap vs switch, so you have to test without SMP to get to this.

yingmingx, 2021-03-04:
Thank you very much for telling me the reason of this problem！
I run “twister -p qemu_x86_64 --coverage -T tests/kernel/sched/ -x=CONFIG_SMP=n ”, then the else branch "return _kernel.ready_q.cache;" is covered."
I'm sure this issue can be closed now.

closed 2021-03-04 by yingmingx