zephyrproject-rtos/zephyr #32822: Code doesn't compile after changing the PWM pin on example "blinky_pwm" on NRF52

selman-nus, 2021-03-03:
**Describe the bug**
I have successfully compiled and run the example (on nrf52DK) without modification and LED 1 is blinking as expected. 
However, when I try to use other LEDs instead of LED 1, it doesn't compile because `#if DT_NODE_HAS_STATUS(PWM_LED0_NODE, okay)` doesn't return true. Is this means I can only use one predefined LED for PWM? 
It just doesn't make sense. Any help much appreciated. 

**To Reproduce**
Steps to reproduce the behavior:
1. Open "blinky_pwm" example and run
2. Change pin from 17 to 18 like below code:
```
pwmleds {
		compatible = "pwm-leds";
		pwm_led0: pwm_led_0 {
			pwms = <&pwm0 18>;
		};
	};
```

3. Also change it from the definition:
```
&pwm0 {
	status = "okay";
	ch0-pin = <18>;
	ch0-inverted;
};
```
4. Run again and observe the error

**Expected behavior**
The second LED on the board would blink as it happened on LED 1 on unmodified code. 

**Impact**
I need to use other pins for PWM for my custom board and I still couldn't figure out how to change this default pin. Yeah, it is very annoying how simple the task sounds but how hard it to make it happen..

**Environment (please complete the following information):**
 - OS: ( Windows)
 - Toolchain (Zephyr SDK, Nordic toolchain manager, segger)

selman-nus, 2021-03-05:
Is there any other file than the board.dts file that I should be configuring?

anangl, 2021-03-05:
> Is there any other file than the board.dts file that I should be configuring?

I'm not sure which file you were actually modifying, but try just creating a file named `nrf52dk_nrf52832.overlay` with the following content:
```
&pwm_led0 {
	pwms = <&pwm0 18>;
};

&pwm0 {
	ch0-pin = <18>;
};
```
and place it in the main directory of the sample or in its subdirectory named `boards/`. Rebuild the sample from scratch. After flashing the board with it, you should see LED2 blinking instead of LED1.

selman-nus, 2021-03-05:
I created the file `nrf52dk_nrf52832.overlay` with the suggested content and copied it to many locations in the project folder.
But I still encounter the same problem. 

Maybe I am not able to rebuild the sample from scratch? 
Here is what I do to rebuild: 

1. - Clean zephyr/zephyr.elf
2. - Clean Solution
3. - Build and Run

[image]

moose-rivieh, 2021-04-20:
Any update on this?

closed 2021-07-05 by github-actions[bot]