zephyrproject-rtos/zephyr #32827: question: Specify size of malloc arena

maxbachmann, 2021-03-03:
I am using the Up Squared board with the following settings:
```
CONFIG_CPLUSPLUS=y
CONFIG_NEWLIB_LIBC=y
CONFIG_LIB_CPLUSPLUS=y
```
For k_malloc I can specify the memory size using:
```
CONFIG_HEAP_MEM_POOL_SIZE=207872000
```
However, I did not find a way to do the same for malloc.
```
CONFIG_MINIMAL_LIBC_MALLOC_ARENA_SIZE=207872000
```
works only when `MINIMAL_LIBC` is used. With no settings, I can only allocate 1mb of memory using malloc.
Is there a different config setting I have to use in my case?

dcpleung, 2021-03-03:
That should be `CONFIG_NEWLIB_LIBC_MAX_MAPPED_REGION_SIZE`.

maxbachmann, 2021-03-04:
Thanks, that's the setting I was searching.

closed 2021-03-04 by maxbachmann