zephyrproject-rtos/zephyr #32836: Remaining integration failures on intel_adsp_cavs15

andyross, 2021-03-03:
As of the (still unmerged) code in https://github.com/andyross/zephyr/tree/xtensa-wip the intel_adsp_cavs15 device now passes a full test run with twister.  But it contains this patch, which excludes the handful of known tests that are known problems, as a hung twister run often spoils the remaining tests (and occasionally requires a reboot of the CAVS device's host x86 system to recover!).

This bug is here to track these failures as they get fixed.  Right now these all have to be run manually:

> tests: samples: Add targetted platform_excludes for intel_adsp_cavs15
> 
> This platform has a tiny handful of remaining tests that fail.  We
> will track them as issues, but let's exclude them from integration
> testing to allow full runs to complete.  Often a hung device in one
> test will break an entire twister run.
> 
> Tests with known (and tracked) failures:
> 
>   samples/application_development/external_lib
>   samples/posix/eventfd
>   samples/userspace/hello_world_user
>   tests/kernel/fatal/message_capture
>   tests/net/socket/socketpair
>   tests/portability/cmsis_rtos_v2
> 
> These tests never fail in isolated testing, but are reliable timeouts
> when run in sequence in a big twister run.  It's possible that the bug
> here may be in twister or the flash/serial scripts:
> 
>   tests/crypto/tinycrypt
>   tests/subsys/logging/log_immediate
>   tests/subsys/logging/log_output

andyross, 2021-03-03:
@KangJianX note the test list here: when this code all gets merged, these tests are going to need to be manually verified.  Everything else should be covered (and **should** be passing!) via a default twister run.

andyross, 2021-03-03:
A few notes so I don't forget:

* The three samples/ produce an immediate device reset (and, occasionally, completely latched DSP!) when they are loaded.  We don't seem to get even as far as z_cstart().  And needless to say, none of these are doing anything weird at all.  This may be some kind of glitch in the build system (vs. some kconfig I'm unaware of) producing a corrupt image, I'm guessing?

* fatal/message_capture is a simple test that sets a fatal error handler, which works and is logged correctly, but control never returns to the main thread so test success is never flagged.  Which... is **correct** I believe?  It's a fatal error!  I need to study why this passes on other platforms and what promises the API is supposed to be making.  As of right now you can register a fatal error handler on Xtensa and reliably get it called on fatal errors, but once that function returns no application code is going to run.

* The cmsis failure is a spurious failure, maybe 30% of the time one of the test cases glitches out and hangs.  I haven't investigated further.

* The socketpair test is a reliable hang of the device in the network code somewhere.  If I had to guess, it's going to turn out to be a KERNEL_COHERENCE violation where cross-thread data (a buffer?) is living on a stack.  If so, and it can't be fixed, we might consider enforcing an antidependency between COHERENCE and NET and file an enhancement instead of a bug.  No existing products are expected to have both.

* The tinycrypt and log tests time out in twister very reliably, but **only** in a large run (i.e. not on their own).  I can't make them fail in any other circumstance.  I have a vague sense that this is some kind of race in the flashing code that shuts down the DSP prematurely for the next test, or something like that.  It feels more like a rig bug than a device bug.

enjiamai, 2021-03-08:
Hi @KangJianX , please notice those message and requirement from @andyross for ASDP.  Also FYI @LeiW000 .

KangJianX, 2021-03-11:
@andyross Should I run all tests on ADSP?

andyross, 2021-03-12:
Yes, please continue to run the test cases detailed in this bug as you have time available (but it's OK to presume they aren't fixed if nothing is committed claiming they have been fixed).  They aren't fixed, they're just excluded because with them in the suite the twister run gets stuck.

KangJianX, 2021-03-14:
You means if testcases run timeout with no output they need be exclude. Maybe I need report issues if testcases run failed with code error?

andyross, 2021-03-16:
@KangJianX : the rules aren't that weird:

1. Report any failures with any test that is not excluded.
2. Please periodically retest these tests manually (no need to be daily or anything) and report any notable differences in behavior from what is described above.

KangJianX, 2021-05-12:
@andyross I run those tests manually
samples/application_development/external_lib
samples/posix/eventfd
samples/userspace/hello_world_user
tests/kernel/fatal/message_capture
tests/net/socket/socketpair
And the behavior **same with you said in description**. 
BTW, tests/portability/cmsis_rtos_v2 **this test is deleted** by someone.

For 
tests/crypto/tinycrypt
tests/subsys/logging/log_immediate
tests/subsys/logging/log_output
**I run those tests together, and they all passed.**

andyross, 2021-05-12:
The cmsis test got moved to tests/subsys/portability/cmsis_rtos_v2.  We move stuff around a lot, but only rarely delete test code.

closed 2021-07-26 by github-actions[bot]