zephyrproject-rtos/zephyr #32853: uninitialized variable in this function

sinkyl, 2021-03-04:
**Describe the bug**
With Network, LwM2M and by default CONFIG_LWM2M_RW_JSON_SUPPORT=y in Zephyr 2.5 this warning show up:

` /zephyrproject/zephyr/subsys/net/lib/lwm2m/lwm2m_rw_json.c:147:16: warning: 'c' may be used uninitialized in this function [-Wmaybe-uninitialized]
  147 |  uint8_t cont, c;`

**Environment (please complete the following information):**
 - OS: Linux,
 - Toolchain: Zephyr SDK

rlubos, 2021-03-04:
How to reproduce this warning? I've tried to build with `zephyr-sdk-0.12.3` for x86 and arm platforms and I didn't get it.

Besides, it doesn't look like a valid warning to me. Looking at the code in https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/net/lib/lwm2m/lwm2m_rw_json.c#L144, the `c` variable can only be used after being set by the `buf_read_u8()` call. If the function call fails, the variable won't be used, otherwise it should hold a valid value.

sinkyl, 2021-03-04:
Try with:
`CONFIG_NEWLIB_LIBC=y` and
`CONFIG_DEBUG=y` -> just in case you don't have it...

rlubos, 2021-03-04:
Thanks, now I got it. I wonder how would these options affect the build to trigger the warning to show up...
Anyway, I still don't see anything wrong in the logic, so I'll just add some initial value to the variable to silence the warning.

rlubos, 2021-03-04:
I've submited a fix: https://github.com/zephyrproject-rtos/zephyr/pull/32882

sinkyl, 2021-03-04:
I don't know either...
We set the default value to `0U`, what you think ??? It should be fine no ?

rlubos, 2021-03-04:
> We set the default value to `0U`, what you think ??? It should be fine no ?

Yes, that's what I did as well

PR #32882 (2021-03-04): net: lwm2m: Initialize the variable to silence compiler warning
The compiler generates a warning regarding a variable being used
w/o being initialized in certian configuration. According to the logic
that's not the case, so just add some initial value to the variable to
silence the compiler.

Fixes #32853
closed 2021-03-04 by nashif