zephyrproject-rtos/zephyr #32860: Periodic advertising/synchronization on nRF52810

kaansenparlak, 2021-03-04:
I have been trying to implement periodic communication between nRF52840 (advertiser) and nRF52810 (synchroniser) but the synchronization is very fragile. Depending on the advertising period and PHY (1M/2M) sometimes the synchronization is never achieved in the first place. For example:

Period: 3s    PHY:   2M -> Rarely establishes a sync.
Period: 1s    PHY:   2M -> Works fine most of the time.
Period: 1s    PHY:   1M -> Almost never terminates sync.
Period: 3s    PHY:   1M -> Works fine most of the time.
Period: 10s  PHY:   1M -> Rarely establishes a sync.

Since LE Coded is not supported on nRF52810, it never works as expected.

To reproduce:

nRF Connect SDK v1.5.0 
zephyr->bluetooth->samples->periodic_adv (nRF52840) and periodic_sync (nRF52810)

The periodic data I send is 26 bytes. I tried increasing sync timeout to prevent sync termination but it did not work.

carlescufi, 2021-03-04:
@kaansenparlak if you are using nRF Connect SDK, then this is not the right place to post an issue. Instead, please open a Nordic Devzone ticket.

carlescufi, 2021-03-04:
Since you are using periodic advertisement, you are surely selecting the Zephyr open source controller. I will let @cvinayak and @Thalley comment.

cvinayak, 2021-03-08:
@kaansenparlak I suggest you to provide a branch (push a draft PR) with your code modification for the different cases you have tested. This will help me to reproduce your observations.

kaansenparlak, 2021-03-09:
Just made the pull request requested. The 2 board files that are added are the devices that I used. (nRF52840 (from nRF9160 DK) and nRF52810 (from a 3rd party))

cvinayak, 2021-03-14:
@kaansenparlak Thanks for your PR, I am suspecting that the missing time reservation calculations in the controller implementation (this is still work in progress) could be causing the overlapping radio events to be aborted. This could be verified if you have a Bluetooth Low Energy sniffer, which could show CRC errors (or incomplete PDUs) on the auxiliary and periodic advertising PDUs when such radio events aborts happen.

Please use this (temporary increased value, until the correct calculations based on the PDU content is implemented) in the below mentioned two places:
```
	slot_us += 3000;
```

https://github.com/zephyrproject-rtos/zephyr/blob/7874052df220caf53e9bb6b949f7e6a4c1903311/subsys/bluetooth/controller/ll_sw/ull_adv_aux.c#L837

and

https://github.com/zephyrproject-rtos/zephyr/blob/7874052df220caf53e9bb6b949f7e6a4c1903311/subsys/bluetooth/controller/ll_sw/ull_adv_sync.c#L387

cvinayak, 2021-03-15:
@kaansenparlak Please verify #33343 also.

PR #33343 (2021-03-30): Bluetooth: controller: Fix SyncInfo with correct event counter
Updated ticker implementation to return lazy value for a
ticker when enumerating active tickers with time
reservations.

This is required to find offsets and to use the lazy value
to correctly calculate event and payload counter values that
needs to be filled into SyncInfo and BigInfo structures.

Fix Periodic Advertising SyncInfo structure to be filled
with correct event counter value considering accumulated
latency and ticker lazy value.

Fixes #32860.
commit 0df70608 (2021-03-30): Bluetooth: controller: Fix SyncInfo with correct event counter

Fix Periodic Advertising SyncInfo structure to be filled
with correct event counter value considering accumulated
latency and ticker lazy value.

Fixes #32860.
closed 2021-03-30 by carlescufi