zephyrproject-rtos/zephyr #32862: MCUboot, use default UART_0 value for RECOVERY_UART_DEV_NAME

mejtejt, 2021-03-04:
### Problem description
I have two seperate applications. One is MCUboot and the other is the actual application. There is a single DTS, configured in application part as follows:
```
	chosen {
		zephyr,sram = &sram0;
		zephyr,flash = &flash0;
		zephyr,can-primary = &can1;
		zephyr,console = &usart3;
		zephyr,code-partition = &slot0_partition;
	};
&flash0 {
	/*
	 * For more information, see:
	 * http://docs.zephyrproject.org/latest/guides/dts/index.html#flash-partitions
	 */
	partitions {
		compatible = "fixed-partitions";
		#address-cells = <1>;
		#size-cells = <1>;
		
		/* 64KB for bootloader */
		boot_partition: partition@0 {
			label = "mcuboot";
			reg = <0x00000000 0x00010000>;
			read-only;
		};

		/* storage: 64KB for settings */
		storage_partition: partition@10000 {
			label = "storage";
			reg = <0x00010000 0x00010000>;
		};

		/* application image slot: 64KB */
		slot0_partition: partition@20000 {
			label = "image-0";
			reg = <0x00020000 0x00010000>;
		};

		/* backup slot: 64KB */
		slot1_partition: partition@30000 {
			label = "image-1";
			reg = <0x00030000 0x00010000>;
		};

		/* swap slot: 64KB */
		scratch_partition: partition@40000 {
			label = "image-scratch";
			reg = <0x00040000 0x00010000>;
		};

	};
};
```
MCU boot conf file for my board has the following inside:
```
CONFIG_CONSOLE=n
CONFIG_CONSOLE_HANDLER=n
CONFIG_UART_CONSOLE=n

CONFIG_MCUBOOT_SERIAL=y
CONFIG_BOOT_SERIAL_DETECT_PORT="GPIOD"
CONFIG_BOOT_SERIAL_DETECT_PIN=10
CONFIG_BOOT_SERIAL_DETECT_PIN_VAL=1
```

When I build MCU with this configuration, for CONFIG_RECOVERY_UART_DEV_NAME I get in boot/zephyr/build-mcuboot/zephyr:
CONFIG_RECOVERY_UART_DEV_NAME="UART_3"

And I am fine with that. What I want is the following: 
```
# Workaround for not being able to have commas in macro arguments
DT_CHOSEN_Z_CONSOLE := zephyr,console

config RECOVERY_UART_DEV_NAME
	string "UART Device Name for Recovery UART"
	default "$(dt_chosen_label,$(DT_CHOSEN_Z_CONSOLE))" if HAS_DTS
	default "UART_0"
	depends on BOOT_SERIAL_UART
	help
	  This option specifies the name of UART device to be used for
	  serial recovery.
```
I don't want RECOVERY_UART_DEV_NAME to be taken from device tree, I want it to get default value (UART_0). How do I achieve this? If I just comment out `zephyr,console = &usart3; ` from device tree, for CONFIG_RECOVERY_UART_DEV_NAME in boot/zephyr/build-mcuboot/zephyr I get:
`CONFIG_RECOVERY_UART_DEV_NAME="".`
Which is understandable, since there is no `zephyr, console` defined in device tree, but device tree is used. How do I modify this to say "do not use device tree for RECOVERY_UART_DEV_NAME, but use default value?"

Thanks for help.

galak, 2021-03-04:
Can you not set `CONFIG_RECOVERY_UART_DEV_NAME="UART_0"` in your prj.conf?

mejtejt, 2021-03-04:
I can, but for some reason, the update then does not work. I managed to circumvent this by defining separate overlay file for MCUboot, and set the wanted uart there. And then I can use different `zepyhr,console` in application and bootloader.

closed 2021-03-05 by mejtejt