zephyrproject-rtos/zephyr #32866: bt_le_ext_adv_create returns -5 with 0x2036 opcode status 1

ppluta, 2021-03-04:
**Describe the bug**
bt_le_ext_adv_create() returns -5 with "opcode 0x2036 status 0x1" log message. No parameter combination could be find for it to work.

**To Reproduce**
Just try own Zephyr sample "periodic_adv".

**Expected behavior**
It should work.

**Impact**
Renders the program useless.

**Logs and console output**
[00:00:44.983,123] <wrn> bt_hci_core: opcode 0x2036 status 0x01

**Environment (please complete the following information):**
Zephyr SDK 2.5.99 west updated today.

**Additional context**
No. It just fails.

joerchan, 2021-03-04:
Status 0x01 is unknown command. Which board did you try to build the sample for?
Anything special in the configuration used?

ppluta, 2021-03-05:
The board was Particle Xenon. However there is already a solution: CONFIG_BT_CTLR_ADV_EXT was missing in cfg.prj.
On the other hand replacing bt_le_adv_start() with bt_le_ext_adv_start() appeared unstable and throws this:

00> ASSERTION FAIL [id != ((uint8_t)((uint8_t)0 - 1))] @ WEST_TOPDIR/zephyr/subsys/bluetooth/controller/ll_sw/ull_adv_aux.c:1082
00> 
00> [00:19:45.391,143] <err> os: r0/a1:  0x00000003  r1/a2:  0x20004e00  r2/a3:  0x200008a0
00> [00:19:45.391,143] <err> os: r3/a4:  0x00000003 r12/ip:  0xa0000000 r14/lr:  0x0000b047
00> [00:19:45.391,174] <err> os:  xpsr:  0x4100001b
00> [00:19:45.391,174] <err> os: Faulting instruction address (r15/pc): 0x0000b052
00> [00:19:45.391,174] <err> os: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0
00> [00:19:45.391,174] <err> os: Fault during interrupt handling
00> 
00> [00:19:45.391,174] <err> os: Current thread: 0x20000920 (unknown)
00> [00:19:45.663,665] <err> os: Halting system

The code running is a loop consisting of 
{	struct bt_le_ext_adv_start_param parameters = {0, 0};
	bt_le_ext_adv_start(advertising_set, &parameters);
	k_sleep(K_MSEC(100));
	bt_le_ext_adv_stop(advertising_set);
}
executed every 5 seconds.

joerchan, 2021-03-05:
@ppluta The extended advertising support in the zephyr controller is still a work in progress.
If you wish to use a stable one you can switch to the SoftDevice controller using NCS.

@cvinayak Do you know what could be the cause of the crash?

cvinayak, 2021-03-05:
Not sure, is the below valid use?
```
	struct bt_le_ext_adv_start_param parameters = {0, 0};
	bt_le_ext_adv_start(advertising_set, &parameters);
```

Could you try using this: https://github.com/zephyrproject-rtos/zephyr/pull/32754

ppluta, 2021-03-05:
bt_le_ext_adv_start() returned 0 for this. Besides, other values than either 0 allowed for at most one successful execution of bt_le_ext_adv_start() and subsequent ones failed with opcode 0x2039 returning 0xc and 0x7, but I did not investigate it more precisely.

joerchan, 2021-03-08:
Vinayak: `struct bt_le_ext_adv_start_param parameters = {0, 0};` is valid, it would be the same as doing `bt_le_ext_adv_start(advertising_set, NULL);`, i.e setting both timeout values (number of completed events, and timeout)  to 0.

cvinayak, 2021-03-15:
@ppluta Please verify this fix: https://github.com/zephyrproject-rtos/zephyr/pull/33328

ppluta, 2021-03-15:
How do I do it? Will west update do?

cvinayak, 2021-03-15:
@ppluta 

‘’’
git fetch git@github.com:cvinayak/zephyr.git github_ext_adv_stop_fix && git cherry-pick FETCH_HEAD
‘’’

ppluta, 2021-03-15:
OK, it ran for an hour and then for another half and nothing happened. So this reset seems to be cured.

PR #33328 (2021-03-17): Bluetooth: controller: Fix assertion on Ext. Advertising disable
Fix an assertion when disabling Extended Advertising with
auxiliary PDUs. Stop the auxiliary PDU scheduling after the
primary PDU scheduling is stopped first. This will avoid the
assertion caused due to a search for a stopped auxiliary PDU
offset by the primary PDU scheduling that is being stopped.

Fixes #32866.
commit 7f62c7d5 (2021-03-17): Bluetooth: controller: Fix assertion on Ext. Advertising disable

Fix an assertion when disabling Extended Advertising with
auxiliary PDUs. Stop the auxiliary PDU scheduling after the
primary PDU scheduling is stopped first. This will avoid the
assertion caused due to a search for a stopped auxiliary PDU
offset by the primary PDU scheduling that is being stopped.

Fixes #32866.
closed 2021-03-17 by carlescufi