zephyrproject-rtos/zephyr #32867: tests/kernel/sched/schedule_api does not start with stm32wb55 on nucleo

FRASTM, 2021-03-04:
**Describe the bug**
when testing the tests/kernel/sched/schedule_api on the nucleo_wb55rg board form STMicroelectronics, the test does not start.

**To Reproduce**
Steps to reproduce the behavior:
1. west build -p auto -b nucleo_wb55rg tests/kernel/sched/schedule_api/
2. west flash

**Expected behavior**
test PASSED

**Logs and console output**
none

**Environment (please complete the following information):**
 - Commit SHA 6eb6c88554ad0c4c55c96cb940442665a53769ee

**Additional context**
The kernel is falling in the 
`__ASSERT((CoreDebug->DHCSR & CoreDebug_DHCSR_C_DEBUGEN_Msk) == 0,`
of the arch/arm/include/aarch32/cortex_m/dwt.h
This is a new ASSERT command which is added by the  PR

FRASTM, 2021-03-04:
It seems that with the nucleo_wb55rg, the CoreDebug  is at address 0xe000edf0: 
```
0xe000edf0:	0x00030003	0x00000000	0x00000000	0x01000000
0xe000ee00:	0x00000000	0x00000000	0x00000000	0x00000000
0xe000ee10:	0x00000000	0x00000000	0x00000000	0x00000000
```

erwango, 2021-03-05:
We actually have a whole bunch of tests failing with this commit on various platforms (tests/drivers/gpio , ...).
Setting prio to high

ioannisg, 2021-03-05:
Will take a look asap, thanks.

ioannisg, 2021-03-05:
@erwango I guess the problem is that you are trying to execute tests on the board with the Debug mode on. But this should not be the case, right? I believe that west flash should never leave the Debug mode on.

erwango, 2021-03-05:
@ioannisg, confirmed this is due to debug mode being enabled by runner. As discussed, waiting for this to be fixed in various runners (openocd at least), let's remove the assert temporarily.

ioannisg, 2021-03-05:
@erwango removing the assert will still make several tests failing due to a raised DebugMonitor exception, it is not a complete workaround. Better to disable null pointer exception detection feature in these stm32 platforms where west flash leaves the debug mode on.

erwango, 2021-03-05:
> Better to disable null pointer exception detection feature in these stm32 platforms where west flash leaves the debug mode on.

@ioannisg That's the whole family then (which is easier). Ok, let me do that.

ioannisg, 2021-03-05:
That' fine with me. As discussed offline, we need to track the openOCD issue (ideally in a different ticket).

galak, 2021-03-05:
> @erwango I guess the problem is that you are trying to execute tests on the board with the Debug mode on. But this should not be the case, right? I believe that west flash should never leave the Debug mode on.
> 
> If some runners do that, I would argue we need to fix the runners instead. I noticed the problem in openOCD, but might be more runners that do the same...

What is the Debug mode referenced here.  Are we talking about building Zephyr with DEBUG enabled or some hardware mode?

ioannisg, 2021-03-05:
Hardware mode. It's about the debugger leaving the DHCSR bit0 to 1 after flashing.

erwango, 2021-03-08:
@ioannisg @galak, I think that https://github.com/zephyrproject-rtos/zephyr/pull/31481 was merged too fast. If https://github.com/zephyrproject-rtos/zephyr/pull/32983 is not a proper fox (which I totally understand), then CONFIG_CORTEX_M_DEBUG_NULL_POINTER_EXCEPTION_DETECTION_DWT should be default n waiting for https://github.com/zephyrproject-rtos/zephyr/issues/32984 to be fixed.

FRASTM, 2021-03-09:
This issue is fixed, **tests/kernel/sched/schedule_api** passed on nucleo_wb55rg 
with master b82a662866eb10c700f31437371bce44694a7604

```
*** Booting Zephyr OS build zephyr-v2.5.0-977-gf9dee026d594  ***
Running test suite threads_scheduling
===================================================================
START - test_bad_priorities
 PASS - test_bad_priorities
===================================================================
START - test_priority_cooperative
 PASS - test_priority_cooperative
===================================================================
START - test_priority_preemptible
 PASS - test_priority_preemptible
===================================================================
START - test_priority_preemptible_wait_prio
 PASS - test_priority_preemptible_wait_prio
===================================================================             
START - test_yield_cooperative                                                  
 PASS - test_yield_cooperative                                                  
===================================================================             
START - test_sleep_cooperative                                                  
 PASS - test_sleep_cooperative                                                  
===================================================================             
START - test_sleep_wakeup_preemptible                                           
 PASS - test_sleep_wakeup_preemptible                                           
===================================================================             
START - test_pending_thread_wakeup                                              
 PASS - test_pending_thread_wakeup                                              
===================================================================             
START - test_time_slicing_preemptible                                           
 PASS - test_time_slicing_preemptible                                           
===================================================================             
START - test_time_slicing_disable_preemptible                                   
 PASS - test_time_slicing_disable_preemptible                                   
===================================================================             
START - test_lock_preemptible                                                   
 PASS - test_lock_preemptible                                                   
===================================================================             
START - test_unlock_preemptible                                                 
 PASS - test_unlock_preemptible                                                 
===================================================================             
START - test_unlock_nested_sched_lock                                           
 PASS - test_unlock_nested_sched_lock                                           
===================================================================             
START - test_sched_is_preempt_thread                                            
 PASS - test_sched_is_preempt_thread                                            
===================================================================             
START - test_slice_reset                                                        
 PASS - test_slice_reset                                                        
===================================================================             
START - test_slice_scheduling                                                   
ABCDEFGHI                                                                       
ABCDEFGHI                                                                       
ABCDEFGHI                                                                       
ABCDEFGHI                                                                       
ABCDEFGHI                                                                       
ABCDEFGHI                                                                       
 PASS - test_slice_scheduling                                                   
===================================================================             
START - test_priority_scheduling                                                
ABCDEFGHI                                                                       
ABCDEFGHI                                                                       
ABCDEFGHI                                                                       
ABCDEFGHI                                                                       
ABCDEFGHI                                                                       
 PASS - test_priority_scheduling                                                
===================================================================             
START - test_wakeup_expired_timer_thread                                        
 PASS - test_wakeup_expired_timer_thread                                        
===================================================================             
START - test_user_k_wakeup                                                      
 PASS - test_user_k_wakeup                                                      
===================================================================             
START - test_user_k_is_preempt                                                  
 PASS - test_user_k_is_preempt                                                  
===================================================================             
Test suite threads_scheduling succeeded                                         
===================================================================             
PROJECT EXECUTION SUCCESSFUL 
```

closed 2021-03-09 by FRASTM