zephyrproject-rtos/zephyr #32885: Zephyr C++ support documentation [probably] conflicts to the code

evgeniy-paltsev, 2021-03-04:
Latest Zephyr documentation related to C++ support ([doc/reference/kernel/other/cxx_support.rst](https://github.com/zephyrproject-rtos/zephyr/blob/master/doc/reference/kernel/other/cxx_support.rst)) explicitly mentions that following features are *not* supported:

* Dynamic object management with the **new** and **delete** operators
* RTTI (runtime type information)
* Exceptions

However the commit 26e0f9a9e1267858848cbad0a145b87a4c07b2c6 says that it

* Add support of C++ exceptions
* Add support of C++ RTTI
* Implements new and delete using k_malloc and k_free

which is conflicting to the documentation statements.

Hence the question to C++ experts:
Do we still have some significant flaws in these features implementation so we consider them as *not* supported or we just have outdated documentation?

evgeniy-paltsev, 2021-03-04:
@nashif, @Benichou34 could you please take a look at this? 
Thanks!

nashif, 2021-03-04:
@evgeniy-paltsev  yes, docs might be outdated, here is a list of what works and what does not: https://github.com/zephyrproject-rtos/zephyr/issues/31281

andygikling, 2022-03-26:
Bump, I think the very brief description in cxx_support.rst is misleading people (me).

andygikling, 2022-03-26:
Never mind I can confirm new is working.  For people who end up looking here, you need to change a few things in your menuconfig:
CONFIG_CPLUSPLUS=y
# Enable 17
CONFIG_STD_CPP17=y
# Newlib 
CONFIG_NEWLIB_LIBC=y
# Standar c++ libraries
CONFIG_LIB_CPLUSPLUS=y
# Support runtime type system
CONFIG_RTTI=y

stephanosio, 2022-03-27:
not stale

stephanosio, 2022-05-26:
Related to https://github.com/zephyrproject-rtos/zephyr/issues/43636

PR #46077 (2022-06-01): Toolchains and language support documentation rework
This series reworks the toolchain and language support-related documentations with various documentation enhancements and bug fixes.

Closes #36905
Fixes #43636 
Fixes #32885

Docs preview: https://builds.zephyrproject.io/zephyr/pr/46077/docs/
closed 2022-06-01 by carlescufi