zephyrproject-rtos/zephyr #32898: hci_usb: crash on repeated pairing attempt

LeBlue, 2021-03-04:
**Describe the bug**
hci_usb FW crashes with imprecise bus error after a repeated pairing attempt with secure only connections.

**To Reproduce**
Steps to reproduce the behavior:
On the central side I use the hci_usb FW with a linux/bluez 5.55 host. The peripheral is a custom FW with secure only connections that uses `bt_conn_set_security(conn, BT_SECURITY_L4)` in the connect cb.

1. Connect and pair the peripheral
2. Disconnect the peripheral and remove its stored keys
3. Try reconnecting. bluez provides the now outdated keys and establishing secure connction fails.
4. Remove the keys in bluez
5. Reconnect the peripheral. The key exchange procedure crashes the central (hci_usb FW)

Unfortunately I was not able to reproduce the issue with the peripheral_sc_only sample.

If step 3. is removed, everything works as expected.
If after step 3. bluez or the FW is reset, everything works as expected.

The behaviour is reproducable almost 100%, sometimes the hci_usb FW just "hangs" instead of crashing.

**Expected behavior**
hci_usb FW should not crash or hang.

**Impact**
Would be showstopper, but workarounds exist.

**Logs and console output**

btmon log

```
< HCI Command: LE Create Conne.. (0x08|0x000d) plen 25  #6065 [hci0] 13669.632058
        Scan interval: 60.000 msec (0x0060)
        Scan window: 60.000 msec (0x0060)
        Filter policy: White list is not used (0x00)
        Peer address type: Random (0x01)
        Peer address: DA:16:BC:11:74:7E (Static)
        Own address type: Random (0x01)
        Min connection interval: 50.00 msec (0x0028)
        Max connection interval: 70.00 msec (0x0038)
        Connection latency: 0 (0x0000)
        Supervision timeout: 420 msec (0x002a)
        Min connection length: 0.000 msec (0x0000)
        Max connection length: 0.000 msec (0x0000)
> HCI Event: Command Status (0x0f) plen 4               #6066 [hci0] 13669.634932
      LE Create Connection (0x08|0x000d) ncmd 1
        Status: Success (0x00)
> HCI Event: LE Meta Event (0x3e) plen 19               #6067 [hci0] 13670.633029
      LE Connection Complete (0x01)
        Status: Success (0x00)
        Handle: 0
        Role: Master (0x00)
        Peer address type: Random (0x01)
        Peer address: DA:16:BC:11:74:7E (Static)
        Connection interval: 70.00 msec (0x0038)
        Connection latency: 0 (0x0000)
        Supervision timeout: 420 msec (0x002a)
        Master clock accuracy: 0x06
@ MGMT Event: Device Connected (0x000b) plen 16      {0x0001} [hci0] 13670.633124
        LE Address: DA:16:BC:11:74:7E (Static)
        Flags: 0x00000000
        Data length: 3
        Flags: 0x06
          LE General Discoverable Mode
          BR/EDR Not Supported
< HCI Command: LE Read Remote U.. (0x08|0x0016) plen 2  #6068 [hci0] 13670.633420
        Handle: 0
> HCI Event: LE Meta Event (0x3e) plen 4                #6069 [hci0] 13670.634051
      LE Channel Selection Algorithm (0x14)
        Handle: 0
[... 207 lines cut ...]
4080		handle = bt_acl_handle(handle);

4081
4082		node_tx = ll_tx_mem_acquire();

4083		if (!node_tx) {
   0x00009ffa <+46>:	mov	r4, r0
   0x00009ffc <+48>:	cmp	r0, #0
   0x00009ffe <+50>:	beq.n	0xa082 <hci_acl_handle+182>

4084			BT_ERR("Tx Buffer Overflow");
   0x0000a082 <+182>:	ldr	r3, [pc, #92]	; (0xa0e0 <hci_acl_handle+276>)
---- snip
```

From another run (`free` value is garbage, should be SRAM addr? Probably dereferencing caused bus error `ll_tx_mem_acquire` -> `mem_aquire(mem_conn_tx)`)
```
(gdb) p/x mem_conn_tx
$6 = {free = 0x5a1820, pool = {0x10, 0x26, 0x0, 0x20, 0x5, 0x0, 0x0, 0xb5, 0x4f, 0xf, 0xde, 0x1e, 0x99, 0x1f, 0x50, 0xf9, 0xf4, 0x87, 0xb, 0x9a, 0xfd, 0x5e, 0x5, 0x15, 0xe0,
    0x5e, 0x6a, 0x29, 0xa7, 0x17, 0x33, 0xd3, 0x3d, 0x1d, 0x0, 0x0, 0x7c, 0x26, 0x0, 0x20, 0x4, 0x0, 0x0, 0x41, 0x0, 0x6, 0x0, 0xc, 0xe2, 0x83, 0x73, 0xca, 0x6a, 0xb0, 0xac,
    0x97, 0x4, 0xd0, 0x90, 0xe7, 0x68, 0xe2, 0x76, 0x47, 0x2d, 0x64, 0x5c, 0xc3, 0xeb, 0xe1, 0x0, 0x0, 0x58, 0x26, 0x0, 0x20, 0x7, 0x0, 0x0, 0x3, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0,
    0x0, 0x0, 0x0, 0x0, 0xf0, 0xaa, 0xad, 0xba, 0x2e, 0x5f, 0x22, 0x30, 0x37, 0xdb, 0x39, 0xfc, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0xec, 0x25, 0x0, 0x20, 0x6, 0x0, 0x0, 0x11, 0x0,
    0x6, 0x0, 0x3, 0x22, 0xe5, 0xc2, 0xd5, 0xe7, 0x11, 0x91, 0x26, 0xb7, 0xb0, 0x25, 0x3d, 0x78, 0x2a, 0x64, 0xe3, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0xa0, 0x26, 0x0,
    0x20, 0x3, 0x0 <repeats 31 times>, 0xc4, 0x26, 0x0, 0x20, 0x0 <repeats 68 times>}}
```

LeBlue, 2021-03-06:
**Further analysis**

https://github.com/zephyrproject-rtos/zephyr/blob/6cf2fdc5d1ebb9534aa980c754b1e13f5daca2fb/subsys/bluetooth/controller/ll_sw/ull_conn.c#L150-L153
```
offsetof(struct node_tx, pdu): 4
offsetof(struct pdu_data, llctrl): 3
sizeof(struct pdu_data_llctrl): 24

sizeof(pdu_data_llctrl): 24
sizeof(pdu_data_llctrl_conn_update_ind): 11
sizeof(pdu_data_llctrl_chan_map_ind): 7
sizeof(pdu_data_llctrl_terminate_ind): 1
sizeof(pdu_data_llctrl_enc_req): 22
sizeof(pdu_data_llctrl_enc_rsp): 12
sizeof(pdu_data_llctrl_start_enc_req): 0
sizeof(pdu_data_llctrl_start_enc_rsp): 0
sizeof(pdu_data_llctrl_unknown_rsp): 1
sizeof(pdu_data_llctrl_feature_req): 8
sizeof(pdu_data_llctrl_feature_rsp): 8
sizeof(pdu_data_llctrl_pause_enc_req): 0
sizeof(pdu_data_llctrl_pause_enc_rsp): 0
sizeof(pdu_data_llctrl_version_ind): 5
sizeof(pdu_data_llctrl_reject_ind): 1
sizeof(pdu_data_llctrl_slave_feature_req): 8
sizeof(pdu_data_llctrl_conn_param_req): 23
sizeof(pdu_data_llctrl_conn_param_rsp): 23
sizeof(pdu_data_llctrl_reject_ext_ind): 2
sizeof(pdu_data_llctrl_ping_req): 0
sizeof(pdu_data_llctrl_ping_rsp): 0
sizeof(pdu_data_llctrl_length_req): 8
sizeof(pdu_data_llctrl_length_rsp): 8
sizeof(pdu_data_llctrl_phy_req): 2
sizeof(pdu_data_llctrl_phy_rsp): 2
sizeof(pdu_data_llctrl_phy_upd_ind): 4
sizeof(pdu_data_llctrl_min_used_chans_ind): 2
CONN_TX_CTRL_BUF_SIZE: 32
```

-> crashes

with:
```
#define CONN_TX_CTRL_BUF_SIZE MROUND(offsetof(struct node_tx, pdu) + \
				     offsetof(struct pdu_data, llctrl) + \
				     sizeof(struct pdu_data_llctrl) + 2)

CONN_TX_CTRL_BUF_SIZE: 36
```
-> OK

**Increased CONN_TX_CTRL_BUF_SIZE**

did a mem dump with the increased size of `CONN_TX_CTRL_BUF_SIZE`

mem layout `mem_conn_tx_ctrl`
1. word: free pointer
each 9 words: 
	1 word node_tx (link/next pointer)
	1 word llctrl + opcode
	rest pdu_ctrl data

After reset:

```
(gdb) p/x &mem_conn_tx_ctrl
$2 = 0x20002918
(gdb) x/100wx 0x20002918
0x20002918 <mem_conn_tx_ctrl>:	0x2000291c	0x20002940	0x00000008	0x00000000
0x20002928 <mem_conn_tx_ctrl+16>:	0x00000000	0x00000000	0x00000000	0x00000000
0x20002938 <mem_conn_tx_ctrl+32>:	0x00000000	0x00000000	0x20002964	0x00000000
[... 57 lines cut ...]
```

and from btmon (note: DATA_LEN == 27)

```
< ACL Data TX: Handle 0 flags 0x00 dlen 27                    #395 [hci0] 409.786150
< ACL Data TX: Handle 0 flags 0x01 dlen 27                    #396 [hci0] 409.786186
< ACL Data TX: Handle 0 flags 0x01 dlen 15                    #397 [hci0] 409.786204
      SMP: Pairing Public Key (0x0c) len 64
        X: 5064bbf998ded5ac6916dad740779f1041a1ed46b03202912e3bb2fbc2add533
        Y: 011608432354e6c6796ab083abe6877f0c6ac6a024b2f6bf326ea6ed7998f6c6
```

Disected:
```
X: 5064bbf9 98ded5ac 6916dad7 40779f10 41a1ed46b0 3202  <- value in node 1 (bytes shown swapped)
```

Why is there a data pdu in the mem_conn_tx_ctrl? thats the BUG.

ASSERTing on the value of `tx_node`

https://github.com/zephyrproject-rtos/zephyr/blob/6cf2fdc5d1ebb9534aa980c754b1e13f5daca2fb/subsys/bluetooth/controller/hci/hci.c#L4082

reveals that `ll_tx_mem_acquire` returns a buffer from the wrong pool.

LeBlue, 2021-03-08:
**1. `ll_tx_mem_acquire` returns buffer from wrong pool**

Therefore the wrong pool is used to construct a data pdu (and overflows buffer, because it is to small for this data).

Following backtraces are on [v2.5-branch@6cf2fdc5d1](https://github.com/zephyrproject-rtos/zephyr/tree/6cf2fdc5d1ebb9534aa980c754b1e13f5daca2fb/subsys/bluetooth/controller/ll_sw)

When reconnecting (step 5) with keys removed from central and peripheral.

```
Breakpoint 2, log_const_source_id (data=<optimized out>) at /Users/matthiaswauer/Projects/nordic_hci_ble/ble_fw/firmware/zephyr/zephyr/subsys/bluetooth/controller/hci/hci.c:4085
4085			BT_ERR("Tx Buffer runnaway: %p", node_tx);
(gdb) bt
#0  log_const_source_id (data=<optimized out>) at /Users/matthiaswauer/Projects/nordic_hci_ble/ble_fw/firmware/zephyr/zephyr/subsys/bluetooth/controller/hci/hci.c:4085
#1  hci_acl_handle (buf=0x20005ed8 <net_buf_hci_acl_pool+24>, evt=evt@entry=0x20003bfc <rx_thread_stack+980>)
    at /Users/matthiaswauer/Projects/nordic_hci_ble/ble_fw/firmware/zephyr/zephyr/subsys/bluetooth/controller/hci/hci.c:4085
#2  0x00008f02 in acl_handle (buf=0x20005ed8 <net_buf_hci_acl_pool+24>)
    at /Users/matthiaswauer/Projects/nordic_hci_ble/ble_fw/firmware/zephyr/zephyr/subsys/bluetooth/controller/hci/hci_driver.c:463
#3  hci_driver_send (buf=0x20005ed8 <net_buf_hci_acl_pool+24>)
    at /Users/matthiaswauer/Projects/nordic_hci_ble/ble_fw/firmware/zephyr/zephyr/subsys/bluetooth/controller/hci/hci_driver.c:463
#4  0x00004d0a in hci_rx_thread () at /Users/matthiaswauer/Projects/nordic_hci_ble/ble_fw/firmware/zephyr/zephyr/subsys/usb/class/bluetooth.c:164
#5  0x0001aa44 in z_thread_entry (entry=0x4ce5 <hci_rx_thread>, p1=<optimized out>, p2=<optimized out>, p3=<optimized out>)
    at /Users/matthiaswauer/Projects/nordic_hci_ble/ble_fw/firmware/zephyr/zephyr/lib/os/thread_entry.c:29
#6  0xaaaaaaaa in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb) l
4080		handle = bt_acl_handle(handle);
4081
4082		node_tx = ll_tx_mem_acquire();
4083		if ((uint32_t) node_tx >= 0x20002918)
4084		{
4085			BT_ERR("Tx Buffer runnaway: %p", node_tx);
4086			LL_ASSERT(!node_tx);
4087			return -ENOBUFS;
4088		}
4089
(gdb) p/x &mem_conn_tx_ctrl
$1 = 0x20002918
(gdb) frame 1
#1  hci_acl_handle (buf=0x20005ed8 <net_buf_hci_acl_pool+24>, evt=evt@entry=0x20003bfc <rx_thread_stack+980>)
    at /Users/matthiaswauer/Projects/nordic_hci_ble/ble_fw/firmware/zephyr/zephyr/subsys/bluetooth/controller/hci/hci.c:4085
4085			BT_ERR("Tx Buffer runnaway: %p", node_tx);
(gdb) p node_tx
$2 = (struct node_tx *) 0x20002940 <mem_conn_tx_ctrl+40>
```
This shows the FW state is already corrupted. The bug must be happening when disconnecting due to authentication failure.

**2. Checking tx pointer on `ll_tx_mem_release(tx)`**

If the `ll_tx_mem_acquire` returns from wrong pool means `ll_tx_mem_release` was called with pointer from `mem_tx_conn_ctrl` pool. Trying to find out where this happens, reveals:

on step 3 (connecting with invalid keys on the central side):

```
< HCI Command: LE Start Encryption (0x08|0x0019) plen 28   #2856 [hci0] 12710.579956
        Handle: 0
        Random number: 0x0000000000000000
        Encrypted diversifier: 0x0000
        Long term key: e3c44d54a9f3b81f12946828d763f57a
> HCI Event: Command Status (0x0f) plen 4                  #2857 [hci0] 12710.582752
      LE Start Encryption (0x08|0x0019) ncmd 1
        Status: Success (0x00)
> HCI Event: LE Meta Event (0x3e) plen 11                  #2858 [hci0] 12710.609753
      LE Data Length Change (0x07)
        Handle: 0
        Max TX octets: 27
        Max TX time: 2704
        Max RX octets: 27
        Max RX time: 2704
> HCI Event: Encryption Change (0x08) plen 4               #2859 [hci0] 12710.669765
        Status: PIN or Key Missing (0x06)
[... 81 lines cut ...]
Seems like `rx_demux_conn_tx_ack` and `ull_conn_tx_ack` fail to recognize the ctrl PDU. Added check on line:

https://github.com/zephyrproject-rtos/zephyr/blob/6cf2fdc5d1ebb9534aa980c754b1e13f5daca2fb/subsys/bluetooth/controller/ll_sw/ull_conn.c#L1576

```
Breakpoint 3, ull_conn_tx_ack (handle=<optimized out>, link=link@entry=0x20002a28 <mem_link_tx+12>, tx=0x2000291c <mem_conn_tx_ctrl+4>)
    at /Users/matthiaswauer/Projects/nordic_hci_ble/ble_fw/firmware/zephyr/zephyr/subsys/bluetooth/controller/ll_sw/ull_conn.c:1602
1602					LL_ASSERT(0);
(gdb) l
1597				return;
1598			} else {
1599				LL_ASSERT(!link->next);
1600				if (tx >= (void*) &mem_conn_tx_ctrl) {
1601
1602					LL_ASSERT(0);
1603				}
1604			}
1605		} else if (handle == 0xFFFF) {
(gdb) p link
$1 = (memq_link_t *) 0x20002a28 <mem_link_tx+12>
(gdb) p link->next
$2 = (struct _memq_link *) 0x0 <z_impl_z_errno>
(gdb) p link->mem
$3 = (void *) 0x2000291c <mem_conn_tx_ctrl+4>
```

LeBlue, 2021-03-10:
**Preliminary conclusions**

The whole crash/memory corruption is triggered by the improper cleanup of a `PDU_DATA_LLCTRL_TYPE_LENGTH_RSP` (after peripheral disconnect request?). The fact that it is always this PDU might be either timing related or specific to this PDU.

The current workaround for me that prevents the bug is just to disable this feature in the central.
```
CONFIG_BT_DATA_LEN_UPDATE=n
```

carlescufi, 2021-03-10:
Setting priority to high since this is a crash in the Link Layer.

LeBlue, 2021-03-11:
@cvinayak

I am not sure if DEBUG_KEYS will be even necessary as the pdu release into wrong pool happens before keys are exchanged. The **crash** happens on next connection and pair attempt where the the buffer pool is already corrupted (step 5). The corruption already happens when bluez uses wrong/outdated keys to connect/update sec level (step 3).

Logs from step 3: reconnecting with invalid keys on central side

Added assertion:
```
void ll_tx_mem_release(void *tx)
{
	BT_ASSERT((tx >= (void *) &mem_conn_tx) && (tx < (void *) (&mem_conn_tx + 1)));
	mem_release(tx, &mem_conn_tx.free);
}
```
sniffer (picture and pcapng):
[image]
[pairing-cap-1.pcapng.zip](https://github.com/zephyrproject-rtos/zephyr/files/6123389/pairing-cap-1.pcapng.zip)

zephyr:
```
ASSERTION FAIL [(tx >= (void *) &mem_conn_tx) && (tx < (void *) (&mem_conn_tx + 1))] @ WEST_TOPDIR/zephyr/subsys/bluetooth/controller/ll_sw/ull_conn.c:233
[00:00:35.264,739] <inf> bt_ctlr_ull_conn: incoming Tx: count= 0, len= 0, rate= 0 bps.
[00:00:35.395,812] <err> os: r0/a1:  0x00000003  r1/a2:  0x0000000a  r2/a3:  0x00000001
[00:00:35.395,843] <err> os: r3/a4:  0x00004fb9 r12/ip:  0x00000000 r14/lr:  0x0000d813
[00:00:35.395,843] <err> os:  xpsr:  0x41000000
[00:00:35.395,843] <err> os: Faulting instruction address (r15/pc): 0x0000d81e
[00:00:35.395,843] <err> os: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0
[00:00:35.395,843] <err> os: Current thread: 0x20000948 (unknown)
[00:00:35.450,195] <err> os: Halting system
```

btmon:
```
< HCI Command: LE Create Connection (0x08|0x000d) plen 25      #443 [hci0] 9168.613674
        Scan interval: 60.000 msec (0x0060)
        Scan window: 60.000 msec (0x0060)
        Filter policy: White list is not used (0x00)
        Peer address type: Random (0x01)
        Peer address: DA:16:BC:11:74:7E (Static)
        Own address type: Random (0x01)
        Min connection interval: 30.00 msec (0x0018)
        Max connection interval: 30.00 msec (0x0018)
        Connection latency: 3 (0x0003)
        Supervision timeout: 7500 msec (0x02ee)
        Min connection length: 0.000 msec (0x0000)
        Max connection length: 0.000 msec (0x0000)
> HCI Event: Command Status (0x0f) plen 4                      #444 [hci0] 9168.614605
      LE Create Connection (0x08|0x000d) ncmd 1
        Status: Success (0x00)
> HCI Event: LE Meta Event (0x3e) plen 19                      #445 [hci0] 9169.614644
      LE Connection Complete (0x01)
        Status: Success (0x00)
        Handle: 0
        Role: Master (0x00)
        Peer address type: Random (0x01)
        Peer address: DA:16:BC:11:74:7E (Static)
        Connection interval: 30.00 msec (0x0018)
        Connection latency: 3 (0x0003)
        Supervision timeout: 7500 msec (0x02ee)
        Master clock accuracy: 0x06
@ MGMT Event: Device Connected (0x000b) plen 16            {0x0001} [hci0] 9169.614723
        LE Address: DA:16:BC:11:74:7E (Static)
        Flags: 0x00000000
        Data length: 3
        Flags: 0x06
          LE General Discoverable Mode
          BR/EDR Not Supported
< HCI Command: LE Read Remote Used Fea.. (0x08|0x0016) plen 2  #446 [hci0] 9169.615031
        Handle: 0
> HCI Event: LE Meta Event (0x3e) plen 4                       #447 [hci0] 9169.615636
      LE Channel Selection Algorithm (0x14)
        Handle: 0
        Algorithm: #2 (0x01)
> HCI Event: Command Status (0x0f) plen 4                      #448 [hci0] 9169.616631
      LE Read Remote Used Features (0x08|0x0016) ncmd 1
        Status: Success (0x00)
> ACL Data RX: Handle 0 flags 0x02 dlen 6                      #449 [hci0] 9169.646625
      SMP: Security Request (0x0b) len 1
        Authentication requirement: Bonding, MITM, SC, No Keypresses (0x0d)
> HCI Event: LE Meta Event (0x3e) plen 12                      #450 [hci0] 9169.676621
      LE Read Remote Used Features (0x04)
        Status: Success (0x00)
        Handle: 0
        Features: 0x7f 0x41 0x01 0x00 0x00 0x00 0x00 0x00
          LE Encryption
          Connection Parameter Request Procedure
          Extended Reject Indication
          Slave-initiated Features Exchange
          LE Ping
          LE Data Packet Length Extension
          LL Privacy
          LE 2M PHY
          Channel Selection Algorithm #2
          Minimum Number of Used Channels Procedure
< HCI Command: LE Start Encryption (0x08|0x0019) plen 28       #451 [hci0] 9169.676752
        Handle: 0
        Random number: 0x0000000000000000
        Encrypted diversifier: 0x0000
        Long term key: 98c9a6d5f1f84142c74e4082f1565b1b
> HCI Event: Command Status (0x0f) plen 4                      #452 [hci0] 9169.679603
      LE Start Encryption (0x08|0x0019) ncmd 1
        Status: Success (0x00)
> HCI Event: Encryption Change (0x08) plen 4                   #453 [hci0] 9169.766608
        Status: PIN or Key Missing (0x06)
        Handle: 0
        Encryption: Disabled (0x00)
< HCI Command: Disconnect (0x01|0x0006) plen 3                 #454 [hci0] 9169.766719
        Handle: 0
        Reason: Authentication Failure (0x05)
> HCI Event: Command Status (0x0f) plen 4                      #455 [hci0] 9169.768602
      Disconnect (0x01|0x0006) ncmd 1
        Status: Success (0x00)
```

For completeness:
v2.5-branch@6cf2fdc5d1

Added to hci_usb `prj.conf`, makes no differences if 1 or multiple `MAX_CONN`
```
CONFIG_LOG=y
CONFIG_UART_CONSOLE=y
CONFIG_INIT_STACKS=y

CONFIG_BT_MAX_CONN=2
CONFIG_BT_CTLR_ADVANCED_FEATURES=y
CONFIG_BT_CTLR_LLCP_CONN=2
```

LeBlue, 2021-03-11:
@cvinayak 
Since I used the nrf52840dk as sniffer instead of debugger there are no gdb logs. Should this be necesssary or any additional info I can rerun the tests with both.

cvinayak, 2021-03-12:
@LeBlue Thank you for the sniffer log, this is sufficient. I am already analyzing a possible implementation bug in the way a paused control PDU is resumed. If I manage a fix, I will post a PR and ask you to help verify it.

PR #33310 (2021-03-18): Bluetooth: controller: Fix regression in ctrl tx queue handling
Fix control Tx buffer leak into data Tx pool that happens
after a cross-over control procedure response was paused due
to currently active encryption setup procedure, and a new
control Tx PDU in addition to the paused one is enqueued
thereafter.

When the control tx PDUs is resumed but not yet enqueued
towards the radio, if there is a new control Tx PDU enqueued
then the paused control Tx PDU is not set as the head of the
control PDUs in the Tx queue. This caused the paused control
Tx PDU to be associated with data Tx pool, hence  [... cut]
commit 7f727ac3 (2021-03-18): Bluetooth: controller: Fix regression in ctrl tx queue handling

Fix control Tx buffer leak into data Tx pool that happens
after a cross-over control procedure response was paused due
to currently active encryption setup procedure, and a new
control Tx PDU in addition to the paused one is enqueued
thereafter.

When the control tx PDUs is resumed but not yet enqueued
towards the radio, if there is a new control Tx PDU enqueued
then the paused control Tx PDU is not set as the head of the
control PDUs in the Tx queue. This caused the paused control
Tx PDU to be associated with data Tx pool, hence [... cut]
commit 0249a626 (2021-03-19): Bluetooth: controller: Fix regression in ctrl tx queue handling

Fix control Tx buffer leak into data Tx pool that happens
after a cross-over control procedure response was paused due
to currently active encryption setup procedure, and a new
control Tx PDU in addition to the paused one is enqueued
thereafter.

When the control tx PDUs is resumed but not yet enqueued
towards the radio, if there is a new control Tx PDU enqueued
then the paused control Tx PDU is not set as the head of the
control PDUs in the Tx queue. This caused the paused control
Tx PDU to be associated with data Tx pool, hence [... cut]
PR #34122 (2021-05-21): [Backport v1.14-branch] Bluetooth: controller: Fix regression in ctrl tx queue handling
Fix control Tx buffer leak into data Tx pool that happens
after a cross-over control procedure response was paused due
to currently active encryption setup procedure, and a new
control Tx PDU in addition to the paused one is enqueued
thereafter.

When the control tx PDUs is resumed but not yet enqueued
towards the radio, if there is a new control Tx PDU enqueued
then the paused control Tx PDU is not set as the head of the
control PDUs in the Tx queue. This caused the paused control
Tx PDU to be associated w [... cut]
closed 2021-03-18 by nashif