zephyrproject-rtos/zephyr #32902: [Coverity CID :219547] Improper use of negative value in tests/drivers/timer/nrf_rtc_timer/src/main.c

zephyrbot, 2021-03-05:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/bd97359a5338b2542d19011b6d6aa1d8d1b9cc3f/tests/drivers/timer/nrf_rtc_timer/src/main.c#L102

Category: Integer handling issues
Function: `test_basic`
Component: Tests
CID: [219547](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=219547)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/bd97359a5338b2542d19011b6d6aa1d8d1b9cc3f/tests/drivers/timer/nrf_rtc_timer/src/main.c#L102

```
96    
97         zassert_true(chan >= 0, "Failed to allocate RTC channel (%d).", chan);
98    
99         k_timeout_t t0 =
100             Z_TIMEOUT_TICKS(Z_TICK_ABS(z_tick_get() + K_MSEC(1).ticks));
101    
>>>     CID 219547:  Integer handling issues  (NEGATIVE_RETURNS)
>>>     "chan" is passed to a parameter that cannot be negative.
102         test_timeout(chan, t0, false);
103    
104         k_timeout_t t1 = K_MSEC(4);
105    
106         test_timeout(chan, t1, false);
107    
```

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

GunZi200, 2021-03-05:
Would a cast be an OK solution? 
```c
test_timeout((uint32_t)chan, t0, false);
```

It would need to be repeated a few times... Or perhaps define

```c
uint32_t channel = chan;
```

after the assert to not repeatedly use a cast.

I would think the `zassert_true` make this a false alarm.

nordic-krch, 2021-04-07:
Fixed by #33408

closed 2021-04-07 by nordic-krch