zephyrproject-rtos/zephyr #32906: [Coverity CID :219543] Out-of-bounds access in samples/boards/nrf/mesh/onoff_level_lighting_vnd_app/src/smp_svr.c

zephyrbot, 2021-03-05:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/bd97359a5338b2542d19011b6d6aa1d8d1b9cc3f/samples/boards/nrf/mesh/onoff_level_lighting_vnd_app/src/smp_svr.c#L46

Category: Memory - corruptions
Function: `smp_svr_init`
Component: Samples
CID: [219543](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=219543)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/bd97359a5338b2542d19011b6d6aa1d8d1b9cc3f/samples/boards/nrf/mesh/onoff_level_lighting_vnd_app/src/smp_svr.c#L46

```
40     STATS_SECT_DECL(smp_svr_stats) smp_svr_stats;
41    
42     void smp_svr_init(void)
43     {
44         int rc;
45    
>>>     CID 219543:  Memory - corruptions  (OVERRUN)
>>>     Overrunning struct type stats_hdr of 24 bytes by passing it to a function which accesses it at byte offset 24.
46         rc = STATS_INIT_AND_REG(smp_svr_stats, STATS_SIZE_32, "smp_svr_stats");
47         __ASSERT_NO_MSG(rc == 0);
48    
49         /* Register the built-in mcumgr command handlers. */
50     #ifdef CONFIG_MCUMGR_CMD_FS_MGMT
51         fs_mgmt_register_group();
```

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

galak, 2021-03-16:
@carlescufi can you assign this to the right person

trond-snekvik, 2021-04-20:
This is a false positive, as the passed `stats_hdr` structure is meant to be accessed out of bounds.

I am not able to put this in coverity though, as I still haven't been granted access.

closed 2021-04-20 by trond-snekvik