zephyrproject-rtos/zephyr #32910: [Coverity CID :219538] Illegal address computation in subsys/bluetooth/controller/ll_sw/nordic/lll/lll_adv_aux.c

zephyrbot, 2021-03-05:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/bd97359a5338b2542d19011b6d6aa1d8d1b9cc3f/subsys/bluetooth/controller/ll_sw/nordic/lll/lll_adv_aux.c#L92

Category: Memory - corruptions
Function: `init_connect_rsp_pdu`
Component: Bluetooth
CID: [219538](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=219538)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/bd97359a5338b2542d19011b6d6aa1d8d1b9cc3f/subsys/bluetooth/controller/ll_sw/nordic/lll/lll_adv_aux.c#L92

```
86    
87         /* Note: AdvA and InitA are updated before transmitting PDU */
88    
89         /* AdvA */
90         dptr += BDADDR_SIZE;
91         /* InitA */
>>>     CID 219538:  Memory - corruptions  (OVERRUN)
>>>     "dptr" evaluates to an address that is at byte offset 7 of an array of 1 bytes.
92         dptr += BDADDR_SIZE;
93    
94         ext_hdr_len = dptr - (uint8_t *)&com_hdr->ext_hdr;
95    
96         /* Finish Common ExtAdv Payload header */
97         com_hdr->adv_mode = 0;
```

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

cvinayak, 2021-04-07:
These are false issues, the dptr is an offset into a static PDU buffer memory in the same file with sufficient size.

carlescufi, 2021-04-07:
Marked as false positive in Coverity.

closed 2021-04-07 by carlescufi