zephyrproject-rtos/zephyr #32911: [Coverity CID :219537] Out-of-bounds access in samples/boards/nrf/system_off/src/retained.c

zephyrbot, 2021-03-05:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/bd97359a5338b2542d19011b6d6aa1d8d1b9cc3f/samples/boards/nrf/system_off/src/retained.c#L117

Category: Memory - corruptions
Function: `ram_range_retain`
Component: Samples
CID: [219537](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=219537)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/bd97359a5338b2542d19011b6d6aa1d8d1b9cc3f/samples/boards/nrf/system_off/src/retained.c

```
109    
110             uint32_t section_mask =
111                 (POWER_RAM_POWERSET_S0RETENTION_On
112                          << (section + POWER_RAM_POWERSET_S0RETENTION_Pos));
113    
114             if (enable) {
>>>     CID 219537:    (OVERRUN)
>>>     Overrunning callee's array of size 9 by passing argument "block" (which evaluates to 9) in call to "nrf_power_rampower_mask_on".
115                 nrf_power_rampower_mask_on(NRF_POWER, block, section_mask);
116             } else {
117                 nrf_power_rampower_mask_off(NRF_POWER, block, section_mask);
118             }
119    
120             /* Move to the first address in the next section. */
111                 (POWER_RAM_POWERSET_S0RETENTION_On
112                          << (section + POWER_RAM_POWERSET_S0RETENTION_Pos));
113    
114             if (enable) {
115                 nrf_power_rampower_mask_on(NRF_POWER, block, section_mask);
116             } else {
>>>     CID 219537:    (OVERRUN)
>>>     Overrunning callee's array of size 9 by passing argument "block" (which evaluates to 9) in call to "nrf_power_rampower_mask_off".
117                 nrf_power_rampower_mask_off(NRF_POWER, block, section_mask);
118             }
119    
120             /* Move to the first address in the next section. */
121             addr += section_size - (addr % section_size);
122         } while (addr < addr_end);
```

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

galak, 2021-03-16:
@carlescufi can you assign this to the right person

carlescufi, 2021-04-19:
Marked as false positive in Coverity.

closed 2021-04-19 by carlescufi