zephyrproject-rtos/zephyr #32913: [Coverity CID :219535] Dereference after null check in drivers/sensor/icm42605/icm42605.c

zephyrbot, 2021-03-05:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/bd97359a5338b2542d19011b6d6aa1d8d1b9cc3f/drivers/sensor/icm42605/icm42605.c#L139

Category: Null pointer dereferences
Function: `icm42605_tap_fetch`
Component: Drivers
CID: [219535](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=219535)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/bd97359a5338b2542d19011b6d6aa1d8d1b9cc3f/drivers/sensor/icm42605/icm42605.c#L139

```
133                 result = inv_spi_read(REG_APEX_DATA4,
134                               drv_data->fifo_data, 1);
135                 if (drv_data->fifo_data[0] & APEX_TAP) {
136                     if (drv_data->tap_trigger.type ==
137                         SENSOR_TRIG_TAP) {
138                         LOG_DBG("Single Tap detected");
>>>     CID 219535:  Null pointer dereferences  (FORWARD_NULL)
>>>     Dereferencing null pointer "drv_data->tap_handler".
139                         drv_data->tap_handler(dev
140                               , &drv_data->tap_trigger);
141                     } else {
142                         LOG_ERR("Trigger type is mismatched");
143                     }
144                 } else if (drv_data->fifo_data[0] & APEX_DOUBLE_TAP) {
```

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

MaureenHelm, 2021-03-05:
@JHKim8212 will you please take a look at this?

JHKim8212, 2021-03-07:
@MaureenHelm I will add null check code at Mar 9th after test.

PR #33199 (2021-03-30): driver: sensor: icm42605: Fixed Dereference afterr null check issue
Fixed Coverity CID: 219524 : Dereference after null check
For fix this issue added null check before call function pointer

Fixes #32913
commit b486c35b (2021-03-30): driver: sensor: icm42605: Fixed Dereference afterr null check issue

Fixed Coverity CID: 219524 : Dereference after null check
For fix this issue added null check before call function pointer

Fixes #32913
closed 2021-03-30 by nashif