zephyrproject-rtos/zephyr #32916: [Coverity CID :219530] Dereference before null check in drivers/modem/ublox-sara-r4.c

zephyrbot, 2021-03-05:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/bd97359a5338b2542d19011b6d6aa1d8d1b9cc3f/drivers/modem/ublox-sara-r4.c#L334

Category: Null pointer dereferences
Function: `send_socket_data`
Component: Drivers
CID: [219530](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=219530)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/bd97359a5338b2542d19011b6d6aa1d8d1b9cc3f/drivers/modem/ublox-sara-r4.c#L334

```
328         }
329    
330         if (!dst_addr && sock->ip_proto == IPPROTO_UDP) {
331             dst_addr = &sock->dst;
332         }
333    
>>>     CID 219530:  Null pointer dereferences  (REVERSE_INULL)
>>>     Null-checking "sock" suggests that it may be null, but it has already been dereferenced on all paths leading to the check.
334         if (!sock) {
335             return -EINVAL;
336         }
337    
338         /*
339          * Binary and ASCII mode allows sending MDM_MAX_DATA_LENGTH bytes to
```

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

galak, 2021-03-16:
@XavierChapron could you look at this?

XavierChapron, 2021-03-17:
@galak Sure! Done in https://github.com/zephyrproject-rtos/zephyr/pull/33411

galak, 2021-03-17:
@XavierChapron thanks!

PR #33411 (2021-03-17): drivers: modem: ublox-sara-r4: Fix dereference before null check
Fixes #32916
commit 30c9588a (2021-03-17): drivers: modem: ublox-sara-r4: Fix dereference before null check

Fixes #32916
closed 2021-03-17 by galak