zephyrproject-rtos/zephyr #32917: [Coverity CID :219529] Unchecked return value in subsys/bluetooth/controller/ll_sw/ull_filter.c

zephyrbot, 2021-03-05:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/bd97359a5338b2542d19011b6d6aa1d8d1b9cc3f/subsys/bluetooth/controller/ll_sw/ull_filter.c#L1121

Category: Error handling issues
Function: `target_resolve`
Component: Bluetooth
CID: [219529](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=219529)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/bd97359a5338b2542d19011b6d6aa1d8d1b9cc3f/subsys/bluetooth/controller/ll_sw/ull_filter.c#L1121

```
1115          * Pass param FILTER_IDX_NONE if RPA can not be resolved,
1116          * or index in cache if it can be resolved
1117          */
1118         if (twork->cb) {
1119             mfy.fp = twork->cb;
1120             mfy.param = (void *) ((unsigned int) j);
>>>     CID 219529:  Error handling issues  (CHECKED_RETURN)
>>>     Calling "mayfly_enqueue" without checking return value (as is done elsewhere 15 out of 17 times).
1121             mayfly_enqueue(TICKER_USER_ID_THREAD,
1122                        TICKER_USER_ID_LLL, 1, &mfy);
1123         }
1124     }
1125    
1126     static uint8_t prpa_cache_try_resolve(bt_addr_t *rpa)
```

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

GunZi200, 2021-03-07:
I couldn’t find any documentation for the mayfly code. https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/bluetooth/controller/util/mayfly.c there are no doxygen comments in `mayfly.h`

Seems to return `0` or `1` but it is unclear to me what those values mean. 

I’m curious what a “mayfly” is 🤔

cvinayak, 2021-03-08:
mayfly is an internal framework in the Bluetooth Low Energy Controller implementation to schedule a function's execution in a different execution context. It can be seen as a software interrupt mechanism.

PR #35426 (2021-05-19): Bluetooth: controller: Various coverity related fixes
Fix side effect in assertion when checking a volatile
variable inside assert check.

Fixes #32904.
Fixes #32923.

Explicitly ignore return value from call to mayfly_enqueue.

Fixes #32917.
Fixes #32961.

Add assert check to avoid deferencing null return value by
ull_adv_set_get() function.

Fixes #35347.
commit 7d1a959d (2021-05-19): Bluetooth: controller: Explicitly ignore return value

Explicitly ignore return value from call to mayfly_enqueue.

Fixes #32917, #32961.