zephyrproject-rtos/zephyr #32931: [Coverity CID :219514] Out-of-bounds access in tests/drivers/timer/nrf_rtc_timer/src/main.c

zephyrbot, 2021-03-05:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/bd97359a5338b2542d19011b6d6aa1d8d1b9cc3f/tests/drivers/timer/nrf_rtc_timer/src/main.c

Category: Memory - corruptions
Function: `test_stress`
Component: Tests
CID: [219514](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=219514)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/bd97359a5338b2542d19011b6d6aa1d8d1b9cc3f/tests/drivers/timer/nrf_rtc_timer/src/main.c#L278

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

GunZi200, 2021-03-05:
I'm wondering if inside `test_timeout()`

```c
int32_t cc_val = z_nrf_rtc_timer_get_ticks(t);
```

could be the issue for this Github issue, and https://github.com/zephyrproject-rtos/zephyr/issues/32943

There is no check to confirm the value is not negative. `z_nrf_rtc_timer_compare_set` expects `uint32_t` not `int32_t`. I can't see any details of what caused this report but maybe something's happening inside `set_absolute_alarm()`?

jenmwms, 2021-03-06:
Thanks @GunZi200! I'm not sure yet, good question. I also saw in Coverity that this was marked Fix Submitted last month (updated by me but I don't recall if it was merged). I'll see if I can find the PR that tried to address this.

closed 2021-03-31 by galak