zephyrproject-rtos/zephyr #32935: [Coverity CID :219510] Improper use of negative value in tests/drivers/timer/nrf_rtc_timer/src/main.c

zephyrbot, 2021-03-05:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/bd97359a5338b2542d19011b6d6aa1d8d1b9cc3f/tests/drivers/timer/nrf_rtc_timer/src/main.c

Category: Integer handling issues
Function: `test_absolute_scheduling`
Component: Tests
CID: [219510](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=219510)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/bd97359a5338b2542d19011b6d6aa1d8d1b9cc3f/tests/drivers/timer/nrf_rtc_timer/src/main.c#L221

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

GunZi200, 2021-03-05:
Could the issue here be that we're missing asserts for `rtc_ticks` to check if its not negative? `z_nrf_rtc_timer_compare_set` expects `uint32_t` but `rtc_ticks` is `int`.

Another issue is `uint64_t evt_uptime_us;` is not initalised.

nordic-krch, 2021-04-07:
Fixed by #33408

closed 2021-04-07 by nordic-krch