zephyrproject-rtos/zephyr #32940: [Coverity CID :219502] Improper use of negative value in tests/net/socket/af_packet/src/main.c

zephyrbot, 2021-03-05:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/bd97359a5338b2542d19011b6d6aa1d8d1b9cc3f/tests/net/socket/af_packet/src/main.c

Category: Integer handling issues
Function: `__test_packet_sockets`
Component: Tests
CID: [219502](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=219502)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/bd97359a5338b2542d19011b6d6aa1d8d1b9cc3f/tests/net/socket/af_packet/src/main.c

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

GunZi200, 2021-03-15:
I believe this is a false positive just like https://github.com/zephyrproject-rtos/zephyr/issues/32942

@jukkar final coverity af_packet issue 😃

jukkar, 2021-03-17:
Yep, this is false positive. @GunZi200 thanks for all the help with these issues :+1:

closed 2021-03-17 by jukkar