zephyrproject-rtos/zephyr #32942: [Coverity CID :219499] Argument cannot be negative in tests/net/socket/af_packet/src/main.c

zephyrbot, 2021-03-05:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/bd97359a5338b2542d19011b6d6aa1d8d1b9cc3f/tests/net/socket/af_packet/src/main.c

Category: Error handling issues
Function: `prepare_udp_socket`
Component: Tests
CID: [219499](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=219499)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/bd97359a5338b2542d19011b6d6aa1d8d1b9cc3f/tests/net/socket/af_packet/src/main.c#L177

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

GunZi200, 2021-03-13:
I think this is a false positive since `sock` is checked with an assertion above:

https://github.com/zephyrproject-rtos/zephyr/blob/bd97359a5338b2542d19011b6d6aa1d8d1b9cc3f/tests/net/socket/af_packet/src/main.c#L168

jukkar, 2021-03-15:
Yep, Coverity does not understand that zassert() does not return.

closed 2021-03-15 by jukkar