zephyrproject-rtos/zephyr #32944: [Coverity CID :219497] Logically dead code in subsys/bluetooth/host/gatt.c

zephyrbot, 2021-03-05:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/bd97359a5338b2542d19011b6d6aa1d8d1b9cc3f/subsys/bluetooth/host/gatt.c

Category: Control flow issues
Function: `cf_set_value`
Component: Bluetooth
CID: [219497](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=219497)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/bd97359a5338b2542d19011b6d6aa1d8d1b9cc3f/subsys/bluetooth/host/gatt.c#L510

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

joerchan, 2021-03-05:
This is logically dead because there is not enough bits yet defined to take this branch. Once more bits are added this code-path will be needed.

@jhedberg You agree that we should close this as intentional?

carlescufi, 2021-05-21:
> This is logically dead because there is not enough bits yet defined to take this branch. Once more bits are added this code-path will be needed.
> 
> @jhedberg You agree that we should close this as intentional?

I've looked at the code now and I agree completely. Will close this.

carlescufi, 2021-05-21:
Marked as false positive in Coverity.

closed 2021-05-21 by carlescufi