zephyrproject-rtos/zephyr #32950: [Coverity CID :219490] Unsigned compared against 0 in drivers/wifi/esp/esp.c

zephyrbot, 2021-03-05:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/bd97359a5338b2542d19011b6d6aa1d8d1b9cc3f/drivers/wifi/esp/esp.c

Category: Control flow issues
Function: `esp_dns_work`
Component: Drivers
CID: [219490](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=219490)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/bd97359a5338b2542d19011b6d6aa1d8d1b9cc3f/drivers/wifi/esp/esp.c#L297

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

mniestroj, 2021-03-16:
False positive, as ESP_MAX_DNS can be > 0 when DNS resolver is enabled.

closed 2021-03-16 by mniestroj

ceolin, 2021-03-16:
> False positive, as ESP_MAX_DNS can be > 0 when DNS resolver is enabled.

Hi @mniestroj please next time either you mark as false positive in Coverity (with the justification) or ask someone to do it, otherwise this issue will automatically be re-opened in the next scan. I've done it for this one though.

mniestroj, 2021-03-16:
@ceolin Thanks for letting me know. How can I get Coverity account/permission to do that next time?