zephyrproject-rtos/zephyr #32957: [Coverity CID :219483] Improper use of negative value in tests/drivers/timer/nrf_rtc_timer/src/main.c

zephyrbot, 2021-03-05:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/bd97359a5338b2542d19011b6d6aa1d8d1b9cc3f/tests/drivers/timer/nrf_rtc_timer/src/main.c

Category: Integer handling issues
Function: `test_reseting_cc`
Component: Tests
CID: [219483](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=219483)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/bd97359a5338b2542d19011b6d6aa1d8d1b9cc3f/tests/drivers/timer/nrf_rtc_timer/src/main.c#L314

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

GunZi200, 2021-03-05:
I just took a look at this ... (just looking at the code)

I am going to guess the issue is with the `chan` variable. Though I'm not certain how this report was generated.

`chan` is defined locally here: https://github.com/zephyrproject-rtos/zephyr/blob/bd97359a5338b2542d19011b6d6aa1d8d1b9cc3f/tests/drivers/timer/nrf_rtc_timer/src/main.c#L296

` z_nrf_rtc_timer_chan_alloc()` is defined here  https://github.com/nrfconnect/sdk-zephyr/blob/a2d2a5e169f7aec8d9de16787af99e59c56c09b7/drivers/timer/nrf_rtc_timer.c#L280
and returns `int` as expected.

The problem:

`z_nrf_rtc_timer_compare_set()` is defined like this: https://github.com/zephyrproject-rtos/zephyr/blob/dccfe76663c2f59b479c6533479428e3b6f17016/drivers/timer/nrf_rtc_timer.c#L216

It is expecting `chan` to be a 32-bit unsigned integer `uint32_t`. But we're inputing a simple `int`.

Again, I'm no expert but would this inconsistency be enough to generate the report?

GunZi200, 2021-03-05:
Another thought... is `test_reseting_cc` perhaps missing a `zassert_true(chan >= 0, "Failed to allocate RTC channel.");` statement? And thus if a negative value gets passed unchecked the zephyr bot complains?🤔

nordic-krch, 2021-04-07:
Assert was added by #32970

closed 2021-04-07 by nordic-krch